raspberrypi/linux #984: Strange MMC errors booting from Samsung EVO SDcards

PythonNut, 2015-05-25:
```
[   85.426681] request_module: runaway loop modprobe binfmt-0000
[   85.428508] request_module: runaway loop modprobe binfmt-0000
[   85.449229] request_module: runaway loop modprobe binfmt-0000
[   85.449778] request_module: runaway loop modprobe binfmt-0000
[   85.472247] [<800256f8>] (warn_slowpath_null) from [<80410298>] (bcm2835_mmc_transfer_dma+0x1a8/0x1e0)
[   85.485867] [<80410298>] (bcm2835_mmc_transfer_dma) from [<80411324>] (bcm2835_mmc_request+0xa8/0xc0)
[   85.498750] [<80411324>] (bcm2835_mmc_request) from [<803f9aa8>] (mmc_start_request+0xd4/0xf8)
[   85.511034] [<803f9aa8>] (mmc_start_request) from [<803fa834>] (mmc_start_req+0x2a8/0x37c)
[   85.523053] [<803fa834>] (mmc_start_req) from [<804090a4>] (mmc_blk_issue_rw_rq+0xc8/0xafc)
[   85.535168] [<804090a4>] (mmc_blk_issue_rw_rq) from [<80409bdc>] (mmc_blk_issue_rq+0x104/0x4d8)
[   85.547683] [<80409bdc>] (mmc_blk_issue_rq) from [<8040aedc>] (mmc_queue_thread+0xb8/0x158)
[   85.559891] [<8040aedc>] (mmc_queue_thread) from [<8003ffa0>] (kthread+0xe0/0xfc)
[   85.571240] [<8003ffa0>] (kthread) from [<8000eb88>] (ret_from_fork+0x14/0x20)
[   85.580610] ---[ end trace f869ecbd2c135221 ]---
[   85.587513] EXT4-fs error (device mmcblk0p2): ext4_iget:3921: inode #5515: comm udevd: bad extra_isize (5787 != 256)
[   85.587572] mmcblk0: unknown error -5 sending read/write command, card status 0x900
[   85.587586] blk_update_request: I/O error, dev mmcblk0, sector 2048
[   95.597794] mmc0: Timeout waiting for hardware interrupt.
[   95.605645] mmcblk0: error -110 transferring data, sector 801952, nr 8, cmd response 0x900, card status 0xb00
[   95.620166] mmcblk0: retrying using single block read
```

popcornmix, 2015-05-26:
Duplicate of https://github.com/raspberrypi/firmware/issues/397
Can you try updating to latest rpi-update firmware and possibly switching to the "sdhost" sdcard driver.

PythonNut, 2015-05-26:
Strangely, enough expanding the filesystem seems to have fixed the issue. Does that make any sense?

PythonNut, 2015-05-26:
Updating to the latest firmware fixed the issue.

Is there any reason I should switch to the `sdhost` driver given that my problem is already solved?

closed 2015-05-26 by PythonNut

Cy4n1d3, 2015-05-27:
sdhost is still in testing afaik, main reasons would be improved performance in certain scenarios.