raspberrypi/linux #985: MMC falling to PIO

kukabu, 2015-05-26:
I update RPi via rpi-update to 4f95c894c6a96b265d98f04a889e469833f7e996 and see it in the dmesg:

[    1.376992] mmc-bcm2835 20300000.mmc: mmc-bcm2835: Unable to initialise DMA channels. Falling back to PIO

kukabu, 2015-05-26:
May be it helps.

root@rpi:~# for F in /proc/device-tree/soc/mmc@7e300000/* ; do echo $F; hexdump -C $F; done/proc/device-tree/soc/mmc@7e300000/bus-width
00000000  00 00 00 04                                       |....|
00000004
/proc/device-tree/soc/mmc@7e300000/clocks
00000000  00 00 00 02                                       |....|
00000004
/proc/device-tree/soc/mmc@7e300000/compatible
00000000  62 72 63 6d 2c 62 63 6d  32 38 33 35 2d 6d 6d 63  |brcm,bcm2835-mmc|
00000010  00                                                |.|
00000011
/proc/device-tree/soc/mmc@7e300000/dma-names
00000000  74 78 00 72 78 00                                 |tx.rx.|
00000006
/proc/device-tree/soc/mmc@7e300000/interrupts
00000000  00 00 00 02 00 00 00 1e                           |........|
00000008
/proc/device-tree/soc/mmc@7e300000/linux,phandle
00000000  00 00 00 10                                       |....|
00000004
/proc/device-tree/soc/mmc@7e300000/name
00000000  6d 6d 63 00                                       |mmc.|
00000004
/proc/device-tree/soc/mmc@7e300000/phandle
00000000  00 00 00 10                                       |....|
00000004
/proc/device-tree/soc/mmc@7e300000/reg
00000000  7e 30 00 00 00 00 01 00                           |~0......|
00000008
/proc/device-tree/soc/mmc@7e300000/status
00000000  6f 6b 61 79 00                                    |okay.|
00000005
root@rpi:~# 

ps. Raspberry Pi Model B Revision 0x0f

skogsmaskin, 2015-05-26:
Same here, with Arch Linux ARM's official packages. According to <a href="http://archlinuxarm.org/forum/viewtopic.php?f=60&t=8874">this thread</a> it happened 3.18.13-1 -> 3.18.13-2.

notro, 2015-05-26:
See comment on commit https://github.com/raspberrypi/linux/commit/10c0f306936a7955d48e938422758cad49dd1056

notro, 2015-05-27:
Ping @popcornmix @pelwell 
MMC DMA is broken in rpi-3.18.y
https://github.com/raspberrypi/linux/commit/10c0f306936a7955d48e938422758cad49dd1056#commitcomment-11371485

kukabu, 2015-05-28:
@notro, thanks, it works

gregt590, 2015-05-29:
I just did a sudo apt-get update && sudo apt-get -y upgrade && sudo rpi-update to update my  Raspberry Pi model B to the latest standard distribution Raspian software and am now always getting the "Unable to initialize DMA channels" error as well upon reboots. 

As recorded in dmesg:
pi@talon5 ~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 3.18.14+ (dc4@dc4-XPS13-9333) (gcc version 4.8.3 20140303 (prerelease) (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03) ) #792 PREEMPT Sat May 23 15:53:19 BST 2015
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] Machine model: Raspberry Pi Model B
...
[    1.149003] bcm2835-cpufreq: min=700000 max=900000
[    1.149945] sdhci: Secure Digital Host Controller Interface driver
[    1.149969] sdhci: Copyright(c) Pierre Ossman
[    1.150445] mmc-bcm2835 20300000.mmc: mmc-bcm2835: Unable to initialise DMA channels. Falling back to PIO
[    1.191202] sdhci-pltfm: SDHCI platform and OF driver helper
...

Anything you want me to capture and post to help someone find the problem?

notro, 2015-05-29:
The problem is fixed in the source but there's no new kernel/firmware release yet.
You can go back to the previous firmware like this:
```
sudo rpi-update 56569584e2a65d50408ea56d467336446af69925
```

popcornmix, 2015-05-31:
@kukabu can you test with latest rpi-update firmware?

gregt590, 2015-05-31:
FYI, I first went ahead and re-installed the last production kernel/firmware release via:
sudo apt-get install --reinstall libraspberrypi-bin libraspberrypi-dev libraspberrypi-doc libraspberrypi0 raspberrypi-bootloader
to correct the problem falling back to:
 $ uname -a
Linux talon5 3.18.11+ #781 PREEMPT Tue Apr 21 18:02:18 BST 2015 armv6l GNU/Linux
and confirmed the problem did not occur.

Then today I did a rpi-update to test with the updated firmware again, now running:
 $ uname -a
Linux talon5 3.18.14+ #793 PREEMPT Sat May 30 13:15:19 BST 2015 armv6l GNU/Linux
and re-tested my application.  I can report I do not see the error reported and that my application appears to runs fine.

Thanks,
Greg

kukabu, 2015-05-31:
Fixed. Thanks.

closed 2015-05-31 by kukabu