raspberrypi/linux #990: LED control problem to build from sources

ivanfdez, 2015-05-29:
I have compiled the kernel:

git clone --depth=1 https://github.com/raspberrypi/linux

...
.

popcornmix, 2015-05-29:
Does it work with latest official kernel (got from rpi-update?)

ivanfdez, 2015-05-29:
with latest official kernel downloaded by rpi-update works (3.18.14) but not compiled from sources

popcornmix, 2015-05-29:
Are you on 3.18.y branch?
Have you copied the dtb files (from arch/arm/boot/dts) to sdcard card (in addition to kernel.img).

ivanfdez, 2015-05-29:
Yes, I use 3.18.y branch

Now I copied all dtb files into /boot, but not work

popcornmix, 2015-05-30:
Can you capture output of ``dmesg`` and ``sudo vcdbg log msg`` with the rpi-update kernel and your own one and I suspect we'll see a difference.

ivanfdez, 2015-05-30:
I have compiled the kernel 3.19 with the same results

I try to build from raspberry pi

ivanfdez, 2015-05-30:
with compiled kernel

.....
....

000862.605: Read command line from file 'cmdline.txt'
dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait debug

000867.757: Loading 'Image' from SD card
001380.167: No kernel trailer (run mkknlimg to fix) - assuming not DT-capable
001381.482: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
002956.836: vchiq_core: vchiq_init_state: slot_zero = 0x5bc80000, is_master = 1
002960.608: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
002964.642: gpioman: gpioman_get_pin_num: pin LEDS_RUNNING not defined
002964.665: gpioman: gpioman_get_pin_num: pin LEDS_NAND_ACTIVITY not defined
002964.687: gpioman: gpioman_get_pin_num: pin LEDS_USB_ACTIVITY not defined
002964.708: gpioman: gpioman_get_pin_num: pin LEDS_FATAL_ERROR not defined
002964.729: gpioman: gpioman_get_pin_num: pin LEDS_APP_OK not defined
002964.752: gpioman: gpioman_get_pin_num: pin LEDS_APP_FAILED not defined
002964.772: gpioman: gpioman_get_pin_num: pin LEDS_HDCP_AUTH not defined
002964.794: gpioman: gpioman_get_pin_num: pin LEDS_HDCP_UNAUTH not defined
002964.816: gpioman: gpioman_get_pin_num: pin LEDS_HDMI_ON not defined
002964.837: gpioman: gpioman_get_pin_num: pin LEDS_DVI_ON not defined
002964.859: gpioman: gpioman_get_pin_num: pin LEDS_HDMI_HPD_UP not defined
002964.880: gpioman: gpioman_get_pin_num: pin LEDS_REMOTE_CONTROL not defined
002964.900: gpioman: gpioman_get_pin_num: pin LEDS_ARM_CONTROLLED not defined

ivanfdez, 2015-05-31:
When compiled kernel /proc/device-tree not exist

ivanfdez, 2015-05-31:
I used mkknlimg for convert zImage to image.img and now work!!!

does not understand the tool mkknlimg , I have not seen any  kernel compilation manual

sorry for my English , I really hard to express

thx

popcornmix, 2015-05-31:
Some info on mkknlimg here: https://github.com/raspberrypi/documentation/blob/master/configuration/device-tree.md

Ruffio, 2016-08-16:
@ivanfdez has your issue been resolved? If so, please close this issue. Thanks.

ivanfdez, 2016-08-19:
The problem was solved, did not follow the right steps

Thank you

closed 2016-08-19 by pelwell