raspberrypi/linux #991: TP-Link TL-WN725N issue

ric96, 2015-05-29:
TP-Link TL-WN725N does not work out of the box for raspberry pi, a simple workaround is to 
copy the firmware file (if you don't have it already) from https://github.com/lwfinger/rtl8188eu/blob/master/rtl8188eufw.bin to /lib/firmware/rtlwifi/
referencing #640 
it would be nice to have it pre-installed.
thank you

ric96, 2015-05-29:
and now i am getting kernel panics :p

XECDesign, 2015-05-29:
pretty sure this will be in jessie

ric96, 2015-05-29:
But we are not upgrading to it soon, are we?

XECDesign, 2015-05-29:
as soon as possible, but can't give any time frame.

Ruffio, 2016-08-16:
@ric96 has your issue been resolved? If so, please close this issue. Thanks.

ric96, 2016-08-16:
Work with the latest release of the OS and firmware.

closed 2016-08-16 by ric96