raspberrypi/linux #1003: spi-bcm2835 driver does something odd with chip selects?

jimbojr, 2015-06-03:
It seems to set SPI chip select pins as GPIO outputs rather than using the native peripheral chip selects.

pelwell, 2015-06-03:
@msperl Any comments?

pelwell, 2015-06-03:
A quick glance suggests that the sense of the test near the start of bcm2835_spi_setup is wrong:
```
	/*
	 * sanity checking the native-chipselects
	 */
	if (spi->mode & SPI_NO_CS)
		return 0;
	if (gpio_is_valid(spi->cs_gpio))  <=== Shouldn't this be !gpio_is_valid()?
		return 0;
	if (spi->chip_select > 1) {
		/* error in the case of native CS requested with CS > 1
		 * officially there is a CS2, but it is not documented
		 * which GPIO is connected with that...
...
	/* and calculate the real CS */
	spi->cs_gpio = chip->base + 8 - spi->chip_select;  <=== cs_gpio is an input to this function, not an output
```
My understanding is that the SPI framework initialises its cs_gpios array from the "cs-gpios" property, leaving absent entries invalid. When creating a device it picks the entry from this array indexed by the chip-select number, and uses this to set cs_gpio in the spi_device structure.

If spi->cs_gpio is valid then driver has to initialise a GPIO to an output and drive it appropriately. If this value is not valid then either the hardware is driving it directly (now that pinctrl has set the appropriate alt-function) or chip select is not being used ((spi->mode & SPI_NO_CS) != 0).

pelwell, 2015-06-03:
Ah, the commit that introduced those lines explains the reasoning:
```
    spi: bcm2835: transform native-cs to gpio-cs on first spi_setup
    
    Transforms the bcm-2835 native SPI-chip select to their gpio-cs equivalent.
    
    This allows for some support of some optimizations that are not
    possible due to HW-gliches on the CS line - especially filling
    the FIFO before enabling SPI interrupts (by writing to CS register)
    while the transfer is already in progress (See commit: e3a2be3030e2)
    
    This patch also works arround some issues in bcm2835-pinctrl which does not
    set the value when setting the GPIO as output - it just sets up output and
    (typically) leaves the GPIO as low.  When a fix for this is merged then this
    gpio_set_value can get removed from bcm2835_spi_setup.
    
    Signed-off-by: Martin Sperl <kernel@martin.sperl.org>
    Signed-off-by: Mark Brown <broonie@kernel.org>
```

msperl, 2015-06-03:
That is true and it is on purpose!
The reason ist that some optimization a are only working 100% reliable when not using native-CS but CS-gpio.
So the code switches the gpios from alt0 to out in case native-io is found and assigns them as cs_gpio to make that work transparently.
It also writes out a message to dmesg that it did switch gpios to output just to state that fact.
In principle you can also set CS-gpios in the device-tree for all CS and everything is fine again, but then the spi-bcm2708 driver will no longer work, as this expects alt0

msperl, 2015-06-03:
Note that reading again: 
The driver is not supposed to set pinmux for normal cases (which need to get set in device tree explicitly)- this only overrides the native-CS to switch to gpio-CS

crishoj, 2016-05-16:
Sorry to piggyback on this old thread, but is it correct that I can address e.g. 6 SPI devices connected in parallel if I set `SPI_NO_CS` and do the chip select myself using GPIO pins?

pelwell, 2016-05-16:
That's a question for the [Interfacing Forum](https://www.raspberrypi.org/forums/viewforum.php?f=44).

msperl, 2016-05-16:
actually you can configure the devie-tree now for you own custom number of CS.

There is no direct overlay for spi0, but the examples for spi1/spi2 also apply in principle:
* add the gpio to the list of cs gpios
* add the gpios to `cs-gpios = ... ` (you may need to set them!) - this controls the order/ids of the devices!

Ruffio, 2016-08-16:
@jimbojr has your issue been resolved? If so, please close this issue. Thanks.

closed 2016-08-17 by jimbojr