raspberrypi/linux #1004: bcm2708-i2s lacks mmap support

jimbojr, 2015-06-04:
If the driver supports mmap then ALSA will do sensible automatic rate and sample size conversion.

The problem (apart from it should support mmap as most drivers seem to) is it isn't obvious to those using I2S based sound cards why a lot of things don't work properly. I've now seen quite a few posts on the R-Pi forums and elsewhere. Just two examples: https://www.hifiberry.com/forums/topic/jackd/ and https://www.raspberrypi.org/forums/viewtopic.php?f=98&t=111982&p=769781#p769781

It would be great to add this if we can.

HiassofT, 2015-06-04:
MMAP support was present in the inital version of bcm2708-i2s but shortly afterwards removed in commit 62c05a0b5328d9376d39c9e74da10b8a2465c234.

It's not quite clear to me what the strange issues actually were/are. 

Ping @koalo, could you elaborate on this?

Ragnar Jensen provides a Wolfson/Cirrus card kernel build with MMAP support enabled and it seems to be working very well.
http://www.element14.com/community/message/149403/l/re-cirrus-logic-audio-card-working-on-the-raspberry-pi-2#149403

I just did a few tests with my Cirrus kernel build (based on top of rpi-4.0.y) with MMAP enabled and it worked fine, too. With MMAP aplay is able to play back 24bit WAVs which won't work without MMAP support.

This is the patch I used when testing:
```
--- a/sound/soc/bcm/bcm2708-i2s.c
+++ b/sound/soc/bcm/bcm2708-i2s.c
@@ -875,7 +875,9 @@ static const struct snd_soc_component_driver bcm2708_i2s_component = {
 };

 static const struct snd_pcm_hardware bcm2708_pcm_hardware = {
-       .info                   = SNDRV_PCM_INFO_INTERLEAVED |
+       .info                   = SNDRV_PCM_INFO_MMAP |
+                                 SNDRV_PCM_INFO_MMAP_VALID |
+                                 SNDRV_PCM_INFO_INTERLEAVED |
                                  SNDRV_PCM_INFO_JOINT_DUPLEX,
        .formats                = SNDRV_PCM_FMTBIT_S16_LE |
                                  SNDRV_PCM_FMTBIT_S24_LE |
```

I think we should do some more testing with the various cards (I only have Wolfson/Cirrus cards here), maybe MMAP works fine with the current kernel tree and it's just a matter of adding back the info flags.

pelwell, 2015-06-04:
If that's all it takes, I can add a custom DT property to enable MMAP support and expose it via a dtparam setting.

koalo, 2015-06-04:
I do not know how exactly MMAP is implemented, but it seems to rely on some kind of advanced memory management of the DMA engine or maybe there is still a bug in the driver itself or ALSA.
Anyway, under some constellations, playing back a file that require MMAP and afterwards recording in 24-bit sometimes result in having a loop of the previously played file in the record instead of the recorded audio signal or no signal at all.
Since I was not able to further debug this issue, I decided to disable MMAP, since don't having a feature is better to have a feature that results in unreliable behavior.
If you decide to enable it per default, it would be very advisable to test it with a wide range of different files with different formats, recording formats and hardware that does support 24-bit and hardware that does not.

pelwell, 2015-06-05:
I've pushed a commit that makes the MMAP support conditional on a DT property. After rebuilding and copying the kernel, modules and overlays to your SD card, add:
```
dtoverlay=i2s-mmap
```
to `config.txt`.

pelwell, 2015-06-08:
The latest firmware (https://github.com/Hexxeh/rpi-firmware/commit/9ad0e7528ce9d356e7c96fd7b2ed6d3103955d28) contains the updated module and overlay. Please try it and comment on your findings.

HiassofT, 2015-06-08:
Thanks a lot for the explanation and the commit!

@pelwell could you include the commit in the 4.0 tree as well?

I did several tests with my Cirrus card and the driver rebased
onto rpi-3.18.y and so far everything went well.

First I played a testfile, tried with different samplerates
and 16/24/32 bit. The (packed) 24bit wavs required MMAP support,
for the others I tried both without MMAP and with forced MMAP
(aplay -M).

Immediately afterwards I recorded an audio file. Tried with 44.1
and 48kHz, 16/24/32 bit and also without and with MMAP (arecord -M).

After recording I played back the recorded file, also sometimes
tried with aplay -M.

I did some 15 play-record-play runs and didn't see any of the
issues koalo wrote about.

The only issue I ran into was playback not starting together with a
```
bcm2708-dmaengine bcm2708-dmaengine: DMA transfer could not be terminated
```
entry in dmesg. But I think that's unrelated to MMAP support, I saw
this before without MMAP support too (not sure if it's a bug in
the Cirrus driver or somewhere else).

pelwell, 2015-06-08:
Pushed.

jimbojr, 2015-06-24:
I am still not sure this is 'proper' mmap support but I'll give it a whirl for some test files I have.

I wonder if @martinezjavier (who I believe very kindly added mmap support to the bcm2835 GPU driver) would mind looking over the code to say yes or no whether it looks like it supports mmap properly, or if we need to do more work here...

BBUK, 2015-08-03:
Thanks for this.  I want to use a Hifiberry Amp+ to run two independent mono streams for a multi room solution (one mapped to the left output and one to the right output).

The easiest route to do this is via the alsa dmix plugin which (I understand) needs MMAP.

The i2s-mmap overlay has enabled proper use of dmix and is confirmed working fine for my use case on a RPi B+ (running mpd).

Let me know if you would like me to assist by undertaking any further tests.

Thanks again

pelwell, 2015-08-10:
@BBUK Thanks for the feedback. Unless we reopen the original problem of why enabling MMAP support by default is problematic, I can't think of anything else for you to test.

iqaudio, 2015-08-17:
@BBUK Can you post details of dmix usage so we can replicate please? I'd like to check this on our boards too.

BBUK, 2015-08-23:
@iqaudio No magic really:

config.txt
> dtoverlay=i2s-mmap
> dtoverlay=hifiberry-amp

asound.conf (I blacklisted the onboard audio driver hence hw:0)
> pcm.onboard {
type hw
card 0
}

> pcm.!default {
    type plug
    slave.pcm "onboard_dmix"
}

> pcm.onboard_dmix {
   type dmix
   ipc_key 1024
   ipc_key_add_uid 1
   ipc_perm 0666
   slave {
       pcm "onboard"
       channels 2
    }
    bindings {
       0 0
       1 1
    }
}

> pcm.zone1 {
type plug
slave.pcm onboard_dmix
slave.channels 2
ttable {
0.0 0.5
1.0 0.5
0.1 0
1.1 0
}
}

> pcm.zone2 {
type plug
slave.pcm onboard_dmix
slave.channels 2
ttable {
0.0 0
1.0 0
0.1 0.5
1.1 0.5
}
}

I then set up two instances of mpd with different config files.  One outputting to zone1 and one to zone2.  So the mpd.conf for zone 1 output would say:
> ...
> audio_output {
        type            "alsa"
        name            "Zone1"
        device          "zone1"
        mixer_type      "software"
}
> ... 

HTH, BBUK

iqaudio, 2015-08-23:
Thanks, useful to see a working config.

Best regards,

Gordon

pelwell, 2015-08-23:
Thanks for sharing. Just for the record, with recent firmware (rpi-update, and the forthcoming new Raspbian image) there should be no need to blacklist the snd-bcm2835 driver - just make sure it isn't in /etc/modules and that you don't have dtparam=audio=on. Unlike in older firmware, the inbuilt audio is off unless asked for, but clean Raspbian installs will still include dtparam=audio=on by default.

iqaudio, 2015-08-23:
Just noticed that if dtparam=audio=on is commented out then alsamixer doesn't work (mplayer still plays audio)

Alsamixer error message: "Cannot open mixer: No such file or directory"

If dtparam=audio=on is listed in config.txt then alsamixer works and shows additional i2s audio cards.

Any clues?

G

pelwell, 2015-08-23:
What is in your ~/.asoundrc? A quick search suggests that it may have a reference to card 1 when you no longer have one. Try deleting/renaming it and see if that fixes it.

iqaudio, 2015-08-23:
Yep, that's it. Thanks Phil.

Gordon

mr-berndt, 2015-10-05:
Is there any way to enable this without DT-overlay? I run a small busybox with a self-compiled kernel and have only looked into DT-overlay far enough to figure that I'd better not mess with it. I (until now?!) don't need it and from what I understand it would require quite some changes in my system to work (I don't use udev, but mdev etc..).

I am trying to get jackd to run with an iqaudio dac. Any help would be great!

pelwell, 2015-10-05:
All you need to use Device Tree and overlays is:
1) Any firmware since January/February, but we would recommend the latest. This will include the base DTBs - bcm2708-rpi-b-plus.dtb, bcm2709-rpi-2-b.dtb etc. - and the overlays directory.
2) A kernel built with Device Tree support.
3) A small trailer on the kernel to tell the firmware that the firmware understands Device Tree.

I'd recommend you make this small change soon - we won't be supporting non-DT kernels forever.

But if that's too much effort, the second post in this thread shows you how to enable MMAP support unconditionally.

mr-berndt, 2015-10-05:
OK, I'll look into it, than. I was told DT and overlays /depend/ on udev. So I understand this is not the case, right?

One thing puzzles me, though: I can load the kernel-module for the onboard sound (snd-bcm2835) and happily use MMAP and therefor jackd with it. If MMAP depends on "activation" in bcm2708-i2s I wonder how this comes? Is the snd-bcm2835 module somehow able to activate MMAP in snd-bcm2835? If so, it might make sense for other drivers to do similarely and spare users the configuration trouble.

pelwell, 2015-10-05:
> OK, I'll look into it, than. I was told DT and overlays /depend/ on udev. So I understand this is not the case, right?

Yes - that's _not_ the case.

> One thing puzzles me, though: I can load the kernel-module for the onboard sound (snd-bcm2835) and happily use MMAP and therefor jackd with it. If MMAP depends on "activation" in bcm2708-i2s I [...]

This is about MMAP support in the ALSA drivers - not whether the whole kernel supports MMAP because they all do. Each of the drivers may or may not support MMAP - they are independent. This thread is about MMAP support in the I2S driver.

mr-berndt, 2015-10-05:
Tried the DT and overlay way and it works - Thx a lot!

About the drivers: I thought, all of the drivers we are talking about here, use the bcm2708-i2s driver and therefor it's MMAP feature. So is the snd-bcm2835 independant from it and comes with it's own MMAP-support? Anyway, I might have gotten it wrong but nevermind, I have MMAP working now and I don't use the onboard sound anyway.

pelwell, 2015-10-05:
Congratulations on your DT success - you've made the right move.

The i2s driver is only for external DACs and ADCs that are attached using the I2S digital audio bus. Apart from being an ALSA driver and running on the same chip there is no connection to the snd_bcm2835 driver.

agherzan, 2017-04-25:
@popcornmix Any idea why this overlay (i2s-mmap) is not available anymore in 4.9?

HiassofT, 2017-04-25:
> Any idea why this overlay (i2s-mmap) is not available anymore in 4.9?

bcm2835-i2s now supports MMAP by default so this overlay is no longer
needed.

agherzan, 2017-04-25:
Sounds good. Do you have any idea where is this enabled now in the source?

HiassofT, 2017-04-25:
> Sounds good. Do you have any idea where is this enabled now in the source?

sound/soc/soc-generic-dmaengine-pcm.c

All I2S devices using soc-generic-dmaengine implicitly have MMAP
capabilities.

see dmaengine_pcm_set_runtime_hwparams:
...
        hw.info = SNDRV_PCM_INFO_MMAP | SNDRV_PCM_INFO_MMAP_VALID |
                        SNDRV_PCM_INFO_INTERLEAVED;

agherzan, 2017-05-01:
Thanks @HiassofT . That makes sense.

closed 2017-05-17 by P33M