raspberrypi/linux #1012: PCM5122 driver uses non-standard control names - breaks apps

andrum99, 2015-06-08:
Recently there was a change to the PCM5122 driver that changed the mixer control names. The new control names are non-standard (per the documentation in the Linux kernel at https://www.kernel.org/doc/Documentation/sound/alsa/ControlNames.txt) and cause several applications problems, including mpd and Kodi. Applications are now unable to see the master volume control on PCM5122-based cards (for example. the HifiBerry DAC+). The master volume control is now named 'Digital Playback Volume', which is on the deprecated list. It should be renamed 'PCM Playback Volume'. Also, there are two controls that have the source name as 'Analogue'. The word 'analogue' does not appear anyway in the documentation for control names. ('Analog' does but only in reference to loopback controls). These two controls also need to be renamed to something compliant with the standard.

pelwell, 2015-06-08:
The commit in question (f646e040a619bcea31a6cab378ccaccb6f4cb659) is in the mainline kernel, approved by Mark Brown and Greg Kroah-Hartman. Although I agree (at first glance) that the change looks wrong, you need to raise the issue upstream.

closed 2015-06-08 by pelwell

popcornmix, 2015-06-08:
Ping @hmbedded for comments.

andrum99, 2015-06-08:
Bug #99631 opened upstream (https://bugzilla.kernel.org/show_bug.cgi?id=99631)

hmbedded, 2015-06-08:
The main volume control on PCM5122 was originally named 'Playback Digital Volume' and I seem to remember that mpd had problems with that name also (no experience with Kodi). It was changed by Mark Brown (alsa maintainer) in this commit (https://github.com/raspberrypi/linux/commit/1c6d36805fcbd9f84b6d9252b4d022653df8d1fa) so as to comply with ControlNames.txt according to the commit comments.

The addition of Analogue to a couple of controls was done by me in agreement with Mark Brown. The reason was to fix an issue that users of the PCM5122 were seeing when upgrading from the rpi-3.12 to rpi-3.18:
Since rpi-3.12 Mark Brown had added this commit (https://github.com/raspberrypi/linux/commit/5be2fc20b101b5138c4f54a584dc11790ef16598) which exposes a couple of analogue volume controls on the PCM5122. The one labelled 'Playback Volume' is in fact intended to set the full scale output voltage of the device and has just two levels available: 0db and -6dB. Unfortunately the mechanism that alsa uses to restore volume control settings at boot time has a default of -20dB for a control named 'Playback Volume' and so after an upgrade to rpi-3.18 users were finding that there output volume had halved as 'Playback Volume' had been forced to -6dB, rather than the devices default of 0dB.
In my opinion it made no sense to expose these controls to userland as the full scale output voltage is really a PCB design decision. However, apparently alsa standard practise is to expose everything to avoid kernel rebuilds etc. so Mark suggested adding a new 'Analogue' prefix to ensure that the controls were ignored by existing software. This seemed like a reasonable compromise so I went ahead and submitted a patch (https://github.com/raspberrypi/linux/commit/24dd64320e45b5d840aaee282f58c59e3f3333c4).