raspberrypi/linux #1013: Explanation or short description for the rpi-linux branches and the workflow

c0d3z3r0, 2015-06-08:
It would be great if there was an explanation or short description for the diffrerent branches in the wiki and the process for creating new branches ( not every single git command but a short workflow description ;-) ).

My main questions would be:
- What is the difference between rpi-3.18.y and rpi-3.18.y-rebase. Is there any beside the rebase?
- What is the workflow of creating new branches for new versions? Do you simply rebase the old version against mainline and then apply the patches to the new branch?
- What is rpi-patches?

Thanks in advance
Michael

popcornmix, 2015-06-09:
In general we support two kernel trees. Stable and latest.

Stable is the kernel tree used by raspbian and rpi-update. Currently that is 3.18.y.

Latest is the latest upstream kernel that is purely used by self-compilers or distributions that like to use latest kernel (e.g. OpeneELEC). We tend to use latest for more experimental features and future features (like switching to upstream drivers and more use of device tree). Currently latest if 4.0.y.

In general we merge in kernel bumps in stable tree (to make the history clear when identifying source associated with an official build), but we rebase the unstable trees (keeping Pi specific commits on top). 

3.18.y-rebase was used by OpenELEC for a while before they moved to 4.0.y tree and it was just a rebase of 3.18.y tree (our commits moved to end). It should possibly be deleted now for causing confusion.

rpi-patches was the very first tree (3.1.9 kernel) before we started including the kernel branch in the branch name.

c0d3z3r0, 2015-06-09:
Thanks a lot! That makes everything much more clear to me :-)

closed 2015-10-13 by c0d3z3r0