raspberrypi/linux #1015: kexec - vchiq: vchiq_initialise: videocore not initialized

binarylemon, 2015-06-09:
I try run same kernel with same rootfs via kexec.

First boot (boot after powerup) without any issue:
[    2.382497] vc-sm: Videocore shared memory driver
[    2.387217] [vc_sm_connected_init]: start
[    2.392388] [vc_sm_connected_init]: end - returning 0

but after kexec, videocore initialization failed:
```
$ kexec -l --append="$(cat /proc/cmdline)" /boot/Image--3.18.14+git39bde2cb339980677dbceb03336246b6de1d9095-r0-raspberrypi-20150608232906.bin 
$ kexec -e
```
[    2.390164] vc-sm: Videocore shared memory driver
[    2.394883] [vc_sm_connected_init]: start
[    2.399097] vchiq_get_state: g_state.remote->initialised != 1 (0)
[    2.405787] vchiq_get_state: g_state.remote->initialised != 1 (0)
[    2.412587] vchiq_get_state: g_state.remote->initialised != 1 (0)
[    2.419154] Indeed it is in host mode hprt0 = 00021501
[    2.428547] vchiq_get_state: g_state.remote->initialised != 1 (0)
[    2.436856] vchiq_get_state: g_state.remote->initialised != 1 (0)
[    2.443103] mmc0: host does not support reading read-only switch, assuming write-enable
[    2.458468] vchiq_get_state: g_state.remote->initialised != 1 (0)
[    2.465062] vchiq_get_state: g_state.remote->initialised != 1 (0)
[    2.473936] mmc0: new high speed SDHC card at address 0002
[    2.488697] mmcblk0: mmc0:0002 SD8GB 7.46 GiB
[    2.495095]  mmcblk0: p1 p2
[    2.498522] vchiq_get_state: g_state.remote->initialised != 1 (0)
[    2.505117] vchiq_get_state: g_state.remote->initialised != 1 (0)
[    2.528573] vchiq_get_state: g_state.remote->initialised != 1 (0)
[    2.535183] vchiq: vchiq_initialise: videocore not initialized
[    2.535183]
[    2.557977] [vc_sm_connected_init]: failed to initialise VCHI instance (ret=-1)
[    2.565294] [vc_sm_connected_init]: end - returning -5

RPI boot to shell, but any part which depend on vchiq does not work.
For example audio produces this log:
[  114.321115] vchiq_get_state: g_state.remote->initialised != 1 (0)
[  114.328635] vchiq_get_state: g_state.remote->initialised != 1 (0)
[  114.335825] vchiq_get_state: g_state.remote->initialised != 1 (0)
[  114.343866] vchiq_get_state: g_state.remote->initialised != 1 (0)
[  114.351122] vchiq_get_state: g_state.remote->initialised != 1 (0)
[  114.358673] vchiq_get_state: g_state.remote->initialised != 1 (0)
[  114.365845] vchiq_get_state: g_state.remote->initialised != 1 (0)
[  114.373797] vchiq: vchiq_initialise: videocore not initialized
[  114.373797]
[  114.381277] bcm2835_audio_open_connection:386 bcm2835_audio_open_connection: failed to initialise VCHI instance (ret=-1)

tested on RPI B+, Linux kernel version 3.18.14, 3.12.26, default configuration.

Any hint/solution?

pelwell, 2015-06-10:
I haven't looked into the kexec mechanism, but you are effectively rebooting the ARM without the knowledge of the VideoCore processor - remember that in BCM2835 the ARM is effectively a peripheral of the VPU. The VCHIQ driver on the VPU doesn't support that use case, so I'm not surprised it is causing problems.

XECDesign, 2015-06-10:
Duplicate of #27 ?

binarylemon, 2015-06-10:
@XECDesign yes, we can say, it is duplicate :(

Ruffio, 2016-08-16:
@binarylemon has your issue been resolved? If so, please close this issue. Thanks.

popcornmix, 2016-08-17:
Well it's a duplicate, so close in favour of the older one.

closed 2016-08-17 by popcornmix