raspberrypi/linux #1016: Can't cross-compile 4.0.y

voltagex, 2015-06-10:
Build gets up to making modules and then errors out:
```
drivers/built-in.o: In function `bcm2708_ioctl':
devfreq.c:(.text+0x20dc0): undefined reference to `bcm_mailbox_property'
drivers/built-in.o: In function `bcm2708_fb_blank':
devfreq.c:(.text+0x20e60): undefined reference to `bcm_mailbox_property'
drivers/built-in.o: In function `bcm2708_fb_set_par':
devfreq.c:(.text+0x21610): undefined reference to `bcm_mailbox_write'
devfreq.c:(.text+0x2161c): undefined reference to `bcm_mailbox_read'
drivers/built-in.o: In function `vchiq_platform_init':
devfreq.c:(.text+0x8b758): undefined reference to `bcm_mailbox_write'
drivers/built-in.o: In function `bcm2835_get_temp_or_max.isra.0':
devfreq.c:(.text+0x11f35c): undefined reference to `bcm_mailbox_property'
drivers/built-in.o: In function `bcm2835_cpufreq_get_clock':
devfreq.c:(.text+0x127ea4): undefined reference to `bcm_mailbox_property'
drivers/built-in.o: In function `bcm2835_cpufreq_driver_target_index':
devfreq.c:(.text+0x128004): undefined reference to `bcm_mailbox_property'
make: *** [vmlinux] Error 1
```

voltagex, 2015-06-10:
rpi-3.19.y compiles fine.

pelwell, 2015-06-10:
I cross-compile 4.0.y every day, so the cross-compilation itself isn't the problem.

What configuration are you using?

Start with:
```
make ARCH=arm CROSS_COMPILE=... clean
make ARCH=arm CROSS_COMPILE=... bcmrpi_defconfig (or bcm2709_defconfig) zImage modules dtbs
```
which should work.

voltagex, 2015-06-10:
Yep, that worked - must be the config I was using.

closed 2015-06-10 by voltagex

popcornmix, 2015-06-10:
Probably missing:
CONFIG_MAILBOX=y
CONFIG_BCM2708_MBOX=y

N8Fear, 2015-06-10:
In that case: shouldn't the drivers that need that options depend on CONFIG_MAILBOX=y and CONFIG_BCM2708_MBOX=y if it results in a broken build otherwise?

voltagex, 2015-06-11:
In the interests of completeness, the config I used was https://gist.github.com/voltagex/3e5c3c9e8681abca8d82 from https://www.collabora.com/about-us/blog/2015/02/03/debian-jessie-on-raspberry-pi-2/

popcornmix, 2015-06-11:
I've added some depends Kconfig options so your .config now builds without error. It will have some of the modules disabled due to the missing CONFIG_MBOX though.