raspberrypi/linux #1018: 2 MCP2515 dont work at the same time

petit-miner, 2015-06-10:
Two MCP2515 Can Bus Controller IC dont work at the same time.
The PI has two SPI CE Pins so this shouldnt a problem.

pelwell, 2015-06-15:
@msperl Any thoughts?

msperl, 2015-06-15:
I have a setup with 2 mcp2515 and it works fine for me (together with mmc-spi(with patch), fbtft and enc28j26)
I guess it may depend on the dt-overlay...

One thing that I had seen originally was: does it get detected properly after:
rmmod mcp251x;modprobe mcp251x

msperl, 2015-06-15:
There is an issue upstream that could be the root-cause...
As a workaround it might be sufficient to add:
 cs-gpios=<&gpio 8 1> <&gpio 7 1>;
(Or similar) to the spi device in dt to set up CS-gpio immediately and not when the device is probed first...
(That would also remove those warnings)

pelwell, 2015-06-15:
@petit-miner Which kernel are you using? The 3.18 tree is still using the bcm2708-spi driver, while the 4.0 tree has switched to the upstreamed bcm2835-spi driver.

petit-miner, 2015-06-21:
@msperl It would be very nice if you make a short tutorial for me and tell me which kernel version you are using.
That would help me a lot.

msperl, 2015-06-22:
I am running my own build of the kernel based on the 4.0 (otherwise i can not create patches), so you will have to do that as well if you want to use the latest driver.

Note that the foundation seems to be moving to 4.0 already, as their kernel default branch is 4.0.y now...

Besides that you will also need to create a new overlay for can1 by copying:
https://github.com/raspberrypi/linux/blob/rpi-4.0.y/arch/arm/boot/dts/overlays/mcp2515-can0-overlay.dts

and replacing:
* can0 with can1
* spidev0 with spidev1
* default interrupt lines
* reg = <0> with reg = <1>

then compile it with dtc (say as mcp2515-can1-overlay) and then you can load it as well via /boot/config.txt...

pelwell, 2015-06-22:
@petit-miner If you do this and can verify that it works (not that I'm doubting you, @msperl) then we can add the new overlay to the standard distributions.

petit-miner, 2015-06-22:
I am a little bit confused, there are 2 Kernels one for the PI B+ and the PI 2.
Where is the Kernel sourcecode for the PI 2?
But the self build new Kernel isn't necessary?
And how I can compile the dts file?
or @msperl  could you upload the .dtb to dropbox or something else that would be very nice.

pelwell, 2015-06-22:
The two kernels are built from the same source tree, using the same tools, but with a few differences in the configurations. For a Pi 1 we build using bcmrpi_defconfig, but for Pi 2 we use bcm2709_defconfig. See [here](https://www.raspberrypi.org/documentation/linux/kernel/building.md) for details.
You can compile the dts file either by adding it into the kernel source tree and using the included version of dtc, or by running:
```
sudo apt-get update
sudo apt-get install device-tree-compiler
```
Once this is done you just need to:
```
dtc -@ -I dts -O dtb -o mcp2515-can1-overlay.dtb mcp2515-can1-overlay.dts
```
You can find more documentation about Device Tree on the Pi [here](www.raspberrypi.org/documentation/configuration/device-tree.md).

petit-miner, 2015-06-22:
Thank you for the Information :+1: 
I have to replace every can0 with can1 in the file and every spidev0?
What I have to change at the default interrupt lines?

msperl, 2015-06-22:
well with the interrupt line you are using for the second controller - maybe use 24 or 26 as the second interrupt line... (you can override it anyway in config.txt via additional parameters)

petit-miner, 2015-06-22:
I think I made a mistake while editing the file,
it would be easier when you upload the file to dropbox or something else or post the content here.
I get this error:
001046.751: dterror: can't find symbol 'spi1'
001046.768: dtoverlay_merge_overlay('mcp2515-can1-overlay') failed - err 1

msperl, 2015-06-22:
@pelwell: one point maybe: Is it possible to include the SAME overlay twice (but with distinct parameters)?
If so then we could modify the overlay to allow to add additional parameters.

example usage could be:
```
dtoverlay=mcp2515-canX-overlay,cs=0,oscillator=20000000,interrupt=25
dtoverlay=mcp2515-canX-overlay,cs=1,oscillator=16000000,interrupt=24
```
Which would create 2 can interfaces: can0 and can1 based on chipselect 0 and 1.

The downside obviously would be that you would need to add some code to the firmware parser to modify also the identifiers by some scheme (spidev0 translate to spidev1, can0 to can1, can0_osc to can1_osc, can0_pins to can1_pins...).

petit-miner, 2015-06-22:
That would be a great idea, but I am not a programmer. :/

pelwell, 2015-06-22:
@petit-miner There is no spi1 - that should still say spi0, but the 0s in  reg = <0> and mcp2515@0 should be changed to 1s.

@msperl There is nothing stopping an overlay being loaded twice, but as you have worked out the names of the nodes and properties can not be changed based on parameters. I understand that this can result in a proliferation of overlays, but I don't want to spend time on adding a slightly esoteric feature now when I could be working on dynamic Device Tree support.

msperl, 2015-06-22:
I am fine with that - I just wanted to let you know that this could also be an option...

pelwell, 2015-06-22:
I understand, and suggestions are welcome. It sounds like something I could return to at a later date.

msperl, 2015-06-22:
Side-note: Actually there is also spi1 and spi2 - the issue is that the foundation kernel does not have a driver yet and a patch to upstream a working driver has just been posted today...

pelwell, 2015-06-22:
I'll look forward to a PR...

petit-miner, 2015-06-25:
I am not at home, I can't test it out sorry.

petit-miner, 2015-07-01:
It works! :+1:

pelwell, 2015-07-10:
I've pushed your patch to rpi-4.0.y and rpi-4.1.y. Thank you for your efforts.

petit-miner, 2015-07-10:
Ok, thanks for your help :+1:

closed 2015-07-10 by petit-miner