raspberrypi/linux #1019: Oops 80f/17 Unable to handle kernel paging request at virtual address ffffffec

schuhmam, 2015-06-10:
Today I received my Raspberry PI 2 Model B. To be sure to have a proper power supply I bought a kit containing a transparent case and a 2A output power supply and some cooling.
The microSDHC card is a Transcend 16GB class 10 (TS16GUSDHC10U1)

The device is supposed to work as router to replace a virtual machine running Ubuntu on Microsoft Hyper-V (so server can be turned off over night). Anyways. Everything is working but often the system stops reacting. I only see the attached message on the display.

The last time the error appears was after I tried to configure the perl CPAN with perl -MCPAN -e shell. After pressing twice enter (2 questions with default answer 'yes') this error occurs. After a reboot I retried and at the same spot it happened again. Don't get me wrong this happened before - the same error. I am not 100% sure but I think the mentioned virtual address is always the same.

[image]

Sorry for the light bulb but otherwise my Windows Phone is too stupid to make a proper autofocus.

1. Unable to handle kernel paging request at virtual address 7f1dd14c
pgd = 80004000
[7f1dd14c] *pgd=393ad811, *pte=3611a4df, *ppte=3611a65e
Internal error: Oops: 80f [#1] PREEMPT SMP ARM
Modules linked in: [...]
[...]
Process kworker/0:1 (pid: 34)

I tried to raise this value (sorry don't remember the name) in /etc/sysctl.conf to 16384 but it did not help. While googleing I turned off the VLANs in the switch for the PI so that I can use the old virtual machine for internet. During this process the PI run well. Very strange...
This routing, IPv6 and so on is very basic stuff. I can't imagine that this cause so many problems. The memory consumption was about 68 MiB.
Any ideas?

The distribution is a Ubuntu with Kernel 3.18-20

The original source is my recorded video:
https://drive.google.com/open?id=0B0R5aKIdB-XzQ2FIdUhVeDN4ZmM&authuser=0

schuhmam, 2015-06-10:
Additional info:
 - The PI: http://www.amazon.de/dp/B00T3BX0L8
 - The microSDHC: http://www.amazon.de/dp/B00BLHWYWS
 - The power supply: http://www.amazon.de/dp/B00UCSO9G6

popcornmix, 2015-06-11:
I assume you've seen no warning squares in top right of display? (for under-voltage or over-temperature)?

Do you get the same failure with raspbian?
Ubuntu is not something tested or directly supported by us so it's hard to offer advice.
If the issue doesn't occur with raspbian then it may suggest a problem with Ubuntu's kernel configuration or other setup.

schuhmam, 2015-06-11:
The warning square? Well at least I didn't know that they exist. So I think I should have recognized them.

But you are right. Ubuntu is nothing official with the Raspberry. I knew Ubuntu already so I wanted to give it a try first. I already started configuring the system and using Raspian now.

Now I am finished with setting up the system. I will have a look at the stability. Currently it runs fine. No crash so fare. Hope that will stay this way because I have absolutely no idea were the error is hidden.
Thank you so far.

schuhmam, 2015-06-11:
Looking good so far. I will come back if any problems show up.
Thanks.

closed 2015-06-11 by schuhmam