raspberrypi/linux #1021: ARCH_BCM2835 dwc_otg FIQ hack

notro, 2015-06-11:
Looks as if I might have FIQ working on ARCH_BCM2835.
Patch including intrumentation: https://gist.github.com/notro/3c345f09f29a3a8fed62
Overlay so I can switch between dw2 and dwc_otg: https://gist.github.com/notro/3f2be7402629b25c00e4
Add config: ```CONFIG_USB_DWCOTG=y```

I have hardcoded the fiq interrupt number in the ARCH_BCM2835 case, I need to look more into how I can map to the right number from d->hwirq.
I implemented bcm2835_enable_fiq() on both arch's when testing this.

ARCH_BCM2835
```
[    1.870597] dwc_otg 20980000.usb: Calling disable_global_interrupts

[    1.878693] armctrl_xlate(2 0) => 64
[    1.884066] platform_get_irq('20980000.usb', num=0) DT => 56
[    1.891557] dwc_otg 20980000.usb: Calling request_irq(56)
[    1.898711] genirq: request_threaded_irq(irq=56, 'dwc_otg')

[    1.906117] dwc_otg 20980000.usb: Calling dwc_otg_core_init
[    2.113639] Using Buffer DMA mode
[    2.118704] Periodic Transfer Interrupt Enhancement - disabled
[    2.126357] Multiprocessor Interrupt Enhancement - disabled
[    2.133775] OTG VER PARAM: 0, OTG VER FLAG: 0
[    2.139929] dwc_otg 20980000.usb: Calling pcd_init
[    2.146449] Dedicated Tx FIFOs mode

[    2.152046] armctrl_xlate(2 0) => 64
[    2.157373] platform_get_irq('20980000.usb', num=0) DT => 56
[    2.164855] genirq: request_threaded_irq(irq=56, 'dwc_otg_pcd')

[    2.172635] dwc_otg 20980000.usb: Calling hcd_init
[    2.179361] WARN::dwc_otg_hcd_init:1047: FIQ DMA bounce buffers: virt = 0xdb854000 dma = 0x1b854000 len=9024
[    2.192681] FIQ FSM acceleration enabled for :
[    2.192681] Non-periodic Split Transactions
[    2.192681] Periodic Split Transactions
[    2.192681] High-Speed Isochronous Endpoints
[    2.216474] dwc_otg: Microframe scheduler enabled
[    2.223162] WARN::hcd_init_fiq:414: FIQ on core 0 at 0xc041b65c
[    2.230882] WARN::hcd_init_fiq:415: FIQ ASM at 0xc041b8fc length 36
[    2.238895] WARN::hcd_init_fiq:440: MPHI regs_base at 0xf0006000

[    2.246646] armctrl_xlate(1 9) => 41
[    2.251898] platform_get_irq('20980000.usb', num=1) DT => 33
[    2.259253] hcd_init_fiq: pdev=dad29400, irq=33
[    2.265462] bcm2835_enable_fiq(33): d->hwirq=41
[    2.271640] bcm2835_enable_fiq: addr=f000b20c

[    2.277741] armctrl_xlate(2 0) => 64
[    2.283028] platform_get_irq('20980000.usb', num=0) DT => 56
[    2.290455] dwc_otg 20980000.usb: DWC OTG Controller
[    2.297177] dwc_otg 20980000.usb: new USB bus registered, assigned bus number 1
[    2.306298] genirq: request_threaded_irq(irq=56, 'dwc_otg_hcd:usb1')
[    2.314494] dwc_otg 20980000.usb: irq 56, io mem 0x00000000
[    2.321880] Init: Port Power? op_state=1

```

ARCH_BCM2708
```
[    1.990259] dwc_otg 20980000.usb: Calling disable_global_interrupts

[    1.998287] armctrl_xlate(2 0) => 32 32
[    2.003739] platform_get_irq('20980000.usb', num=0) DT => 32
[    2.011062] dwc_otg 20980000.usb: Calling request_irq(32)
[    2.018122] genirq: request_threaded_irq(irq=32, 'dwc_otg')

[    2.025336] dwc_otg 20980000.usb: Calling dwc_otg_core_init
[    2.232389] Using Buffer DMA mode
[    2.237302] Periodic Transfer Interrupt Enhancement - disabled
[    2.244829] Multiprocessor Interrupt Enhancement - disabled
[    2.252075] OTG VER PARAM: 0, OTG VER FLAG: 0
[    2.258071] dwc_otg 20980000.usb: Calling pcd_init
[    2.264450] Dedicated Tx FIFOs mode

[    2.269866] armctrl_xlate(2 0) => 32 32
[    2.275315] platform_get_irq('20980000.usb', num=0) DT => 32
[    2.282635] genirq: request_threaded_irq(irq=32, 'dwc_otg_pcd')

[    2.290253] dwc_otg 20980000.usb: Calling hcd_init
[    2.296762] WARN::dwc_otg_hcd_init:1047: FIQ DMA bounce buffers: virt = 0xdbc14000 dma = 0x5bc14000 len=9024
[    2.309774] FIQ FSM acceleration enabled for :
[    2.309774] Non-periodic Split Transactions
[    2.309774] Periodic Split Transactions
[    2.309774] High-Speed Isochronous Endpoints
[    2.332830] dwc_otg: Microframe scheduler enabled
[    2.339345] WARN::hcd_init_fiq:414: FIQ on core 0 at 0xc040f0c0
[    2.346904] WARN::hcd_init_fiq:415: FIQ ASM at 0xc040f398 length 36
[    2.354754] WARN::hcd_init_fiq:440: MPHI regs_base at 0xdc896000

[    2.362361] armctrl_xlate(1 9) => 9 75
[    2.367651] platform_get_irq('20980000.usb', num=1) DT => 75
[    2.374855] hcd_init_fiq: pdev=dad25000, irq=75
[    2.380916] bcm2835_enable_fiq(75): data=9
[    2.386512] bcm2835_enable_fiq: addr=f200b20c

[    2.392466] armctrl_xlate(2 0) => 32 32
[    2.397874] platform_get_irq('20980000.usb', num=0) DT => 32
[    2.405095] dwc_otg 20980000.usb: DWC OTG Controller
[    2.411698] dwc_otg 20980000.usb: new USB bus registered, assigned bus number 1
[    2.420685] genirq: request_threaded_irq(irq=32, 'dwc_otg_hcd:usb1')
[    2.428729] dwc_otg 20980000.usb: irq 32, io mem 0x00000000
[    2.435898] Init: Port Power? op_state=1
```

notro, 2015-06-11:
I found this which gave me the hwirq -> fiq translation: https://github.com/rickytarr/linux/blob/topic/interrupt/drivers/irqchip/irq-bcm2835.c

```c
void bcm2835_enable_fiq(unsigned int irq)
{
  struct irq_data *d = irq_get_irq_data(irq);
  unsigned int fiq;

  if (!d)
    return;

  /*
   * The hwirq numbering used in this driver is:
   *   BASE (0-7) GPU1 (32-63) GPU2 (64-95).
   * This differ from the one used in the FIQ register:
   *   GPU1 (0-31) GPU2 (32-63) BASE (64-71)
   */
  if (d->hwirq >= 32)
    fiq = d->hwirq - 32;
  else
    fiq = d->hwirq + 64;

  printk("%s(%u): d->hwirq=%lu, fiq=%u\n", __func__, irq, d->hwirq, fiq);
  writel_relaxed(0x80 | fiq, intc.base + REG_FIQ_CONTROL);
}
```

log
```
[    2.246958] armctrl_xlate(1 9) => 41
[    2.252210] platform_get_irq('20980000.usb', num=1) DT => 33
[    2.265762] bcm2835_enable_fiq(33): d->hwirq=41, fiq=9
```

I see three ways to implement this:
- Use bcm2835_enable_fiq() as above
- Add a new hardware interrupt number and irq for all FIQ interrupts. Not sure how this should be specified in DT.
- Extend the irq subsystem with a flag that indicates that we want fiq not normal interrupt

notro, 2015-06-11:
I have a more proper solution now: https://gist.github.com/notro/85c9f45e8822da3c6f6c
It would be nice if someone could verify that this is indeed working as it should.

P33M, 2015-06-12:
My preference would be to extend irqchip-bcm2835 (and forthcoming for bcm2836, which has additional routing on top for IRQ/FIQ affinity) such that we start treating some of the senior bits of the hwirq number as a bitfield (i.e. enable_irq(FIQ | CORE 2 | USB_Interrupt)).

There's an eventual goal that we may want to run more than one FIQ handler at a time, each on a different core handling a specific real-time job - which is possible to accomplish with 2836. That's a story for a different day, though.

pelwell, 2015-06-12:
That sounds good to me.

P33M, 2015-06-12:
This gets a bit more complicated when you factor in the kernel mechanisms for enabling/disabling FIQs:
http://lxr.free-electrons.com/source/arch/arm/kernel/fiq.c

Functioning primarily as a resource tracking API, it's not multicore-aware. FIQs also share a common interrupt vector across cores, so if we went with multicore-capable FIQ handlers we'd have to start poking the MPIDR register and have per-cpu FIQ mode stacks and this all suddenly becomes a bit more complex.

notro, 2015-06-12:
Yes, currently only one FIQ is supported at a time.
Unless there are any objections, I'll polish up the irq part and send it upstream.

P33M, 2015-06-12:
The FIQ selector doesn't use the basic pending bits. The table in the datasheet for FIQ selection (BCM2835 ARM peripherals) selects FIQs as the following:

| *Index* | *Source * |
| 0-63 | GPU interrupts (from GPU IRQ table) |
| 64 | ARM AP804 timer interrupt |
| 65 | ARM Mailbox interrupt |
| 66 | ARM Doorbell 0 interrupt |
| 67 | ARM Doorbell 1 Interrupt |
| 68 | GPU0 halted interrupt |
| 69 | GPU1 halted interrupt |
| 70 | Illegal access type-1 interrupt |
| 71 | Illegal access type-0 interrupt |
| 72-127| Do not use |

notro, 2015-06-12:
> The FIQ selector doesn't use the basic pending bits.

I don't understand, is this an issue with my proposal?

P33M, 2015-06-12:
OK I see what the un-translate is doing now - I suppose interrupt number entries in the DT need to be correct anyway so there's no real need to do range checking when you write to the FIQ register.

notro, 2015-06-12:
armctrl_xlate() does range checking on the DT interrupt numbers.

notro, 2015-06-13:
I will make a PR when the patch is accepted: [[PATCH] irqchip: bcm2835: Add FIQ support](http://lists.infradead.org/pipermail/linux-rpi-kernel/2015-June/001910.html)

closed 2015-06-13 by notro