raspberrypi/linux #1022: WM8783 support?

sandyjmacdonald, 2015-06-11:
Anyone have any tips on getting a WM8783 ADC set up for audio input through I2S?

P33M, 2015-06-12:
You're better off getting advice in the forums.

Closing as this isn't an issue with the Pi Linux kernel.

closed 2015-06-12 by P33M

GTrannoy, 2015-06-12:
Hi,
I've just done the job for WM8750 and you can look what I changed in my repository.

sandyjmacdonald, 2015-06-12:
Cheers! I presume it's just a matter of repeating this, but for the WM8783, and then recompiling the kernel and then it should work for capturing audio with Alsa?

P33M, 2015-06-12:
The workflow for getting support this device (i.e. a device driver) is the following

- Write (or modify) the driver to support your device locally, on your hardware (whatever platform - doesn't have to be Raspberry Pi)
- Submit the patch adding/modifying the driver upstream to the respective kernel maintainer (there's plenty of kernel docs on this subject)
- Ping us once it's been through upstream review (which checks it conforms to kernel coding standards, uses the right subsystems and APIs)
- Submit a PR that simply enables the module(s) in our default driver config.

P33M, 2015-06-12:
Just a note that ASoC drivers are a bit more complicated - these drivers in effect specify a connection arrangement tied to a particular piece of hardware, independent of the codec. This hardware is usually Pi-specific, so we take out-of-tree ASoC drivers (along with the DT overlay to go with them).

GTrannoy, 2015-06-12:
> Cheers! I presume it's just a matter of repeating this, but for the WM8783, and then recompiling the 
> kernel and then it should work for capturing audio with Alsa?

Almost. But I see the driver WM8783.c and WM8783.h in '/sound/soc/codecs' need to be written...
Probably you could be inspired by similar other codecs.
Next:
* Check DT for correctly setting i2s, i2c, ... and your config.txt in '/boot'
* Make Device Tree Overlay for your WM8783
* Make modification for compiling kernel (like I have done)

And with luck and effort, it could work ! ;-)

Just for information: I waste time with bad configuration in Alsa for my specific codec and I believed it was a problem of compilation, overlay, etc... Maybe it can help you.