raspberrypi/linux #1023: bcm2708-i2s does not support Device Tree for GPIO mapping

jimbojr, 2015-06-11:
The bcm2708_i2s_setup_gpio() function in bcm2708-i2s.c needs to be updated to use Device Tree for GPIO location (it currently just sets up fixed GPIOs on bank 0).

jimbojr, 2015-06-11:
Is the fix to simply wrap all of the fixed GPIO setup stuff in bcm2708_i2s_setup_gpio() in a:
if (!pdev->dev.of_node) { <fixed pinsetup code here> }
?

pelwell, 2015-06-11:
Yes, I think so. I have that change sitting in a stash already, it just needs testing.

jimbojr, 2015-06-11:
OK I was going to do a quick change and test but there is no pdev node passed to bcm2708_i2s_setup_gpio();
In fact, shouldn't that GPIO setup be in the probe function?
I'll happily test a patch if you've gont one Phil. (and, aren't you supposed to be on holiday :)

Ruffio, 2016-08-16:
@jimbojr has your issue been resolved? If so, please close this issue. Thanks.

closed 2016-08-17 by jimbojr