raspberrypi/linux #1024: Permission bits for cpuinfo_cur_freq is inconsistent compared with cpu_max_freq & cpu_min_freq

DougieLawson, 2015-06-14:
Why does  ```/sys/devices/system/cpu/cpu*/cpufreq/cpuinfo_cur_freq``` have restrictive permissions (0400) when both ```./cpuinfo_max_freq``` & ```./cpuinfo_min_freq``` are both group and world readable (0444)?

Code in ../drivers/cpufreq/cpufreq.c
```
cpufreq_freq_attr_ro_perm(cpuinfo_cur_freq, 0400);
cpufreq_freq_attr_ro(cpuinfo_min_freq);
cpufreq_freq_attr_ro(cpuinfo_max_freq);
```

popcornmix, 2015-06-14:
This would be better asked upstream, and this is not code we have changed. E.g. on x86 Ubuntu:
```
dc4@dc4-XPS13-9333:~/xbmc_holder/xbmc$ ls -l  /sys/devices/system/cpu/cpu0/cpufreq
total 0
-r--r--r-- 1 root root 4096 Jun 14 13:50 affected_cpus
-r-------- 1 root root 4096 Jun 14 13:50 cpuinfo_cur_freq
-r--r--r-- 1 root root 4096 Jun 14 13:50 cpuinfo_max_freq
-r--r--r-- 1 root root 4096 Jun 14 13:50 cpuinfo_min_freq
-r--r--r-- 1 root root 4096 Jun 14 13:50 cpuinfo_transition_latency
-r--r--r-- 1 root root 4096 Jun 14 13:50 related_cpus
-r--r--r-- 1 root root 4096 Jun 14 13:50 scaling_available_governors
-r--r--r-- 1 root root 4096 Jun 14 13:50 scaling_cur_freq
-r--r--r-- 1 root root 4096 Jun 14 13:50 scaling_driver
-rw-r--r-- 1 root root 4096 Jun  9 18:28 scaling_governor
-rw-r--r-- 1 root root 4096 Jun 14 13:50 scaling_max_freq
-rw-r--r-- 1 root root 4096 Jun 14 13:50 scaling_min_freq
-rw-r--r-- 1 root root 4096 Jun 14 13:50 scaling_setspeed
```
At a guess it's because scaling_cur_freq is the entry you should be looking at.

popcornmix, 2015-06-23:
Closing as behaviour is as expected.

closed 2015-06-23 by popcornmix