raspberrypi/linux #1028: I2C on HDMI

exander77, 2015-06-22:
This is more of a question, than issue.

How to use I2C on HDMI port, I have seen some patches for old kernel, but it should be enabled in kernel, shouldn't be? I can't read or write edid of a display without it.

Patch for 3.10: https://www.raspberrypi.org/forums/viewtopic.php?t=44445&p=452512

popcornmix, 2015-06-22:
I'll quote jdb from the linked thread:

>Yes, but the GPU considers itself to be the sole user of the BSC2 peripheral. Poking the i2c from the ARM CPU (via linux driver) will likely work, but beware if the GPU tries to access the same peripheral then bad things will happen.
As long as you stay away from hotplugging HDMI, or tvservice commands (or anything likely to use tvservice) then the GPU is unlikely cause any traffic on BSC2.

Ping @pelwell. Is this something that can be exposed (with suitable warnings) through device tree?

pelwell, 2015-06-22:
It looks fairly trivial to do. Using DT, Florian/koalo's patch reduces to a tweak to the I2C driver allowing interface number 2 to be used, and (at least initially) an overlay to configure that interface.

Can you think of a warning mechanism? I can make it conditional on a suitably named DT parameter:
```
dtoverlay=i2c2,i_know_what_im_doing=yes
```
perhaps?

popcornmix, 2015-06-22:
I was just imagining a warning in the documentation, but don't mind a "i_know_what_im_doing" type parameter.

We could also only allow it if ``hdmi_ignore_edid=1`` or ``hdmi_edid_file=1``, which I believe means GPU will never access that I2C bus.

Exposing this to linux could be useful. There are tools for rewriting a display's edid, which as incorrect edid's are so common is a useful feature. Obviously advanced users only...

exander77, 2015-06-22:
That seems reasonable to me, it is better than patching and recompiling kernel all the time.

pelwell, 2015-06-22:
OK - watch this space.

pelwell, 2015-06-22:
If you are able to build your own kernel then can you please try the commits in this gist: https://gist.github.com/pelwell/be227c51be0c97dedabe
Only the second patch mentions i2c2, but it has a dependency on the first.

I hope the comments are self-explanatory.

exander77, 2015-06-22:
On what kernel version/commit should I apply it?

pelwell, 2015-06-22:
Top-of-tree rpi-4.0.y (4.0.5) - currently a2ac0d1efb6edb99b0a65f5e656ccdc6ef202eba .

pelwell, 2015-06-22:
I've updated the second patch in the gist - I missed a few DT nodes - but with the new version I've been able to test it to the point that i2c2 comes up and I can run `sudo i2cdetect -y 2` and find nothing.

pelwell, 2015-06-22:
And with an HDMI monitor plugged in I get:
```
pi@raspberrypi:~$ sudo i2cdetect -y 2
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- 37 -- -- 3a -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: 50 -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

exander77, 2015-06-22:
I am compiling kernel now, will repatch a try it. I will report soon.

exander77, 2015-06-22:
Strange, id does not boot.

Do I have to do something special to run 4.* kernel? I used kernel 3.* before.

I built and installed it this way:
```
make -j9 ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf-
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- INSTALL_MOD_PATH=/media/mmcblk0p2-mmc-SL08G_0x00a88086/ modules
sudo make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- INSTALL_MOD_PATH=/media/mmcblk0p2-mmc-SL08G_0x00a88086/ modules_install
sudo cp arch/arm/boot/zImage /media/boot/kernel7.img
```
I am using Raspberry Pi 2 B.

pelwell, 2015-06-22:
From https://www.raspberrypi.org/documentation/linux/kernel/building.md:
```
$ sudo scripts/mkknlimg arch/arm/boot/zImage mnt/fat32/$KERNEL.img
$ sudo cp arch/arm/boot/dts/*.dtb mnt/fat32/
$ sudo cp arch/arm/boot/dts/overlays/*.dtb* mnt/fat32/overlays/
```
which in your case would turn into:
```
$ sudo scripts/mkknlimg arch/arm/boot/zImage /media/boot/kernel7.img
$ sudo cp arch/arm/boot/dts/*.dtb /media/boot/
$ sudo cp arch/arm/boot/dts/overlays/*.dtb* /media/boot/overlays/
```

exander77, 2015-06-22:
Ok, I have it compiled and running, what to add to config.txt, this?
```
dtoverlay=i2c2,i_know_what_im_doing=yes
```

exander77, 2015-06-22:
Works great, I can now read and write edid information.

pelwell, 2015-06-22:
By the way, it has become a parameter instead of an overlay, so use it like this:
```
dtparam=i2c2_iknowwhatimdoing
```

exander77, 2015-06-22:
Yes, already figured, should have read the comments of the patch first.

pelwell, 2015-06-22:
Cool. I'll push that tomorrow.

popcornmix, 2015-06-22:
> Works great, I can now read and write edid information.

Out of interest what tools did you use?

kukabu, 2015-06-23:
@popcornmix http://cgit.freedesktop.org/xorg/app/edid-decode should works

pelwell, 2015-06-23:
That patch will be in the next release. Thanks for the suggestion and testing.

exander77, 2015-06-23:
I am using this two utilities:
https://github.com/bulletmark/edid-rw
https://github.com/ChalkElec/write-edid

kukabu, 2015-06-23:
@exander77, it's overhead :), try to use the eeprom kernel module.

pelwell, 2015-09-16:
@exander77 I think this issue is solved - Close if you agree.

Ruffio, 2016-08-16:
@exander77 has your issue been resolved? If so, please close this issue. Thanks.

P33M, 2017-05-17:
i2c-2 is available by an overlay, use of which requires that you know what you're doing (i.e. don't erase the contents of your monitor's EEPROM).

closed 2017-05-17 by P33M