raspberrypi/linux #1031: Support gpio-poweroff driver to allow switching off power

brycewilkins, 2015-06-22:
I tried to use the device tree GPIO binding gpio-poweroff to switch off power to the Raspberry Pi (via an on/off push-button controller) once the system reaches 'halt' state.  Please see these links for info:
https://github.com/raspberrypi/linux/blob/rpi-4.0.y/Documentation/devicetree/bindings/gpio/gpio-poweroff.txt
and
https://www.raspberrypi.org/forums/viewtopic.php?f=107&t=113789.

It did not work because the standard Raspberry Pi configurations do not include the gpio-poweroff driver which must be enabled using the POWER_RESET and POWER_RESET_GPIO flags.  I would like to request these additions.

I believe this would be an excellent feature to have support for as it will allow the RPi to actually switch off the power to a system it is controlling once it reaches a shutdown state.  Currently the RPi enters a low-power shutdown state, but circuitry controlling the power (and anything connected to it) remains on.

As the gpio-poweroff driver would only be enabled by a device tree overlay it will not cause any conflict/incompatibility with existing systems.

Thank you!

popcornmix, 2015-06-22:
Looks okay. Will add in next update.

brycewilkins, 2015-06-23:
Fantastic! Thank you so much :)  I'll be happy to test and report back when something is ready.

popcornmix, 2015-06-23:
Can you try rpi-update. The options should be in.

brycewilkins, 2015-06-24:
@popcornmix Unfortunately no success yet, here is what I did.  

- Ran rpi-update and got the 4.0.6 kernel
- Recompile dt-blob.dts to /boot/dt-blob.bin
- Recompile my power-off DT overlay and put in /boot/overlays/

Please see my thread in raspberrypi.org forum for details (link in original post above).  

I put an oscilloscope on GPIO 26 which I'm using as the power-off control signal.  The pin goes high during boot when dt-blob.bin is loaded, but when I execute "shutdown -h now" or "poweroff" the pin remains high after the 'halt' state.  (At the end of shutdown the Raspberry Pi green "disk activity" LED blinks a few times then stays off, while the red "power" LED stays on.)

The overlay seems to be loaded as I have directory /proc/device-tree/soc/gpio/power_ctrl/ and files within, but I wonder if the fragment@0 is correct.

How can I help to get it working?  Thank you!

popcornmix, 2015-06-24:
The options you requested appear to be present:

```
pi@raspberrypi:~ $ sudo modprobe configs 
pi@raspberrypi:~ $ zcat /proc/config.gz |grep POWER_RESET
CONFIG_POWER_RESET=y
# CONFIG_POWER_RESET_BRCMSTB is not set
CONFIG_POWER_RESET_GPIO=y
# CONFIG_POWER_RESET_GPIO_RESTART is not set
# CONFIG_POWER_RESET_LTC2952 is not set
# CONFIG_POWER_RESET_RESTART is not set
# CONFIG_POWER_RESET_SYSCON is not set
```
@pelwell may have some ideas.

brycewilkins, 2015-06-24:
Thanks.  I have the same output on my system.  

Interesting that there is a config for LTC2952 which has some extra features compared to LTC2951 I am using.  Seems like this should work.

pelwell, 2015-06-24:
> but I wonder if the fragment@0 is correct.

Good guess. I've commented on the thread.

notro, 2015-06-24:
If I read this correctly, gpio-poweroff won't work because pm_power_off is already set in the platform file.

[arch/arm/mach-bcm2708/bcm2708.c](https://github.com/raspberrypi/linux/blob/rpi-4.0.y/arch/arm/mach-bcm2708/bcm2708.c)
```c
void __init bcm2708_init(void)
{
...
	pm_power_off = bcm2708_power_off;
```

[drivers/power/reset/gpio-poweroff.c](http://lxr.free-electrons.com/source/drivers/power/reset/gpio-poweroff.c)
```c
static int gpio_poweroff_probe(struct platform_device *pdev)
{
        bool input = false;

        /* If a pm_power_off function has already been added, leave it alone */
        if (pm_power_off != NULL) {
                dev_err(&pdev->dev,
                        "%s: pm_power_off function already registered",
                       __func__);
                return -EBUSY;
        }
```

brycewilkins, 2015-06-25:
@notro : @pelwell noted that it should replace the standard pm_power_off handler - [please see here](https://www.raspberrypi.org/forums/viewtopic.php?p=778189#p778189)

But as I've commented [over in the tread](https://www.raspberrypi.org/forums/viewtopic.php?p=779742#p779742), it's still not working...

notro, 2015-06-25:
What does this return:
```
$ dmesg | grep poweroff
```

pelwell, 2015-06-25:
I've got it working (with caveats) - I'm writing a response for the Forum.

pelwell, 2015-06-25:
And you were correct (of course) about the existing pm_power_off handler.

pelwell, 2015-06-25:
Thread updated. Gist here: https://gist.github.com/pelwell/e4c5662c331fcae1e707

notro, 2015-06-25:
You could do:
```c
if (!force && pm_power_off != NULL) {
```

notro, 2015-06-25:
Or just remove the whole if clause, since pm_power_off will always be set in our repo.

brycewilkins, 2015-06-25:
Thanks @pelwell @notro for getting to the root of the problem!  I'd give it a try but not sure how to apply the patch...

pelwell, 2015-06-25:
I've pushed the patch to the rpi-4.0.y branch. It will be in the next firmware release.

brycewilkins, 2015-06-25:
That's really excellent news.  I will post a message here when I test it.  Thank you!

popcornmix, 2015-06-26:
rpi-update firmware should contain @pelwell's patch

brycewilkins, 2015-06-26:
It works **perfectly!**  Thank you @popcornmix @pelwell @notro for all your help in adding this feature.  No doubt a lot of people can make use of it.  You guys are awesome!  

I think this issue can be marked closed.

closed 2015-06-27 by popcornmix