raspberrypi/linux #1032: Please add support for the i2c RTC chip "MCP7941x"

M-Reimer, 2015-06-23:
It's nice that it is now possible to enable an I2C RTC directly at boot.

Unfortunately the RTC, I prefer to use, is not supported, so far.

The chip "MCP7941x" is handled with the "rtc-ds1307" module, so I tried to enable this one using

 dtoverlay=i2c-rtc,ds1307

This prepares I2C and loads the module but, of course, doesn't bind the I2C device.

If I do

 echo "mcp7941x 0x6f" > /sys/bus/i2c/devices/i2c-1/new_device

manually, then my RTC device is successfully initialized.

pelwell, 2015-06-23:
That sounds reasonable, but it has made me realise that there is either a gap in my knowledge or a gap in the device-tree I2C support (or both).

When you write to .../new_device the device string is looked up in the global alias table to locate a suitable driver, but the original device string (more precisely an index into the table of device ids supported by the driver) is passed to the driver so it knows what type of device it is being invoked for.

So far I haven't found the equivalent mechanism when using Device Tree. I thought for a while it might be the name of the node (which in this case would be "mcp7941x@6f"), but that doesn't work.

Thanks for the challenge. 8-)

notro, 2015-06-23:
> So far I haven't found the equivalent mechanism when using Device Tree.

I think it's the compatible string, but it seems the driver only supports ds1307 from DT (id=0 => ds_1307).
```c
enum ds_type {
	ds_1307,
	ds_1337,
	ds_1338,
	ds_1339,
	ds_1340,
	ds_1388,
	ds_3231,
	m41t00,
	mcp794xx,
	rx_8025,
	last_ds_type /* always last */
	/* rs5c372 too?  different address... */
};

static const struct i2c_device_id ds1307_id[] = {
	{ "ds1307", ds_1307 },
	{ "ds1337", ds_1337 },
	{ "ds1338", ds_1338 },
	{ "ds1339", ds_1339 },
	{ "ds1388", ds_1388 },
	{ "ds1340", ds_1340 },
	{ "ds3231", ds_3231 },
	{ "m41t00", m41t00 },
	{ "mcp7940x", mcp794xx },
	{ "mcp7941x", mcp794xx },
	{ "pt7c4338", ds_1307 },
	{ "rx8025", rx_8025 },
	{ }
};
MODULE_DEVICE_TABLE(i2c, ds1307_id);

static const struct of_device_id ds1307_of_match[] = {
	{ .compatible = "maxim,ds1307" },
	{ }
};
MODULE_DEVICE_TABLE(of, ds1307_of_match);
```

notro, 2015-06-23:
Strange that two dts files use a compatible string that has no support:
```
compatible = "microchip,mcp7941x";
```
http://lxr.free-electrons.com/source/arch/powerpc/boot/dts/p1023rdb.dts#L75
http://lxr.free-electrons.com/source/arch/arm/boot/dts/am57xx-beagle-x15.dts?v=3.19#L336

pelwell, 2015-06-24:
That was the conclusion I had come to - that the compatible string was the key and that the driver needed to export a larger list.

Quite a few of the published .dts/i files include compatible strings that don't match anything - "dallas,ds1307" (which is the more logical name), "mxim,ds1339" etc. It turns out this is because the DT compatible string match is only the first step in identifying a driver for an I2C device; after that it tries ACPI, then original I2C device_id mechanism, using the product part of the "vendor,product" compatible string. So:
```
			mcp7941x: mcp7941x@6f {
				compatible = "pelwell,mcp7941x";
				reg = <0x6f>;
				status = "okay";
			};
```
leads to (with a bit of extra tracing):
```
[    8.663987] i2c_device_probe(mcp7941x)
[    8.675733] ds1307_probe(type=8)
[    8.685297] rtc-ds1307: probe of 1-006f failed with error -5
```
i.e. it has correctly located the driver and passed it the correct device index of 8 (mcp794xx).

@M-Reimer I've pushed an update to the overlay source that adds the "mcp7941x" parameter, so you will be able to use `dtparam=i2c-rtc,mcp7941x` after the next firmware build. I don't have one to verify that it functions correctly, but it looks like it should.

notro, 2015-06-24:
> It turns out this is because the DT compatible string match is only the first step in identifying a driver for an I2C device; after that it tries ACPI, then the original I2C device_id mechanism, using the product part of the "vendor,product" compatible string.

That's clever. One more piece in the Device Tree puzzle in place :-)

M-Reimer, 2015-06-25:
Seems to work. I downloaded the updated ".dts" file and compiled this to ".dtb" on my own.

The RTC is now initialized "somewhere later in the boot process". Unfortunately this doesn't mean that my RTC is used to initialize the system time :(

I'm using Arch Linux ARM.

pelwell, 2015-06-25:
Try working through the steps in this forum post: https://www.raspberrypi.org/forums/viewtopic.php?p=770339#p770339

M-Reimer, 2015-06-25:
These instructions are for Debian only. No /lib/udev/hwclock-set on my system. But I think this is something that is distribution specific. I think the issue here is resolved.

closed 2015-06-25 by pelwell

joshperry, 2016-07-27:
@pelwell Did you mean `dtoverlay=i2c-rtc,mcp7941x`, I couldn't get it to work with `dtparam`? Just wondering if I'm using this wrong. Thanks for the overlay!

pelwell, 2016-07-28:
Yes I did.