raspberrypi/linux #1034: AUX_SPI1 kernel driver

vlad-zakharov, 2015-06-24:
Hello everyone,
I am working on a driver for AUX SPI1. I used @joan2937 userspace code for AUX SPI as a reference.
It's pretty simple currently, but I hope to get it into a proper state soon.
If there is any advice or suggestion regarding this driver I would be happy to hear it.
@msperl @notro @joan2937 what do you think?

msperl, 2015-06-24:
there is a post of a patch going upstream already

msperl, 2015-06-24:
as soon as it got accepted into the tree of linux I was planning to ask to merge/cherry-pick it downstream...

vlad-zakharov, 2015-06-24:
Oh, wow! Such a coincidence:)
That's great, I look forward to seeing it upstream.

msperl, 2015-06-24:
note that a quick look at your patch shows that you are not using the latest "recommended" interface for new drivers - you should have used the transfer_one interface and not the transfer interface...
please try the patch - it should work when you use the correct device-tree!

vlad-zakharov, 2015-06-30:
@msperl , Can you give me an advice how to add your patch to the kernel? And you wrote about the correct device-tree, what did you mean? I added to file linux/arch/arm/boot/dts/bcm2708.dtsi next lines from your patch:
spi1 <at> 7e215080 {
	compatible = "brcm,bcm2835-aux-spi";
	reg = <0x7e215080 0x40>, <0x7e215000 0x08>;
	interrupts = <1 29>;
	clocks = <&clk_spi>;
	#address-cells = <1>;
	#size-cells = <0>;
};
but anything happened. What I'm doing wrong?

msperl, 2015-06-30:
Driver requiers that you set cs-gpios with the chipselects to use and you also have to set the correct alt functions for the miso, mosi and sck pins as well as setting Chipselects pins  as output not alt!!!

pelwell, 2015-06-30:
I think you need to create children of the gpio node that look like this:
```
   	spi1_pins: spi1_pins {
		brcm,pins = <19 20 21>;
		brcm,function = <3>; /* alt4 */
	};
   	spi1_cs_pins: spi1_cs_pins {
		brcm,pins = <18 17 16>;
		brcm,function = <1>; /* out */
	};
```
and then reference them from the spi1 node:
```
spi1@7e215080 {
    compatible = "brcm,bcm2835-aux-spi";
    pinctrl-names = "default";
    pinctrl-0 = <&spi1_pins &spi1_cs_pins>;
    reg = <0x7e215080 0x40>, <0x7e215000 0x08>;
    interrupts = <1 29>;
    clocks = <&clk_spi>;
    gpio-cs = <&gpio 18 0>, <&gpio 17 0>, <&gpio 16 0>;
    #address-cells = <1>;
    #size-cells = <0>;
};
```
Things to bear in mind:
1) Please consider this a thumbnail sketch - I'm guessing a bit at the chip select support, but I'm sure @msperl will correct me if I'm wrong.
2) Since we are using GPIO chip selects you can have any number you like and use any pins you like, but this example mirrors the pins used by the three native chip selects for SPI1. If you need fewer you can shrink the sp1_cs_pins section accordingly.

I hope this is at least a pointer in the right direction.

[ P.S. @msperl The upstreamed documentation patch looks wrong:
> gpio-cs: the gpio-cs (native cs is NOT supported)

 Shouldn't that be cs-gpio ?
And it would be helpful if the example included the cs-gpio/gpio-cs usage. ]

vlad-zakharov, 2015-07-01:
@pelwell I have done changes you adviced, but I still have no results. May be the problem I have isn't in the device tree? Do I have to change some configuration files? When building kernel and modules I don't get any message about spi-bcm2835-aux. 
@msperl, May be you can push your code with this patch on GitHub?

msperl, 2015-07-06:
Well, the first version did not get accepted, so I am about to push a version 3 that makes use of the sysconf framework for access to the enable register.
If this is accepted, then we can starting to merge it.

vlad-zakharov, 2015-07-06:
@msperl I'd like to test yout driver. Please, let me know when new version would be available. It would be great if you could upload it on Github somewhere.

frasersdev, 2016-02-15:
...so...
I notice the spi1aux module is in the kernel (4.4.y / 4.5.y) and I would like to try and get it going.
I've built it on a rPi2 (note I had to change the spi Kconfig for it to show up in menuconfig)
The module loads, 
```
$ lsmod | grep spi
spi_bcm2835aux          4427  0 
spi_bcm2835             7286  0 
```

However it doesn't show up in sys/class etc, no log entries nothing.
I'm guessing thats a absence of device tree definition issue.

I'm new to device trees so I may need a little help here,
I can see the SPI module needs to be defined to 2 places to work,
bcm2708_common.dtsi
bcm2???-rpi-*.dts (one for each type of rPi board)
Is there anywhere else?

Once I've made the changes can I use the device-tree-compiler package in jessie to compile it?
There is a note from a few years ago that the package was too old but is this still the case?
When I compile the dtb using the jessie package I get an error
```
$ dtc -I dts -O dtb -o  bcm2709-rpi-2-b.dtb bcm2709-rpi-2-b.dts
Error: bcm2709-rpi-2-b.dts:3.1-9 syntax error
FATAL ERROR: Unable to parse input tree
```
Which I think is telling me syntax error is on line 3, characters 1-9 -> the '#include' ?

Any guidance would be much appreciated.

frasersdev, 2016-02-15:
I'm getting close now. For the record here is what I have had to do (if this is not the right place for this then I apologise in advance)

Edit ./drivers/spi/Kconfig and change the config SPI_BCM2835AUX depends on line to
`depends on ARCH_BCM2835 || ARCH_BCM2708 || ARCH_BCM2709 || COMPILE_TEST `

Edit ./arch/arm/boot/dts/bcm2708_common.dtsi and add to aliases
`spi1 = &spi1;`
and then add to soc:
```
                spi1: spi1@7e215080 {
                        compatible = "brcm,bcm2835-aux-spi";
                        reg = <0x7e215080 0x40>, <0x7e215000 0x08>;
                        interrupts = <1 29>;
                        clocks = <&clk_spi>;
                        #address-cells = <1>;
                        #size-cells = <0>;
                        status = "disabled";
                        cs-gpios = <0>, <0>, <0>;
                };
```
Edit ./arch/arm/boot/dts/bcm2709-rpi-2-b.dts (I'm using a PI2) and add to gpio:
```
        spi1_pins: spi1_pins {
                brcm,pins = <19 20 21>;
                brcm,function = <3>; /* alt4 */
        };
        spi1_cs_pins: spi1_cs_pins {
                brcm,pins = <18 17 16>;
                brcm,function = <1>; /* out */
        };
```
then add  a spi1 node (i'm not sure of this bit):
```
&spi1 {
        pinctrl-names = "default";
        pinctrl-0 = <&spi1_pins &spi1_cs_pins>;
        cs-gpios = <&gpio 18 1>, <&gpio 17 1>, <&gpio 16 1>;

        spidev@0{
                compatible = "spidev";
                reg = <0>;      /* CE0 */
                #address-cells = <1>;
                #size-cells = <0>;
                spi-max-frequency = <500000>;
        };

        spidev@1{
                compatible = "spidev";
                reg = <1>;      /* CE1 */
                #address-cells = <1>;
                #size-cells = <0>;
                spi-max-frequency = <500000>;
        };

        spidev@2{
                compatible = "spidev";
                reg = <2>;      /* CE2 */
                #address-cells = <1>;
                #size-cells = <0>;
                spi-max-frequency = <500000>;
        };
};
```
Finally add an override
`spi_aux = <&spi1>,"status";`

run make menuconfig, pick the aux spi driver, build the kernel, install the kernel, reboot
add `dtparam=spi_aux=on` to /boot/config.txt and reboot.

This gives me:
* the spi_bcm2835aux module is loaded and running automatically
* the gpio alt functions get set on pins 19,20,21
* /sys/devices/platform/soc/3f215080.spi1 exists contains information about the module
```
$ ls -l /sys/devices/platform/soc/3f215080.spi1
total 0
-rw-r--r-- 1 root root 4096 Feb 15 18:53 driver_override
-r--r--r-- 1 root root 4096 Feb 15 18:53 modalias
lrwxrwxrwx 1 root root    0 Feb 15 18:53 of_node -> ../../../../firmware/devicetree/base/soc/spi1@7e215080
lrwxrwxrwx 1 root root    0 Feb 15 18:53 subsystem -> ../../../../bus/platform
-rw-r--r-- 1 root root 4096 Feb 15 18:53 uevent
```

However
* /sys/class/spi_master/ only has spi0
* no device nodes

msperl, 2016-02-15:
You will typically also need the auxiliar clock driver installed (as only then the clock gate is enabled - but that is also upstream only).
But that should only impact functionality not configuration

Please share ```dmesg```when you have loaded the module (maybe put the module on the blacklist first) as well as the corresponding section for spi1 from ```dtc -Ifs /proc/device-tree```

frasersdev, 2016-02-15:
Thanks. yeah the clock pretty obvious now.
The phandle error is interesting. 

dmesg has the following
```
[    4.418263] /soc/spi1@7e215080: could not find phandle
[    4.418310] spi-bcm2835aux 3f215080.spi1: could not get clk: -2
[    4.428681] spi-bcm2835aux: probe of 3f215080.spi1 failed with error -2
```

dtc -Ifs /proc/device-tree produces 
```
        soc {
                spi1@7e215080 {
                        reg = <0x7e215080 0x40 0x7e215000 0x8>;
                        interrupts = <0x1 0x1d>;
                        pinctrl-0 = <0xd 0xe>;
                        compatible = "brcm,bcm2835-aux-spi";
                        cs-gpios = <0xa 0x12 0x1 0xa 0x11 0x1 0xa 0x10 0x1>;
                        clocks = <0xdeadbeef>;
                        status = "okay";
                        #address-cells = <0x1>;
                        phandle = <0x1e>;
                        #size-cells = <0x0>;
                        pinctrl-names = "default";
                        linux,phandle = <0x1e>;

                        spidev@0 {
                                reg = <0x0>;
                                compatible = "spidev";
                                spi-max-frequency = <0x7a120>;
                                #address-cells = <0x1>;
                                #size-cells = <0x0>;
                        };

                        spidev@1 {
                                reg = <0x1>;
                                compatible = "spidev";
                                spi-max-frequency = <0x7a120>;
                                #address-cells = <0x1>;
                                #size-cells = <0x0>;
                        };

                        spidev@2 {
                                reg = <0x2>;
                                compatible = "spidev";
                                spi-max-frequency = <0x7a120>;
                                #address-cells = <0x1>;
                                #size-cells = <0x0>;
                        };
                };

<snip>

                gpio@7e200000 {
                        spi1_pins {
                                phandle = <0xd>;
                                brcm,function = <0x3>;
                                brcm,pins = <0x13 0x14 0x15>;
                                linux,phandle = <0xd>;
                        };

                        spi1_cs_pins {
                                phandle = <0xe>;
                                brcm,function = <0x1>;
                                brcm,pins = <0x12 0x11 0x10>;
                                linux,phandle = <0xe>;
                        };

<snip>

        __fixups__ {
                clk_spi = "/soc/spi1@7e215080:clocks:0";
        };

<snip>

        __symbols__ {
                spi1 = "/soc/spi1@7e215080";
                spi1_pins = "/soc/gpio@7e200000/spi1_pins";
                spi1_cs_pins = "/soc/gpio@7e200000/spi1_cs_pins";
<snip>

        aliases {
                spi1 = "/soc/spi1@7e215080";
<snip>

        __overrides__ {
                spi_aux = [00 00 00 1e 73 74 61 74 75 73 00];
```

pelwell, 2016-02-15:
`clk_spi` doesn't exist - the <0xdeadbeef> is an unresolved symbol. I think you want `clk_core`.

frasersdev, 2016-02-15:
real close now

```
$ ls -l /dev/spi*
crw-rw---- 1 root spi 153, 0 Feb 15 22:53 /dev/spidev0.0
crw-rw---- 1 root spi 153, 1 Feb 15 22:53 /dev/spidev0.1
crw-rw---- 1 root spi 153, 2 Feb 15 22:53 /dev/spidev1.0
crw-rw---- 1 root spi 153, 3 Feb 15 22:53 /dev/spidev1.1
crw-rw---- 1 root spi 153, 4 Feb 15 22:53 /dev/spidev1.2
```
but sending a message fails, config (ioctl's) work though
```
$ ./spidev_test -D /dev/spidev1.0 -s 10000 -H -O
spi mode: 3
bits per word: 8
max speed: 10000 Hz (10 KHz)
can't send spi message: Connection timed out
Aborted
```
==> indicates a clock issue?

FYI spi1 is using the following clock now
```
		clock@6 {
			reg = <0x6>;
			#clock-cells = <0x0>;
			compatible = "fixed-clock";
			clock-frequency = <0x124f800>;
			clock-output-names = "osc";
			phandle = <0x3>;
			linux,phandle = <0x3>;
		};
```

So this is tomorrows challenge:
> You will typically also need the auxiliar clock driver installed (as only then the clock gate is enabled - but that is also upstream only).

msperl, 2016-02-15:
As said: you will need a clock-driver that will bring up the clock-gate.
The clk-bcm2835-aux driver that does this should come along in 4.5.

Then it should be possible to use that one directly on a downstream kernel - maybe still with a fixed clock behind it instead of the new clock framework for now...

msperl, 2016-02-15:
The osc clock you use is running at 19.2MHz which is definitely wrong and will result in unexpected results - i.e: the SPI clock will be much to fast (by a factor of 13.3)! It needs to be the VPU/core clock running (typically) at 250MHz!

If you are willing you can also just poke 0x00000007 into register 0x7e215004 and then the clock gates for spi1/2 and uart1 are also enabled.

Or you add this:
```
                aux: aux@0x7e215000 {
                        compatible = "brcm,bcm2835-aux";
                        #clock-cells = <1>;
                        reg = <0x7e215000 0x8>;
                        clocks = <&clocks BCM2835_CLOCK_VPU>;
                };
```

But for clocks choose a different clock that is available downstream - not osc, but maybe : ```<&clk_core>```

frasersdev, 2016-02-17:
well I have got it working
```
$ ./spidev_test -D /dev/spidev1.0 --verbose
spi mode: 0x0
bits per word: 8
max speed: 500000 Hz (500 KHz)
TX | FF FF FF FF FF FF 40 00 00 00 00 95 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF F0 0D  | ......@....?..................?.
RX | FF FF FF FF FF FF 40 00 00 00 00 95 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF F0 0D  | ......@....?..................?.
```

But try as I might I just could enable the aux clock gate with via the device tree.

I'm new to device trees so perhaps I have made a mistake. This has been added to bcm2708_common.dtsi in the 'soc' section
```
                aux: aux@0x7e215000 {
                        compatible = "brcm,bcm2835-aux";
                        #clock-cells = <1>;
                        reg = <0x7e215000 0x8>;
                        clocks = <&clk_core>;
                };
```
and its there 
```
/sys/devices/platform/soc/3f215000.aux $ ls -l
total 0
lrwxrwxrwx 1 root root    0 Feb 17 17:01 driver -> ../../../../bus/platform/drivers/bcm2835-aux-clk
-rw-r--r-- 1 root root 4096 Feb 17 17:01 driver_override
-r--r--r-- 1 root root 4096 Feb 17 17:01 modalias
lrwxrwxrwx 1 root root    0 Feb 17 17:01 of_node -> ../../../../firmware/devicetree/base/soc/aux@0x7e215000
lrwxrwxrwx 1 root root    0 Feb 17 17:01 subsystem -> ../../../../bus/platform
-rw-r--r-- 1 root root 4096 Feb 17 17:01 uevent
```
But I just could not work out how to enable to aux clock properly...

NOTE I needed to update this kernel makefile: https://github.com/raspberrypi/linux/blob/rpi-4.5.y/drivers/clk/bcm/Makefile as it was only building the bcm clocks for CONFIG_ARCH_BCM2835 where the rPI2 is CONFIG_ARCH_BCM2709 and all other rPis are CONFIG_ARCH_BCM2708

frasersdev, 2016-02-17:
oh - I should add I got the SPi1 module going by poking 0x7 into 0x7e215004

msperl, 2016-02-17:
as said: you will need to enable the clk-bcm2835-aux driver (which is by default not enabled for the downstream architecture 270x, but only for 283X).

Note that it may get enabled with a 4.5 kernel as it will be included there...

frasersdev, 2016-02-17:
sorry, I don't understand what you mean 
I have been using this kernel https://github.com/raspberrypi/linux/tree/rpi-4.5.y as it contains both the spi1aux and clkaux
I rebuilt the kernel with clk-bcm2835-aux enabled for the bcm2709 architecture
```
fraser@rpi:~/kernel/linux $ make -j4 zImage modules dtbs
  CHK     include/config/kernel.release
  CHK     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
make[1]: 'include/generated/mach-types.h' is up to date.
  CHK     include/generated/timeconst.h
  CHK     include/generated/bounds.h
  CHK     include/generated/asm-offsets.h
  CALL    scripts/checksyscalls.sh
  CHK     include/generated/compile.h
  CHK     kernel/config_data.h
  CC      drivers/clk/bcm/clk-bcm2835.o
  CC      drivers/clk/bcm/clk-bcm2835-aux.o
  LD      drivers/clk/bcm/built-in.o
  LD      drivers/clk/built-in.o
  LD      drivers/built-in.o
  LINK    vmlinux
  LD      vmlinux.o
  MODPOST vmlinux.o
  GEN     .version
  CHK     include/generated/compile.h
  UPD     include/generated/compile.h
  CC      init/version.o
  LD      init/built-in.o
  KSYM    .tmp_kallsyms1.o
  KSYM    .tmp_kallsyms2.o
  LD      vmlinux
  SORTEX  vmlinux
  SYSMAP  System.map
  OBJCOPY arch/arm/boot/Image
  Building modules, stage 2.
  Kernel: arch/arm/boot/Image is ready
  Kernel: arch/arm/boot/Image is ready
  GZIP    arch/arm/boot/compressed/piggy.gzip
  AS      arch/arm/boot/compressed/piggy.gzip.o
  MODPOST 1455 modules
  LD      arch/arm/boot/compressed/vmlinux
  OBJCOPY arch/arm/boot/zImage
  Kernel: arch/arm/boot/zImage is ready
...

fraser@rpi:~/kernel/linux $ sudo scripts/mkknlimg arch/arm/boot/zImage /boot/$KERNEL.img
[sudo] password for fraser: 
Version: Linux version 4.5.0-rc3-v7+ (fraser@rpi) (gcc version 4.9.2 (Raspbian 4.9.2-10) ) #2 SMP Tue Feb 16 19:15:09 AEDT 2016
```

I am running that kernel
```
$ uname -a
Linux rpi 4.5.0-rc3-v7+ #2 SMP Tue Feb 16 19:15:09 AEDT 2016 armv7l GNU/Linux
```

I have got this in the bcm2708_common.dtsi 
```
                aux: aux@0x7e215000 {
                        compatible = "brcm,bcm2835-aux";
                        #clock-cells = <1>;
                        reg = <0x7e215000 0x8>;
                        clocks = <&clk_core>;
                };
```

Is there something else I need to do to enable clk-bcm2835-aux?

msperl, 2016-02-17:
did you now use the aux clock in spi1/spi2?

should look like this:
```
                spi1: spi@7e215080 {
                       compatible = "brcm,bcm2835-aux-spi";
                       reg = <0x7e215080 0x40>;
                       interrupts = <1 29>;
                       clocks = <&aux BCM2835_AUX_CLOCK_SPI1>;
                       status = "disabled";
               };

                spi2: spi@7e2150C0 {
                       compatible = "brcm,bcm2835-aux-spi";
                       reg = <0x7e2150C0 0x40>;
                       interrupts = <1 29>;
                       clocks = <&aux BCM2835_AUX_CLOCK_SPI2>;
                       status = "disabled";
               };
```
Because only then the aux clock is used and enables the gate.

frasersdev, 2016-02-17:
> clocks = <&aux BCM2835_AUX_CLOCK_SPI1>;

that was it!  :+1: 
Obvious once its pointed out...

One tiny tiny little problem but I'll look at that tomorrow - when I compiled the dtb is didn't recognise the define (BCM2835_AUX_CLOCK_SPI1) but worked fine if I substituted the value (1). So this file: include/dt-bindings/clock/bcm2835-aux.h needs to be referenced somehow.

anyway, thanks again for your help. 
I'm going to put spi1 through its paces now and see how it holds up

frasersdev, 2016-02-20:
FYI I have submitted a pull request which if nothing else neatly wraps up the changes I needed to make.
The pull request is for kernel 4.4 as it has been raised as a LTS kernel so figured that now is a good time to mull over making the auxiliary SPI devices available.

https://github.com/raspberrypi/linux/pull/1295

chaosjug, 2016-04-05:
4.6rc1 contains some fixes for the auxiliary SPI driver.
It would be nice to include those as well.

frasersdev, 2016-04-21:
Hi,

I've created a pull request for the aux clock fix in 4.6
https://github.com/raspberrypi/linux/pull/1423

Ruffio, 2016-08-16:
@vlad-zakharov has your issue been resolved? If so, please close this issue. Thanks.

closed 2016-08-16 by vlad-zakharov