raspberrypi/linux #1038: Possible errors in linux/arch/arm/boot/dts/overlays/README

Thuffir, 2015-06-27:
The above mentioned file states the following information for the lirc_rpi overlay:

```
        gpio_in_pull             Pull up/down/off on the input pin
                                 (default "down")
```
Which suggests that the keyword __`off`__ should be used to turn off the pull resistors on the input GPIO.
Using the keyword __`off`__ actually leaves the pull-down resistor turned on. After some trying around, the keyword which really turned off both pull resistors was __`none`__.

```
        sense                    Override the IR receive auto-detection logic:
                                   "1" = force active high
                                   "0" = force active low
                                   "-1" = use auto-detection
                                 (default "-1")
```
It's actually the other way around. __`0`__ forces an active high receiver and __`1`__ forces an active low.

pelwell, 2015-06-29:
Thanks for the feedback. I agree with you on the sense of the "sense" parameter - that will be fixed in the next update.

I wasn't able to reproduce your findings with regard to the gpio_in_pull parameter though. Because it isn't possible to read back the pull settings, I instrumented the bcm2835_pinconf_set function to log pin numbers and settings. Here's what I found:
```
0 = Off disable pull-up/down
1 = Enable Pull Down control
2 = Enable Pull Up control
3 = Reserved

dtoverlay=lirc-rpi
[   11.203072] pinconf_set(18,1)

dtoverlay=lirc-rpi,gpio_in_pull=down
[   11.041564] pinconf_set(18,1)

dtoverlay=lirc-rpi,gpio_in_pull=up
[   10.942981] pinconf_set(18,2)

dtoverlay=lirc-rpi,gpio_in_pull=off
[   11.199396] pinconf_set(18,0) 

dtoverlay=lirc-rpi,gpio_in_pull=none
[   11.185545] pinconf_set(18,1)
```

This shows that "off" is functioning as intended, and that "none" is equivalent to "down". The reason for this is that the firmware doesn't have an integer value associated with "none", whereas "off" is 0, "down" is 1 and "up" is 2. If you run `sudo vcdbg log msg` you will find an error message:
```
001641.825: dtparam: gpio_in_pull=none
001642.117: dterror: invalid override value 'none' - ignored
```
The parameter is being ignored, so the default ("down", 1) is being used.

Thuffir, 2015-06-29:
I have just repeated my measurements with the latest kernel and I can also confirm your findings regarding the pull values. Sorry for the confusion.

closed 2015-06-29 by Thuffir