raspberrypi/linux #1039: rpi-4.1.y: only 1 out of 4 CPUs brought up

Kasreyn, 2015-06-29:
Only 1 out of 4 CPUs start on branch rpi-4.1.y d9fa1180.

"[bcm2709_boot_secondary] cpu:1 failed to start (9380)"

All CPUs are brought up (same hardware/unit) by latest raspbian.

Is there a known fix for this?

Uncompressing Linux... done, booting the kernel.
[    0.000000] Booting Linux on physical CPU 0xf00
[    0.000000] Linux version 4.1.0-gd9fa118 (i@ildjarn) (gcc version 4.9.1 (crosstool-NG 1.20.0) ) #4 SMP PREEMPT Thu Jun 25 16:30:21 CEST 2015
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: Raspberry Pi 2 Model B
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] [bcm2709_smp_init_cpus] enter (9380->f3003010)
[    0.000000] [bcm2709_smp_init_cpus] ncores=4
[    0.000000] PERCPU: Embedded 7 pages/cpu @c7ecd000 s7936 r0 d20736 u32768
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 32512
[    0.000000] Kernel command line: earlyprintk console=tty0 console=ttyAMA0 root=/dev/mmcblk0p2 rootwait rw
[    0.000000] PID hash table entries: 512 (order: -1, 2048 bytes)
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Memory: 125156K/131072K available (3319K kernel code, 242K rwdata, 696K rodata, 160K init, 158K bss, 5916K reserved, 0K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xc8800000 - 0xff000000   ( 872 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xc8000000   ( 128 MB)
[    0.000000]       .text : 0xc0008000 - 0xc03f3eec   (4016 kB)
[    0.000000]       .init : 0xc03f4000 - 0xc041c000   ( 160 kB)
[    0.000000]       .data : 0xc041c000 - 0xc0458988   ( 243 kB)
[    0.000000]        .bss : 0xc045b000 - 0xc0482bf4   ( 159 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] NR_IRQS:608
[    0.000000] Architected cp15 timer(s) running at 19.20MHz (virt).
[    0.000000] clocksource arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000009] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000029] Switching to timer-based delay loop, resolution 52ns
[    0.000192] Console: colour dummy device 80x30
[    0.001022] console [tty0] enabled
[    0.001066] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.001140] pid_max: default: 4096 minimum: 301
[    0.001294] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.001341] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.002210] CPU: Testing write buffer coherency: ok
[    0.002665] CPU0: update cpu_capacity 1024
[    0.002712] CPU0: thread -1, cpu 0, socket 15, mpidr 80000f00
[    0.002752] [bcm2709_smp_prepare_cpus] enter
[    0.002847] Setting up static identity map for 0x8280 - 0x82b4
[    0.080334] [bcm2709_boot_secondary] cpu:1 failed to start (9380)
[    1.080929] CPU1: failed to come online
[    1.101266] [bcm2709_boot_secondary] cpu:2 failed to start (9380)
[    2.101865] CPU2: failed to come online
[    2.122212] [bcm2709_boot_secondary] cpu:3 failed to start (9380)
[    3.122806] CPU3: failed to come online
[    3.122853] Brought up 1 CPUs
[    3.122889] SMP: Total of 1 processors activated (38.40 BogoMIPS).
[    3.122930] CPU: All CPU(s) started in SVC mode.
[    3.123729] devtmpfs: initialized
[    3.149718] VFP support v0.3: implementor 41 architecture 2 part 30 variant 7 rev 5
[    3.150060] clocksource jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    3.150265] pinctrl core: initialized pinctrl subsystem
[    3.151115] NET: Registered protocol family 16
[    3.156556] DMA: preallocated 4096 KiB pool for atomic coherent allocations
[    3.157280] bcm2709.uart_clock = 3000000
[    3.162164] Serial: AMBA PL011 UART driver
[    3.162380] 3f201000.uart: ttyAMA0 at MMIO 0x3f201000 (irq = 83, base_baud = 0) is a PL011 rev2
[    3.523588] console [ttyAMA0] enabled
[    3.615159] bcm2708-dmaengine 3f007000.dma: DMA legacy API manager at f3007000, dmachans=0x7f35
[    3.624546] bcm2708-dmaengine 3f007000.dma: Load BCM2835 DMA engine driver
[    3.631449] bcm2708-dmaengine 3f007000.dma: dma_debug:0
[    3.637404] usbcore: registered new interface driver usbfs
[    3.642989] usbcore: registered new interface driver hub
[    3.648531] usbcore: registered new device driver usb
[    3.655127] Switched to clocksource arch_sys_counter
[    3.660859] simple-framebuffer eb75000.framebuffer: framebuffer at 0xeb75000, 0x85400 bytes, mapped to 0xc8d00000
[    3.671196] simple-framebuffer eb75000.framebuffer: format=r5g6b5, mode=656x416x16, linelength=1312
[    3.683988] Console: switching to colour frame buffer device 82x26
[    3.695602] simple-framebuffer eb75000.framebuffer: fb0: simplefb registered!
[    3.716376] NET: Registered protocol family 2
[    3.724221] TCP established hash table entries: 1024 (order: 0, 4096 bytes)
[    3.733709] TCP bind hash table entries: 1024 (order: 1, 8192 bytes)
[    3.742427] TCP: Hash tables configured (established 1024 bind 1024)
[    3.751204] UDP hash table entries: 128 (order: 0, 4096 bytes)
[    3.759404] UDP-Lite hash table entries: 128 (order: 0, 4096 bytes)
[    3.768273] NET: Registered protocol family 1

Kasreyn, 2015-06-29:
Kernel config used inserted below:

#
# Automatically generated file; DO NOT EDIT.
# Linux/arm 4.1.0 Kernel Configuration
#
CONFIG_ARM=y
CONFIG_SYS_SUPPORTS_APM_EMULATION=y
CONFIG_HAVE_PROC_CPU=y
CONFIG_STACKTRACE_SUPPORT=y
CONFIG_LOCKDEP_SUPPORT=y
CONFIG_TRACE_IRQFLAGS_SUPPORT=y
CONFIG_RWSEM_XCHGADD_ALGORITHM=y
CONFIG_GENERIC_HWEIGHT=y
CONFIG_GENERIC_CALIBRATE_DELAY=y
CONFIG_NEED_DMA_MAP_STATE=y
CONFIG_ARCH_SUPPORTS_UPROBES=y
CONFIG_FIQ=y
CONFIG_VECTORS_BASE=0xffff0000
CONFIG_ARM_PATCH_PHYS_VIRT=y
CONFIG_NEED_MACH_IO_H=y
CONFIG_NEED_MACH_MEMORY_H=y
CONFIG_GENERIC_BUG=y
CONFIG_PGTABLE_LEVELS=2
CONFIG_DEFCONFIG_LIST="/lib/modules/$UNAME_RELEASE/.config"
CONFIG_IRQ_WORK=y
CONFIG_BUILDTIME_EXTABLE_SORT=y

#
# General setup
#
CONFIG_INIT_ENV_ARG_LIMIT=32
CONFIG_CROSS_COMPILE=""
# CONFIG_COMPILE_TEST is not set
CONFIG_LOCALVERSION=""
CONFIG_LOCALVERSION_AUTO=y
CONFIG_HAVE_KERNEL_GZIP=y
CONFIG_HAVE_KERNEL_LZMA=y
CONFIG_HAVE_KERNEL_XZ=y
CONFIG_HAVE_KERNEL_LZO=y
CONFIG_HAVE_KERNEL_LZ4=y
CONFIG_KERNEL_GZIP=y
# CONFIG_KERNEL_LZMA is not set
# CONFIG_KERNEL_XZ is not set
# CONFIG_KERNEL_LZO is not set
# CONFIG_KERNEL_LZ4 is not set
CONFIG_DEFAULT_HOSTNAME="(none)"
CONFIG_SWAP=y
CONFIG_SYSVIPC=y
CONFIG_SYSVIPC_SYSCTL=y
# CONFIG_POSIX_MQUEUE is not set
# CONFIG_CROSS_MEMORY_ATTACH is not set
# CONFIG_FHANDLE is not set
# CONFIG_USELIB is not set
# CONFIG_AUDIT is not set
CONFIG_HAVE_ARCH_AUDITSYSCALL=y

#
# IRQ subsystem
#
CONFIG_GENERIC_IRQ_PROBE=y
CONFIG_GENERIC_IRQ_SHOW=y
CONFIG_GENERIC_IRQ_SHOW_LEVEL=y
CONFIG_HARDIRQS_SW_RESEND=y
CONFIG_IRQ_DOMAIN=y
CONFIG_HANDLE_DOMAIN_IRQ=y
# CONFIG_IRQ_DOMAIN_DEBUG is not set
CONFIG_IRQ_FORCED_THREADING=y
CONFIG_GENERIC_TIME_VSYSCALL=y
CONFIG_GENERIC_CLOCKEVENTS=y
[... 1964 lines cut ...]
# CONFIG_XZ_DEC_POWERPC is not set
# CONFIG_XZ_DEC_IA64 is not set
# CONFIG_XZ_DEC_ARM is not set
# CONFIG_XZ_DEC_ARMTHUMB is not set
# CONFIG_XZ_DEC_SPARC is not set
# CONFIG_XZ_DEC_BCJ is not set
# CONFIG_XZ_DEC_TEST is not set
CONFIG_GENERIC_ALLOCATOR=y
CONFIG_HAS_IOMEM=y
CONFIG_HAS_IOPORT_MAP=y
CONFIG_HAS_DMA=y
CONFIG_CPU_RMAP=y
CONFIG_DQL=y
CONFIG_NLATTR=y
CONFIG_ARCH_HAS_ATOMIC64_DEC_IF_POSITIVE=y
CONFIG_AVERAGE=y
# CONFIG_CORDIC is not set
# CONFIG_DDR is not set
CONFIG_LIBFDT=y
CONFIG_FONT_SUPPORT=y
# CONFIG_FONTS is not set
CONFIG_FONT_8x8=y
CONFIG_FONT_8x16=y
# CONFIG_ARCH_HAS_SG_CHAIN is not set
# CONFIG_VIRTUALIZATION is not set

pelwell, 2015-06-29:
Try using bcm2709_defconfig:
```
make ARCH=arm ... bcm2709_defconfig zImage modules ...
```
(i.e. follow the instructions [here](https://www.raspberrypi.org/documentation/linux/kernel/building.md))

If that works then there is a fault in your configuration. If it doesn't work, I suggest you run rpi-update to update your firmware, check that boots correctly, then try your kernel again.

Kasreyn, 2015-07-01:
Thanks for the reply.

It looks to be an issue with scripts in U-Boot. It may not be something that is an issue in the kernel.

I did some research..

The image from Raspbian can boot both RPi 1 and RPi 2. Raspbian doesn't use U-Boot it boots kernel.img (or kernel7.img) directly from firmware.

I tried doing the same, my own config and defconfig can boot directly from firmware (without U-Boot) and the problem goes away.

The following options must be added to my config however:
CONFIG_ATAGS=y
CONFIG_CMDLINE_FROM_BOOTLOADER=y

I have passed the question on to U-Boot's mailing list:
http://lists.denx.de/pipermail/u-boot/2015-June/217771.html

jeremymc526, 2015-07-16:
Try this, it worked for me:

    U-Boot> fatload mmc 0:1 0x10000000 boot.scr.uimg
    reading boot.scr.uimg
    324 bytes read in 135 ms (2 KiB/s)
    U-Boot> source 0x10000000
    ## Executing script at 10000000
    switch to partitions #0, OK 
    mmc0 is current device
    reading zImage
    4020912 bytes read in 2580 ms (1.5 MiB/s)
    reading bcm2709-rpi-2-b.dtb
    9804 bytes read in 139 ms (68.4 KiB/s)
    Kernel image @ 0x1000000 [ 0x000000 - 0x3d5ab0 ]
    ## Flattened Device Tree blob at 02000000
    Booting using the fdt blob at 0x2000000
    Loading Device Tree to 3ab44000, end 3ab4964b ... OK

If all cores come online as expected, simply update
your U-Boot environment variables with the address. 

    U-Boot> sete scriptaddr 0x10000000 
    U-Boot> savee 
    Saving Environment to FAT...
    writing uboot.env
    done
    U-Boot>

Ruffio, 2016-08-16:
@Kasreyn has your issue been resolved? If so, please close this issue. Thanks.

Kasreyn, 2016-08-16:
@Ruffio Yes, i think it was never an issue with the rpi-4.1.y kernel, especially since you can make a workaround for it in U-Boot.

closed 2016-08-16 by Kasreyn