raspberrypi/linux #1044: Unable to locate package linux-headers-4.0.7

weijian19391, 2015-07-03:
Hi 

I'm installing Jool, a NAT64 program, and it prompted me to do a ````apt-get install linux-headers-$(uname -r)````. However, upon running that command, i was presented the error of : 
````
Reading state information... Done
E: Unable to locate package linux-headers-4.0.7
E: Couldn't find any package by regex 'linux-headers-4.0.7'
````
May i know whats the problem here??

Thank you!

XECDesign, 2015-07-03:
The official kernel doesn't come with a headers package (and never has). Although fixing this is on my todo list, unfortunately, it's near the bottom.

weijian19391, 2015-07-03:
@XECDesign I see. Thanks for your prompt reply!

XECDesign, 2015-07-03:
I'll close this, since it's not really a linux kernel issue. If you would like to track it, you can re-open it here https://github.com/RPi-Distro/repo/.

closed 2015-07-03 by XECDesign