raspberrypi/linux #1046: Wrong I2S alt settings in DT

HiassofT, 2015-07-04:
The I2S alt settings changes in commits 93e9bd0 / 0bc6395 don't seem to be correct for the B+ and 2B (and I guess also for the CM). On GPIOs 18-21 they should be kept at Alt0. That's also what the Alt table in the datasheet says.

I've only tested with the (not in tree) Cirrus driver on a RPi2, using Alt0 works fine, Alt2 breaks audio.

ping @pelwell @popcornmix

pelwell, 2015-07-04:
Yes, you are correct. That certainly explains the lack of complaints about the "errors" it was "fixing".

Thank you for reporting this so promptly.

"Fixed".

HiassofT, 2015-07-05:
Thanks for the quick fix!

Kernel 4.1 is working fine both on RPi B (non-plus) and RPi2.

closed 2015-07-05 by HiassofT

popcornmix, 2015-07-08:
rpi-update firmware now has this fix