raspberrypi/linux #1048: `./scripts/recordmcount: Syntax error: “(” unexpected` is reported when I compile a simple module.

d0u9, 2015-07-06:
I have built a new kernel (3.18.10) via cross-compile tools on my Debian 8 amd64. This kernel can boot correctly but report error below when I compile a simple `hello-world` module.

    make -C /lib/modules/3.18.10-d0u9/build M=/home/pi/Linux-Device-Driver-3.18/1_Hello-World modules
    make[1]: Entering directory '/usr/src/linux-source-3.18'
        CC [M]  /home/pi/Linux-Device-Driver-3.18/1_Hello-World/main.o
    ./scripts/recordmcount: 1: ./scripts/recordmcount: Syntax error: "(" unexpected
    scripts/Makefile.build:257: recipe for target '/home/pi/Linux-Device-Driver-3.18/1_Hello-      World/main.o' failed
    make[2]: *** [/home/pi/Linux-Device-Driver-3.18/1_Hello-World/main.o] Error 2
    Makefile:1398: recipe for target '_module_/home/pi/Linux-Device-Driver-3.18/1_Hello-World' failed
    make[1]: *** [_module_/home/pi/Linux-Device-Driver-3.18/1_Hello-World] Error 2
    make[1]: Leaving directory '/usr/src/linux-source-3.18'
    Makefile:19: recipe for target 'modules' failed
    make: *** [modules] Error 2

This module is very simple and you can find the code here: <https://github.com/d0u9/Linux-Device-Driver-3.18/blob/master/1_Hello-World/main.c>.

Then I recompile the kernel(same source code) directly on my Pi2, the module is compiled OK. 

So what happened to my Pi2?

lurch, 2015-11-04:
It sounds like you're doing your cross-compilation wrong - see http://stackoverflow.com/questions/17282461/scripts-recordmcount-syntax-error-when-i-try-to-build-a-linux-kernel-module-o

https://www.raspberrypi.org/documentation/linux/kernel/building.md has a section on cross-compiling the Linux kernel for RPi.

Ruffio, 2016-08-16:
@d0u9 has your issue been resolved? If so, please close this issue. Thanks.

closed 2017-05-17 by JamesH65