raspberrypi/linux #1050: DTS for Adafruit PiTFT 2.8 capacitive touchscreen ?

ondrej1024, 2015-07-09:
Hi,
Is anyone working on a DT overlay for the Adafruit PiTFT 2.8 capacitive touchscreen? 
https://www.adafruit.com/products/1983

Adafruit provide their own kernel with an overlay which can be found here:
https://github.com/adafruit/Adafruit-Pi-Kernel-o-Matic/blob/pitft/boot/overlays/pitft28c-overlay.dtb

But if would be much better of course, if the stock RPi firmware would support this display, so we wouldn't have to mess with third party kernels.

There is already an overlay pitft28-resistive-overlay.dtb for the resistive version of the display.

Thanks, Ondrej

pelwell, 2015-07-09:
We'd be happy to accept it if somebody creates a pull request (or just points me at the source). Perhaps @ladyada or @notro can help?

ondrej1024, 2015-07-09:
I wasn't able to find the original source dts file from Adafriut but from the dtb referenced above you can generate it with the device tree compiler:
`dtc -I dtb -O dts pitft28c-overlay.dtb > pitft28c-overlay.dts`

However I'm afraid that some modifications will be necessary to integrate it nicely into the RPI kernel. 

For the resistive display, the pitft28-resistive-overlay.dtb from the RPi kernel and pitft28r-overlay.dtb from Adafruit look substantially different.

ladyada, 2015-07-09:
there's a lot more than just the dts file - we have to also send over the capacitive device driver and our tweaks to the resistive device driver, both of which we know aren't going to be accepted without quite a bit of effort on my part :(

ondrej1024, 2015-07-09:
What about pitft28-resistive-overlay.dtb from the stock RPi kernel? Where did that come from and do we still need the Adafruit custom kernel for this resistive display?

ladyada, 2015-07-09:
1) no idea, i didnt submit it :)
2) i recommend it! you also get the power switch driver. we recently recompiled the kernel a few weeks ago to track this repository

pelwell, 2015-07-09:
That overlay is by @notro, who is also at least partly responsible for upstreaming the drivers used and who has contributed many similar overlays, as well as work in many other areas.

ondrej1024, 2015-07-09:
OK, I get the picture. Eventually I also found the original Adafruit dts files:
[pitft28c-overlay.dts](https://github.com/adafruit/Adafruit-Pi-Kernel-o-Matic/blob/pitft/pitft28c-overlay.dts)
[pitft28r-overlay.dts](https://github.com/adafruit/Adafruit-Pi-Kernel-o-Matic/blob/pitft/pitft28r-overlay.dts)

I was looking at the wrong files before and now I see that the dts file for the resistive display is actually the same for both Adafruit and RPi native kernel. So I expect this to work.

Looks like the main problem ATM is the touch driver for the capacitive display (Focaltech ft6x06) which is not included in the RPi native kernel. Too bad. Hope this can be sorted out sooner or later. Having to use the Adafruit kernel is not an ideal solution.

pssc, 2015-07-09:
I have the adafruit display working with the stock kernel and  pitft28-resistive-overlay.dtb touch screen and all.

pelwell, 2015-07-09:
But is it the capacitive model? It uses a completely different touchscreen controller.

pssc, 2015-07-09:
My bad I will now go sit in the corner its the restive 2.8 and 3.5in versions, I do have an additional dt overplay for mapping the buttons on the 28" to cursor keys by default (over-ridable) using the gpio_keys module.

ondrej1024, 2015-07-14:
Now I have received both the resistive and the capacitive touch screen and tested them with the stock RPi kernel (4.0.8). As expected, the resistive display works fine OOTB with the provided overlay and the kernel drivers. The capacitive display works also fine but the touch interface is not functional due to the missing [ft6x06_ts](https://github.com/adafruit/adafruit-raspberrypi-linux/blob/rpi-3.18.y/drivers/input/touchscreen/ft6x06_ts.c) input driver.
@ladyada  @notro: Is there a chance to see this driver in the mainline or at least the RPi kernel any time soon?
Thanks.

ladyada, 2015-07-14:
@ondrej1024 no change since https://github.com/raspberrypi/linux/issues/1050#issuecomment-119942798

ondrej1024, 2015-07-14:
I take that for a "no" :smirk:
Thanks anyway.

NickWaterton, 2015-08-06:
I know this is a few weeks old now, but I have successfully included the touch screen drivers (capacitive) in the latest kernel (V 4.0.9-v7+) for the RPi 2B. You just need a couple of files (and some instructions). I can't attach files here, but let me know if you need the tarball.

ladyada, 2015-08-06:
@NickWaterton can you email them to support@adafruit.com please and include a link to this thread (so i remember what it is)? i would greatly appreciate it!

NickWaterton, 2015-08-06:
E-Mail sent, let me know if security strips the tar file.

notro, 2015-10-02:
The ft6236 driver has just been [accepted](https://lkml.org/lkml/2015/10/2/645) for mainline: https://git.kernel.org/cgit/linux/kernel/git/dtor/input.git/commit/?id=92deea1350f9fa97a841d45cd1f8228d5e8aa667
I will make a PR backporting it to rpi-4.1.y after the driver shows up in [linux-next](https://git.kernel.org/cgit/linux/kernel/git/next/linux-next.git/).

ondrej1024, 2015-10-03:
@notro that sounds very good. Glad to see that things got moving eventually. Looking forward to test the driver and the Adafruit PiTFT with the rpi-4.1.y kernel when it will be available.

notro, 2015-10-30:
The driver is in the latest rpi-update kernel #1176

ondrej1024, 2015-10-30:
Thanks @notro, will try it ASAP.

ondrej1024, 2015-11-04:
I tried the latest RPi stock kernel 4.1.12 with the ft6236 driver. Now the touch input of the Adafruit PiTFT capacitive touchscreen works fine. 

Here is the specific DT overlay I wrote for the display.

<pre>
/*
 * Device Tree overlay for Adafruit PiTFT 2.8" capavitive touch screen
 *
 */

/dts-v1/;
/plugin/;

/ {
        compatible = "brcm,bcm2835", "brcm,bcm2708", "brcm,bcm2709";

        fragment@0 {
                target = <&spi0>;
                __overlay__ {
                        status = "okay";

                        spidev@0{
                                status = "disabled";
                        };

                        spidev@1{
                                status = "disabled";
                        };
                };
        };

        fragment@1 {
                target = <&gpio>;
                __overlay__ {
                        pitft_pins: pitft_pins {
                                brcm,pins = <24 25>;
                                brcm,function = <0 1>; /* in out */
                                brcm,pull = <2 0>; /* pullup none */
                        };
                };
        };

        fragment@2 {
                target = <&spi0>;
                __overlay__ {
                        /* needed to avoid dtc warning */
                        #address-cells = <1>;
                        #size-cells = <0>;

                        pitft: pitft@0{
                                compatible = "ilitek,ili9340";
                                reg = <0>;
                                pinctrl-names = "default";
                                pinctrl-0 = <&pitft_pins>;

                                spi-max-frequency = <32000000>;
                                rotate = <90>;
                                fps = <25>;
                                bgr;
                                buswidth = <8>;
                                dc-gpios = <&gpio 25 0>;
                                debug = <0>;
                        };

                };
        };

        fragment@3 {
                target = <&i2c1>;
                __overlay__ {
                        /* needed to avoid dtc warning */
                        #address-cells = <1>;
                        #size-cells = <0>;

                        /*pitft_ctp {*/
                        ft6236: ft6236@38 {
                                compatible = "focaltech,ft6236";
                                reg = <0x38>;

                                /*irq-gpio = <&gpio 24 2>;*/
                                interrupt-parent = <&gpio>;
                                interrupts = <24 2>;

                                touchscreen-size-x = <320>;
                                touchscreen-size-y = <480>;
                        };
                };
        };

        __overrides__ {
                speed =   <&pitft>,"spi-max-frequency:0";
                rotate =  <&pitft>,"rotate:0";
                fps =     <&pitft>,"fps:0";
                debug =   <&pitft>,"debug:0";
                invx =    <&ft6236>,"touchscreen-inverted-x?";
                invy =    <&ft6236>,"touchscreen-inverted-y?";
                swapxy =  <&ft6236>,"touchscreen-swapped-x-y?";
        };
};
</pre>

pelwell, 2015-11-05:
That looks good. Although there's technically no need to disable spidev@1 since only CE0 is used, having something plugged into the whole 40-way header probably makes it a moot point.

Can you create a pull request for this?

pelwell, 2015-11-05:
Ah, I see that the header pins are still accessible, so it would be better to remove the:
```
                        spidev@1{
                                status = "disabled";
                        };
```

ondrej1024, 2015-11-06:
I removed the `spidev@1` section as proposed and made some minor changes.
Created pull request #1192.

closed 2015-11-06 by ondrej1024

notro, 2016-08-25:
Just a heads up, the ft6236 driver will be gone in 4.9. It turned out that edt-ft5x06 also supports ft6236.
The kconfig symbol for this driver is: TOUCHSCREEN_EDT_FT5X06

Input: remove duplicate ft6236 driver:
https://git.kernel.org/cgit/linux/kernel/git/next/linux-next.git/commit/drivers/input/touchscreen?id=d18716542d137beafef2859bdf90d4e0c9a25523

DT binding doc patch:
https://git.kernel.org/cgit/linux/kernel/git/next/linux-next.git/commit/Documentation/devicetree/bindings/input/touchscreen/edt-ft5x06.txt?id=d18716542d137beafef2859bdf90d4e0c9a25523

pelwell, 2016-08-26:
Thanks for the warning.