raspberrypi/linux #1051: Changing SPI frequency/speed in fbtft driver causes usb based audio to play at the wrong speed

pssc, 2015-07-09:
I am seeing this on multiple pis of differing types and kernel versions setting dtparam to anything other than 25Mhz causes issues.

```
config.txt
disable_audio_dither=1

dtdebug=1
dtoverlay=pitft28-resistive-overlay
dtparam=fps=6
dtparam=speed=25000000
#dtparam=speed=32000000
#dtparam=speed=42000000
dtoverlay=pitft28-switches
#dtparam=debounce=50
```

```
Linux raspberrypi 3.18.14+ #794 PREEMPT Sun Jun 7 12:02:04 BST 2015 armv6l GNU/Linux
```

```
root@raspberrypi:/home/pi# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: CODEC [USB Audio CODEC], device 0: USB Audio [USB Audio]
  Subdevices: 0/1
  Subdevice #0: subdevice #0
```
```

Bus 001 Device 002: ID 0424:9512 Standard Microsystems Corp. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0 Unused
  bDeviceProtocol         2 TT per port
  bMaxPacketSize0        64
  idVendor           0x0424 Standard Microsystems Corp.
  idProduct          0x9512 
  bcdDevice            2.00
  iManufacturer           0 
  iProduct                0 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           41
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                2mA
    Interface Descriptor:
[... 1722 lines cut ...]
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              6 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      52
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              10
```

pelwell, 2015-07-09:
Have you determined the relationship between the "speed" parameter and the audio playback rate?

pssc, 2015-07-09:
speed set to speed=25000000 works ok
speed set to 42MHz and slow with stuttering is the consensus in the office
speed set to 32MHz struggle to tell the difference from 42Hz perhaps more stuttering and noise
speed set to 16MHz no issues.

```
Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=1920 bcm2708_fb.fbheight=1080 bcm2708.boardrev=0xf bcm2708.serial=0x333e0666 smsc95xx.macaddr=B8:27:EB:3E:06:66 bcm2708_fb.fbswap=1 sdhci-bcm2708.emmc_clock_freq=250000000 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=ttyAMA0,38400 kgdboc=ttyAMA0,38400 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait forcefsck dwc_otg.fiq_fsm_mask=0xf
```

I have frpi-bcp running at a interval of 6fps atm, to replicate onto the fbtft device.

pssc, 2015-07-09:
ok distortion slowness only occurs when fbcp is running testing with additional noise a guess that is putting traffic onto the spi bus
```
 speaker-test -D default:CARD=CODEC -t wav -c 2
```
Is this a contention issue for service the usb frames?

pssc, 2015-07-09:
On board audio is unaffected.

notro, 2015-07-09:
This might be related: #971

pssc, 2015-07-09:
```
 graphics fb1: fb_ili9340 frame buffer, 320x240, 150 KiB video memory, 0 KiB DMA buffer memory, fps=6, spi0.0 at 42 MHz
```

```
diff --git a/arch/arm/boot/dts/overlays/pitft28-resistive-overlay.dts b/arch/arm/boot/dts/overlays/pitft28-resistive-overlay.dts
index d506eae..8b0eb45 100644
--- a/arch/arm/boot/dts/overlays/pitft28-resistive-overlay.dts
+++ b/arch/arm/boot/dts/overlays/pitft28-resistive-overlay.dts
@@ -49,6 +49,7 @@
 				pinctrl-0 = <&pitft_pins>;
 
 				spi-max-frequency = <32000000>;
+				txbuflen = <152>;
 				rotate = <90>;
 				fps = <25>;
 				bgr;
@@ -108,6 +109,7 @@
 
 	__overrides__ {
 		speed =   <&pitft>,"spi-max-frequency:0";
+		txbuflen = <&pitft>,"txbuflen:0";
 		rotate =  <&pitft>,"rotate:0";
 		fps =     <&pitft>,"fps:0";
 		debug =   <&pitft>,"debug:0";
```
seems to work so should

```
fbtft-core.c

       /* Transmit buffer */
        if (txbuflen == -1)
                txbuflen = vmem_size + 2; /* add in case startbyte is used */

#ifdef __LITTLE_ENDIAN
        if ((!txbuflen) && (bpp > 8))
                txbuflen = PAGE_SIZE; /* need buffer for byteswapping */
#endif

        if (txbuflen > 0) {
                if (dma) {
                        dev->coherent_dma_mask = ~0;
                        txbuf = dmam_alloc_coherent(dev, txbuflen, &par->txbuf.dma, GFP_DMA);
                } else {
                        txbuf = devm_kzalloc(par->info->device, txbuflen, GFP_KERNEL);
                }
                if (!txbuf)
                        goto alloc_fail;
                par->txbuf.buf = txbuf;
                par->txbuf.len = txbuflen;
        }
```
My reading is this should allocated 152?  which is 4k aligned (if not explicitly allocated)
I dont get why the DMA buffer is now 0, mayhap this is why the audio is now ok as no DMA?

pelwell, 2015-07-09:
Because (int)(152/1024) = 0?

notro, 2015-07-09:
> Because (int)(152/1024) = 0?

Yes, the code does not expect a buffer less than 1 KiB:
```c
        if (par->txbuf.buf)
                sprintf(text1, ", %zu KiB %sbuffer memory",
                        par->txbuf.len >> 10, par->txbuf.dma ? "DMA " : "");
```

Anyways, you won't get DMA in the official kernel since the in-kernel SPI driver doesn't support it yet: #1036

pssc, 2015-07-09:
I though the txbuflen parameter was in K from #971

pelwell, 2015-07-09:
From that thread:

    The fbtft default is 4k=4096

pssc, 2015-07-09:
@pelwell yup brain failure on my part
So default value is page size on little endian(eg arm... 4096 in this case) rather than vmem

pssc, 2015-07-10:
@notro  txbuf by lowering this what limitations and affect does this have? do we effective stall updates once the txbuffer is full?

pssc, 2015-07-10:
txbuflen=152k (Failure clear Stutter)
txbuflen=76k (Failure clear Stutter)
txbuflen=8096 (worse tones with no audio playing...)
txbuflen=4096 fail 
txbuflen=2048 fail
txbuflen=1024 fail (close)
txbuflen=512 ok

@32Mhz txbuflen=1024 -- seems good 2k fails

So dropping txbuflen makes the USB audio work with an SPI frequency of 42Mhz, distortion is related to system load and spi activity, so why... We have no DMA, is the SPI txbuf copy taking to long when clocking at 42Mhz?

And have I negated the advantage of up clocking the SPI bus as my buffer transfers per cycle are now smaller and therefore more of them needed...

pssc, 2015-07-10:
Revised @32Mhz txbuflen=512 ok
Further testing revealed the odd blip and distortion with 1k

So @42-32Mhz 512 coincidence...

pssc, 2015-07-10:
@notro does the current rpi-update give DMA?

notro, 2015-07-10:
> txbuf by lowering this what limitations and affect does this have? do we effective stall updates once the txbuffer is full?

The tx buffer is needed because the SPI controller driver does not support 16-bit words (RGB565 pixels), so we have to swap the bytes (little to big endian) before transferring them to the display controller.
The effect of a small txbuf is that the whole display update takes longer, because of the setup time for each SPI transfer (waking a kernel thread).
But to be honest, I don't know why it helps to lower the txbuflen value.
Can the tx buffer copying stall USB DMA?
The SPI hw FIFO is ~120 bytes, so one interrupt per 120 bytes. I can't imagine that this should stall anything.
Without DMA we max out around [2MB/s](https://github.com/notro/fbtft/wiki/Performance)

> does the current rpi-update give DMA?

No, not yet. See #1036 

In the Adafruit forum where this sort of problem surfaced for the first time, an [out-of-tree SPI DMA driver](https://github.com/notro/spi-bcm2708) was used. I believe Adafruit still uses that driver in their kernels. I also used that driver in my [FBTFT kernels](https://github.com/notro/fbtft/wiki#install) (discontinued).

pssc, 2015-07-11:
> Can the tx buffer copying stall USB DMA?

This would be my guess given the load happened once the systems loaded and audio one of the few applications thats time sensitive , is that happening in the interrupt handler or similar?  Does they buyte swapping occur before this in a kernel thread?
would setting txbuf len be better at 120 or say 240 bytes?

notro, 2015-07-11:
Here's a writeup of what happens:

When someone writes to /dev/fb1, [fb_deferred_io_mkwrite](http://lxr.free-electrons.com/ident?i=fb_deferred_io_mkwrite) is called which schedules delayed work to happen in a kernel thread.
After a delay (default HZ/20) the worker thread is woken up:
[fbtft_deferred_io](http://lxr.free-electrons.com/ident?i=fbtft_deferred_io) -> [fbtft_update_display](http://lxr.free-electrons.com/ident?i=fbtft_update_display) -> [fbtft_write_vmem16_bus8](http://lxr.free-electrons.com/ident?i=fbtft_write_vmem16_bus8) (txbuf is filled) -> [fbtft_write_spi](http://lxr.free-electrons.com/ident?i=fbtft_write_spi)
-> [spi_sync](http://lxr.free-electrons.com/ident?i=spi_sync)->[__spi_sync](http://lxr.free-electrons.com/ident?i=__spi_sync) -> [__spi_pump_messages](http://lxr.free-electrons.com/ident?i=__spi_pump_messages) -> [bcm2835_spi_transfer_one](http://lxr.free-electrons.com/ident?i=bcm2835_spi_transfer_one)
Short transfer \<30us: [bcm2835_spi_transfer_one_poll](http://lxr.free-electrons.com/ident?i=bcm2835_spi_transfer_one_poll) Loops filling FIFO, no interrupt.
Other transfers: [bcm2835_spi_transfer_one_irq](http://lxr.free-electrons.com/ident?i=bcm2835_spi_transfer_one_irq) (fill FIFO) ----> [bcm2835_spi_interrupt](http://lxr.free-electrons.com/ident?i=bcm2835_spi_interrupt)

I haven't calculated when a polling transfer kicks in at your speeds.

Note: the issue I linked to links to an Adafruit forum thread. That kernel used a DMA SPI driver, so the operation would differ from this one. No polling mode on small transfers for instance.

> would we be using start bytes here?

Not on this display.

pssc, 2015-07-13:
Thought on above reducing txbuf means we end up in Short transfer?

Any other testing I can do to shed light on this?

notro, 2015-07-13:
I have calculated the max transfer length for polled transfers:
```
speed  cdiv    len
--------------------------
62MHz    4     208 bytes
41MHz    6     138
31MHz    8     104
25MHz   10      83

len = 30*250/(9*cdiv)
```
@msperl If the FIFO is 120 bytes, shouldn't we always use polled mode for len\<120?

If you can build the kernel yourself, you can try and disable the transfer buffer and thus the copying, to see if that changes anything. You will get strange colors.
Remove or comment out this part to allow txbuflen to be zero:
[fbtft_framebuffer_alloc](http://lxr.free-electrons.com/ident?i=fbtft_framebuffer_alloc)
```c
//#ifdef __LITTLE_ENDIAN
//        if ((!txbuflen) && (bpp > 8))
//                txbuflen = PAGE_SIZE; /* need buffer for byteswapping */
//#endif

```

This will lead to writing the framebuffer memory directly over SPI:
```c
/* 16 bit pixel over 8-bit databus */
int fbtft_write_vmem16_bus8(struct fbtft_par *par, size_t offset, size_t len)
{
...
        /* non buffered write */
        if (!par->txbuf.buf)
                return par->fbtftops.write(par, vmem16, len);
```
Note: this codepath have hardly seen any testing.

msperl, 2015-07-13:
@notro: in you asked about spi-Fifo : it is 64 Bytes in size.

But what I would guess is that you have transfers like this: 4096, 4096,...,4096,64
Then all transfers would be using interrupt mode except for the last one with 64 bytes, which would be polling for 30us (more so if there is contention on the CPU and it gets interrupted).

From what I have seen is that I get "glitches" on USB audio on upstream kernels (without USB fiq) reliably.

I am now still in a business trip, but when I get back I will try to cherry-pick the dma patch (which is in 4.2rc already) for a 4.1 kernel and then that may change the picture...
But note that there may be other sources for glitches besides CPU: in the dma case it could also be access to the ram/saturation on axi-bus that could slow things down (and here because of dma)

pssc, 2015-07-14:
hmmm ok  well from my testing it doesn’t seem related to clocking other than not 25Mhz  have set txbuflen on 2x pi's and its ok  600 this is well above the polling limit, will mod up a kernel this week and rule it out completely.

pssc, 2015-07-30:
Wondering if some of the issue s that have come to light in #1077 might be relevant, I am emerging from my SDl hacking(now does the job of fbcp in and accelerated fb driver for the pi), what should i test?

notro, 2015-07-30:
The latest rpi-update kernel has SPI DMA support. Have you tried that?
```
dtoverlay=spi-dma
```

pssc, 2015-08-03:
I have now 
...[   12.574636] graphics fb1: fb_ili9340 frame buffer, 320x240, 150 KiB video memory, 4 KiB DMA buffer memory, fps=12, spi0.0 at 32 MHz

Linux raspberrypi 4.0.9+ #807 PREEMPT Fri Jul 24 15:14:04 BST 2015 armv6l GNU/Linux

cat /dev/random > /dev/fb1  & speaker-test -c 2 -t wav  -D default:CARD=CODEC
...

Results in distortion, distortion only occurs when updating the framebuffer.

msperl, 2015-08-04:
Well, that would point to latencies/delays during interrupts which would explain the situation.
With dma we have a very low number of interrupts - say 1 interrupt every 4k with dma vs. 1 int for every 64 bytes. As USB also triggers irqs this can mean some additional latencies in irq handling so that the USB-sound card runs out of buffer to "bridge" the gap...

But there is another option - especially with a rpi model 1 (single CPU):
It may be about CPU consumed by:
* filling frame-buffer (cat)
* updating the screen from the frame buffer
* spi-interrupts
* playing a sound (speaker-test)
* transfer USB ( from wav buffer to USB-card)
* USB-interrupts.

All of those consume CPU and if there is any time where there is lots of activity not enough CPU may be made available to speaker-test as everything is spent handling interrupts. So the sound output may stall just because of that...

Here again dma reduces the number of interrupts, but it still would be a rare event that you would get you in such a situation, so it may not be as "recognizable" as the situation without dma

pssc, 2015-08-06:
Ok thoughts that no point does ALSA indicate a buffer under run from the sound card (this happens on the internal sound occasionally ) of course the usb driver and or HW(tested with 3 different usb sound chips)  could be broken.

msperl, 2015-08-07:
When I am back at my logic-analyzer I can try to make some measurements again measuring spi and the analog wave-form (and maybe instrument the interrupt handlers).

But what I have seen on an upstream kernel (without fast irq) was that there were situations when we have been inside an interrupt for a few ms. And 1ms is 40 samples so with 8-bit stereo 80 bytes that are consumed.

So obviously there is some buffer (say 512bytes) on the device, but eventually it will drain and if no new data comes in to fill up the data, then there will be a glitch in the audio signal.

And I think I have seen some times when we ran in interrupt-handlers for 32ms (sic), which would mean 2560 bytes getting taken from the buffer in the meantime...

So the USB latency seems really important here, but fast-irq should reduce this.

msperl, 2015-08-17:
OK  - a quick initial test shows that when I run:
```aplay /usr/share/sounds/alsa/Front_Center.wav```
there is a USB dataframe of 192 bytes going to PID A04/0 E01
every 0.001 seconds.

Assuming that is is 16 bit stereo this means 48 samples.
And with 48khz this means the frame is played in 0.001s.

This means that there needs to be a data-packet sent every 0.001s or the sound will stop (assuming no buffering).

So with SPI-Interrupts consuming CPU this will change.

I still need to reproduce the exact situation with DMA disabled for SPI to proof that this is the case...

msperl, 2015-08-17:
Here now replicated with a stock 4.1 kernel from the foundation with DMA disabled:
[image]
Playing Big Buck Bunny as well as a 440Hz signal on audio.

Zooming in it becomes obvious that while there is no SPI traffic (=screen update happening) the waveform is fine and the "sound packets" go ove USB every 1ms.

But as soon as there is SPI traffic these USB-audio packets become very rare - and result in a loss of audio - you only get to hear noise...
[image]
(Note those markers show those USB audio packages while there is no SPI traffic)

That is the situation with DMA disabled - with DMA enabled there are minimal interrupts, so  this is not becoming an "issue" as a single one time glitch is not easily detectable...

Still it may be recommended that the interrupts are not ALL running on CPU 0 but are handled by all CPUs...

Here /proc/interrupts:
```
           CPU0       CPU1       CPU2       CPU3
 16:          0          0          0          0   ARMCTRL  16 Edge      bcm2708_fb dma
 24:        118          0          0          0   ARMCTRL  24 Edge      DMA IRQ
 25:       2436          0          0          0   ARMCTRL  25 Edge      DMA IRQ
 32:     297354          0          0          0   ARMCTRL  32 Edge      dwc_otg, dwc_otg_pcd, dwc_otg_hcd:usb1
 49:          0          0          0          0   ARMCTRL  49 Edge      3f200000.gpio:bank0
 50:          0          0          0          0   ARMCTRL  50 Edge      3f200000.gpio:bank1
 65:         97          0          0          0   ARMCTRL  65 Edge      3f00b880.mailbox
 66:          2          0          0          0   ARMCTRL  66 Edge      VCHIQ doorbell
 75:          1          0          0          0   ARMCTRL  75 Edge
 80:    9030976          0          0          0   ARMCTRL  80 Edge      3f204000.spi
 83:          9          0          0          0   ARMCTRL  83 Edge      uart-pl011
 84:       6115          0          0          0   ARMCTRL  84 Edge      mmc0
 99:      19156      14312      13204      13129   ARMCTRL  99 Edge      arch_timer
484:          0          0          0          0  pinctrl-bcm2835   4 Edge      ads7846
FIQ:              usb_fiq
IPI0:          0          0          0          0  CPU wakeup interrupts
IPI1:          0          0          0          0  Timer broadcast interrupts
IPI2:     101674     109421     175736     147783  Rescheduling interrupts
IPI3:         11         11         12         11  Function call interrupts
IPI4:          3          5          0          2  Single function call interrupts
IPI5:          0          0          0          0  CPU stop interrupts
IPI6:          0          0          0          0  IRQ work interrupts
IPI7:          0          0          0          0  completion interrupts
Err:          0
```

P33M, 2015-08-17:
On BCM2835 you have a single core which has to handle all interrupts. The USB fiq was implemented because of various hardware requirements meaning any interrupt latency over 125uS caused problems. In this case, the maximum acceptable latency then turns into ~500uS for this application - approximately half the USB device update rate. Very few commonly-encountered kernel operations disable interrupts for this length of time.

If the SPI driver is spending a lot of time with interrupts disabled or in a HardIRQ context, then this is a bad behaviour. Using DMA is a solution for this (one interrupt on completion of DMA).

On BCM2836, there is no vectored interrupt controller. There is simply a mux that selects which core GPU IRQ or GPU FIQ interrupts go to - which is then chained into the core-local interrupt handling. One possibility that I have yet to implement (had working, but then a kernel version update broke it) is to (ab)use a CPU local mailbox such that USB interrupts occur on a single designated core, but this will not be possible to implement for any other interrupt source.

msperl, 2015-08-17:
Actually the SPI driver just does a complete in the irq context (plus a few register writes):
https://github.com/raspberrypi/linux/blob/rpi-4.1.y/drivers/spi/spi-bcm2835.c#L139

But the big problem is that this interrupt may happen in very short intervals (about every 64 bytes so depending on SPI-frequency it may trigger every 0.0000005s) , so it is essentially consuming CPU handling IRQ (including overhead) as well as context switching leaving no time for anything else when you have to transfer 128000 bytes resulting in 2000 interrupts in 0.0005s (assuming 125MHz)...

That is 400 interrupts per 1 ms - and with the 20us per interrupt overhead (measured a few month ago) we are at 0.8ms/ms that we just spend in handling interrupts (without any completion) and this stalls USB sound...

Obviously DMA solves that, but only to some extent.

Those IPI2 interrupts seem to move the interrupts to CPU 0, so maybe a bit more logic could get applied so that instead of rescheduling the interrupt some of them are handled locally...

P33M, 2015-08-17:
IPIs don't move interrupts off cores.
http://lxr.free-electrons.com/source/kernel/sched/core.c#L1543
IPI2 is simply a request from one core to another core to run schedule() at some point. This picks a task to wake up (if sleeping) or schedule a higher priority task (if something's running).

There's no sane way you can cause GPU IRQs to be routed to arbitrary cores. There's a single line coming from the old ARMCTRL block and this can be selected to a single core only. I briefly considered what would happen if you automatically changed the mux setting mid-IRQ handler but that grows into a horrendous monstrosity of handling code very quickly and is not worth doing.

Unless they come from the per-CPU local interrupt sources on 2836, all interrupts are handled by a single core.

msperl, 2015-08-18:
OK that IPI2 explains why the total IPI2 counts mostly correlate with interrupts - most interrupts (at least the ones I use) all make use of complete() to wake a kernel-thread and that is what IPI2 essentially does...

Ruffio, 2016-08-16:
@pssc has your issue been resolved? If so, please close this issue. Thanks.

pssc, 2016-08-16:
@Ruffio I will retest and take appropriate action

P33M, 2017-05-17:
SPI has had DMA enabled by default for ages - closing.

closed 2017-05-17 by P33M