raspberrypi/linux #1055: Motion not working after firmware upgrade.

ifermon, 2015-07-10:
This release (d4945b3b77d29cc5bb3777734422c048c1f1d003 ) broke motion for me. Got kernel panics. Motion stated in the grey screen (video served on 8080) that it was unable to open device (/dev/video0). Validated that device existed and motion log did not mention issues with device, it said that it was unable to use the video format (8). Running rasbian on an un-overclocked (non-overclocked?) rPi v2. Everything updated / upgraded. Using the pi camera module. Rolled back and everything worked fine, but I'm happy to roll back forward to break it again and do some debugging, just let me know if I can help. Rolled back to 19debde. Apologies if this is the wrong place to put this.

6by9, 2015-07-10:
Please provide logs of the kernel panic text.
Motion can be a pain to set up, so it would be useful to avoid wasting time having to try to get it running before getting anything useful.

6by9, 2015-07-11:
@pelwell This is likely to be linked to the VCHQ changes :-(

popcornmix, 2015-07-11:
@ifermon I believe this update included vchiq changes to both kernel and firmware.
Can you test with latest kernel, but previous firmware?
(e.g. update to latest firmware, then overwrite start*.elf and fixup*.dat from the previous firmware)
That will narrow down the issue to gpu firmware or kernel.

ifermon, 2015-07-11:
I will do that. It will have to wait until Monday.

6by9, 2015-07-12:
I'm not sure whether reverting just the firmware will help too much. Phil's commit text on https://github.com/raspberrypi/linux/commit/5d5a5ceb829329f490dd1c5cb634da66ef5d0f92 says that the firmware added a DT entry to say that it supported the new mechanism, and the kernel changes only activated if it found this.
It might exclude a silly error, but not sure it will really narrow down the issue.

The V4L2 driver does some flushing itself, so it's possible that is interfering with the new flushing. Not convinced though as the logs quoted on https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=62364&start=350#p787559 appear to show it being the setup phase failing (no reports of kernel panics). 
I'll try to do some testing this afternoon/evening. Seeing whether the steps in the first entry of https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=62364 work for basic capturing would be a start.

6by9, 2015-07-13:
Just reproduced. HEAD/d4945b3 fails, but 19debde is OK, so it does look like the VCHIQ change.

Setting up the driver with ```v4l2-ctl -v width=1280,height=720,pixelformat=YU12``` works fine, but starting streaming with ```v4l2-ctl --stream-mmap=3 --stream-count=100 --stream-to=/dev/null``` has caused a kernel oops due to a paging request. Call stack is showing it in ieee80211_phy0: rt2x00usb_vendor_request (yes, I'm using wifi), but I suspect it's a memory trample.
The system does recover though on ctrl-C, though throws the videobuf2_core warning due to not cleaning up all buffers properly (issue already logged, but not had a chance to investigate).

I will try with the previous firmware and current kernel, but as I commented above, I think that will just disable the kernel changes. 
Testing raspistillyuv may also be useful as that will start throwing around large buffers via bulk receives, so I'll give that a go.

popcornmix, 2015-07-13:
There were actual vchiq changes to firmware, as well as the DT changes.
So using latest firmware without latest kernel may be a valid test.

6by9, 2015-07-13:
Yup. I'm just looking at the diffs to see what changed and how messy latest firmware with old kernel could be.
Sounds like a rogue memcpy at the moment though, at a guess on fully aligned transfers (original issue was with H264 which are random length transfers. YU12 will be nicely aligned)

6by9, 2015-07-13:
Not got any further in testing this, and need to work on something else again now.
raspiyuv seems happy at 640x480, so alignment may be a red herring.

pelwell, 2015-07-13:
I'm just finishing up another issue, so I can pick this up.

ifermon, 2015-07-13:
At this point, is there anything I can do to help? It sounds like you've already been able to recreate the issue. I'd like to help, but my C skills are weak at best. No way could I find a memory leak in a driver. But if you want me to do some other tests or get some logs then please let me know. I'd also be happy to test a fix - I can do builds.

pelwell, 2015-07-13:
Thanks for the offer, but with a reliable way to reproduce this (6by9's v4l2-ctl command) it shouldn't take long to fix.

pelwell, 2015-07-14:
Disappointing - it was a cut-and-paste error into a code path not tested by my existing tests. I've pushed a patch to the 4.0 and 4.1 kernel branches, and with it included I see no new(*) problems with the v4l2-ctl test.

(*) At shutdown time, the bcm2835-v4l2 driver causes a kernel WARNING, both with and without the recent VCHIQ changes. I'll create a new issue for that.

pelwell, 2015-07-14:
Ah - it looks like I've rediscovered https://github.com/raspberrypi/linux/issues/817.

popcornmix, 2015-07-14:
Can you test after running rpi-update?

ifermon, 2015-07-15:
You guys rock. Just ran rpi-update and running motion with no errors.  Happy to help in the future, just send me a message - appreciate all you guys / gals do to move this forward.

closed 2015-07-15 by popcornmix