raspberrypi/linux #1056: Please tag commit bb6b4b6b331680bed807605685572d727638bb51 as rpi-bootloader-v1.20150421-1

diederikdehaas, 2015-07-11:
If someone wants to build a kernel module for the kernel released with `raspberrypi-bootloader` version 1.20150421-1, one has to go through a complicated process to figure out that the corresponding commit for the kernel source is bb6b4b6b331680bed807605685572d727638bb51.
You need that commit id to check that commit out so you can then prepare the kernel for module compilation.

If you would tag that commit with a name that's easily associable with that release of `raspberrypi-bootloader`, it would make the process of module compilation a LOT easier.
(Tagging commit 6cc27e02dde4f1e02cfddb3d0f23d54be87680cb of https://github.com/raspberrypi/firmware in a similar way would then make it easier to get `Module.symvers` and/or `Module7.symvers` for that release).

As a bonus, GitHub would then tell you how many commits there have been since that release which would then allow ppl to use the `--depth` parameter to `git clone`.

See also https://github.com/raspberrypi/linux/issues/915#issuecomment-89548327

P33M, 2015-07-27:
Ping @XECDesign - is this something we can integrate into the workflow for package updates?

XECDesign, 2015-07-27:
It wouldn't be much extra effort, but does it make sense? These repos have been distro agnostic so far. Plus, there is #915.

Maybe tagging in https://github.com/RPi-Distro/firmware makes more sense? When the linux kernel has it's own package, I can add package version tags there as well.

popcornmix, 2015-07-27:
The firmware repo contains the git hash of the linux tree it was built from:
https://github.com/raspberrypi/firmware/blob/master/extra/git_hash

I guess that could be installed somewhere with raspberrypi-bootloader, then the exact linux source can be easily identified.

XECDesign, 2015-07-27:
The way to get to the kernel commit right now is to check the 'firmware as of' string in the changelog, then that git_hash file. So there's certainly a way, it's just not pretty.
```
wget https://github.com/raspberrypi/linux/archive/$(wget -qO- "https://raw.githubusercontent.com/raspberrypi/firmware/$(zcat /usr/share/doc/raspberrypi-bootloader/changelog.Debian.gz | egrep -m 1 'firmware as of ([a-f0-9]+)' | sed 's/.* //')/extra/git_hash").tar.gz -O linux-pi-current.tar.gz
```

diederikdehaas, 2015-07-27:
I'm currently trying to write up a document on how to do it (see https://github.com/diederikdehaas/cknow.org/blob/master/rpi/compile-kernel-module-on-raspberrypi.md and https://www.raspberrypi.org/forums/viewtopic.php?p=790803&sid=dfa593d4011dfbea34a229652ac2175a#p790803). 
Fixing #915 would make that _process_ easier.

But this issue is meant quite literary: tag commit bb6b4b6b331680bed807605685572d727638bb51 as the one which is used for raspberrypi-bootloader version 1.20150421-1:
```
git tag -a rpi-bootloader-v1.20150421-1 bb6b4b6b331680bed807605685572d727638bb51
<enter tag comment/message>
git push --tags
```

diederikdehaas, 2016-03-02:
Even though this specific commit wasn't tagged, I still consider the issue fixed as the releases now do get tagged properly.
Thanks @XECDesign :+1:

closed 2016-03-02 by diederikdehaas