raspberrypi/linux #1058: Multiple 1-Wire busses?

phmarek, 2015-07-13:
I've got two DS18B20 in the immediate vicinity of my RPi, so querying them via GPIO works fine.

Now I'd like to add some more; I'm afraid that the length of cables required (star, not bus) won't work anymore, so I'd like to ask about the state of multiple w1 buses on multiple GPIO pins.

ISTR that there have been patches floating around for that already; could that be picked too, please?

pelwell, 2015-07-13:
What's missing is a way to parameterise DT node/property names as well as values. Without that you need one overlay for each concurrent w1 bus - something we've already seen with the two mcp2515 overlays.

ctc, 2016-03-22:
This is a patch for 10 fixes bus-number 1wire busses with dt overlay and dynamic bind/unbind support and some other special addons.

there is also a bug with devm_kzalloc and 1wire - i have only patched it to work with dt mode.
See "devm_kzalloc is freed on unload!" comments and https://lkml.org/lkml/2015/3/5/64

dt sample:
dtoverlay=w1-gpio-10
dtparam=pullup1=y
dtparam=pullup2=y
dtparam=pullup3=y
dtparam=pullup4=y
dtparam=pullup5=y
dtparam=pullup6=y
dtparam=pullup7=y
dtparam=pullup8=y
dtparam=pullup9=y
dtparam=pullup10=y
dtparam=bus1=1
dtparam=bus2=2
dtparam=bus3=3
dtparam=bus4=4
dtparam=bus5=5
dtparam=bus6=6
dtparam=bus7=7
dtparam=bus8=8
dtparam=bus9=9
dtparam=bus10=10

```
diff --git a/arch/arm/boot/dts/overlays/Makefile b/arch/arm/boot/dts/overlays/Makefile
index 75692d0..e13a8e9 100644
--- a/arch/arm/boot/dts/overlays/Makefile
+++ b/arch/arm/boot/dts/overlays/Makefile
@@ -63,6 +63,7 @@ dtb-$(RPI_DT_OVERLAYS) += vc4-kms-v3d-overlay.dtb
 dtb-$(RPI_DT_OVERLAYS) += vga666-overlay.dtb
 dtb-$(RPI_DT_OVERLAYS) += w1-gpio-overlay.dtb
 dtb-$(RPI_DT_OVERLAYS) += w1-gpio-pullup-overlay.dtb
+dtb-$(RPI_DT_OVERLAYS) += w1-gpio-10-overlay.dtb
 
 targets += dtbs dtbs_install
 targets += $(dtb-y)
diff --git a/arch/arm/mach-bcm2708/bcm2708.c b/arch/arm/mach-bcm2708/bcm2708.c
index 1b9c3be..037a887 100644
--- a/arch/arm/mach-bcm2708/bcm2708.c
+++ b/arch/arm/mach-bcm2708/bcm2708.c
@@ -80,9 +80,28 @@
 #define DMA_MASK_BITS_COMMON 32
 
 // use GPIO 4 for the one-wire GPIO pin, if enabled
-#define W1_GPIO 4
+#define W1_GPIO1 4
+#define W1_GPIO2 17
+#define W1_GPIO3 18
+#define W1_GPIO4 27
+#define W1_GPIO5 22
+#define W1_GPIO6 23
+#define W1_GPIO7 24
+#define W1_GPIO8 10
+#define W1_GPIO9 9
+#define W1_GPIO10 25
+
 // ensure one-wire GPIO pullup is disabled by default
-#define W1_PULLUP -1
+#define W1_PULLUP1 -1
+#define W1_PULLUP2 -1
+#define W1_PULLUP3 -1
+#define W1_PULLUP4 -1
+#define W1_PULLUP5 -1
+#define W1_PULLUP6 -1
+#define W1_PULLUP7 -1
[... 920 lines cut ...]
+				<&w1_5_pins>,"brcm,pins:0";
+		pullup5 =	<&w1_5>,"rpi,parasitic-power:0";
+
+		gpiopin6 =	<&w1_6>,"gpios:23",
+				<&w1_6_pins>,"brcm,pins:0";
+		pullup6 =	<&w1_6>,"rpi,parasitic-power:0";
+
+		gpiopin7 =	<&w1_7>,"gpios:24",
+				<&w1_7_pins>,"brcm,pins:0";
+		pullup7 =	<&w1_7>,"rpi,parasitic-power:0";
+
+		gpiopin8 =	<&w1_8>,"gpios:10",
+				<&w1_8_pins>,"brcm,pins:0";
+		pullup8 =	<&w1_8>,"rpi,parasitic-power:0";
+
+		gpiopin9 =	<&w1_9>,"gpios:9",
+				<&w1_9_pins>,"brcm,pins:0";
+		pullup9 =	<&w1_9>,"rpi,parasitic-power:0";
+
+		gpiopin10 =	<&w1_10>,"gpios:25",
+				<&w1_10_pins>,"brcm,pins:0";
+		pullup10 =	<&w1_10>,"rpi,parasitic-power:0";
+	};
+};
```

pelwell, 2016-03-22:
We aren't accepting any more changes to the board support code (bcm2708.c and bcm2709.c) - RPi releases will be switching to 4.4 in a matter of days, and the code you have modified no longer exists. All platform devices must now be instantiated using Device Tree, and non-DT operation is no longer supported.

The w1-gpio changes are not Pi-specific and should go upstream. Reading that thread you linked to it sounds as though Markus Pargmann would have started upstreaming the crash fix in April last year if you had replied to him.

That 10-channel overlay is an inelegant solution because it always tries to create all 10 channels, with 10 unique GPIOs, regardless of whether the user wants that many. I have a relatively simple change in the pipeline that allows fragments to be conditionally enabled and disabled. A further change to allow parametric node names (load a single-channel overlay 10 times) needs more work.

Ruffio, 2016-08-16:
@phmarek has your issue been resolved? If so, please close this issue. Thanks.

P33M, 2017-05-17:
There is a dtoverlay tweak where you can specify multiple 1-wire overlay instances.
https://www.raspberrypi.org/forums/viewtopic.php?f=107&t=156734

closed 2017-05-17 by P33M