raspberrypi/linux #1060: Consider enabling GPT/GUID partition-table support (CONFIG_EFI_PARTITION)

srcshelton, 2015-07-13:
With a custom kernel, RPi hardware/firmware boots entirely correctly with a GPT present (although I did create a Hybrid MBR, so that the firmware has no issues finding the FAT32 partition - I haven't tested a pure GPT setup).

Given that the overhead of additionally supporting GPT in the stock kernel is minimal and it makes rebuilding installs which don't want to have to contend with logical partitions much easier, would it be possible to set CONFIG_EFI_PARTITION=y by default?

Thanks in advance,

Stuart

popcornmix, 2015-07-14:
Are you sure it's not enabled?
$ sudo modprobe xonfigs
$ zcat /proc/config.gz |grep EFI
CONFIG_EFI_PARTITION=y

srcshelton, 2015-07-14:
Note entirely - I've had a somewhat forced move from an RPi to an RPi2, and the stock kernel doesn't get any further than waiting for the identified by UUID root device.
The bcmrpi defconfig doesn't specify EFI_PARTITION that I could see though, and so I assumed this was the source of my problem.
Sounds as if it's back to the drawing board... :(

popcornmix, 2015-07-14:
bcmrpi_defconfig only lists non-default options. It's likely EFI_PARTITION is enabled by another config option.

srcshelton, 2015-07-14:
In that case, well anticipated for including EFI support ;)

Now I just need to figure out the /real/ reason why it's waiting forever for the device containing the root filesystem to appear...

Ferroin, 2015-07-14:
You may want to try booting with the option 'gpt' on the kernel commandline.

srcshelton, 2015-07-15:
Good thought, but adding 'gpt' has had no effect, unfortunately.  This all worked fine on several RPiA/B/B+ machines, so I'm missing something here...

pelwell, 2015-07-15:
How about posting the active contents of your config.txt and cmdline.txt? And any bit of kernel boot log you can get, e.g. via the serial port on pins 6, 8 & 10.

And to answer your earlier question, bcm2709_defconfig is only used for Pi2.

srcshelton, 2015-07-15:
Sure thing!

Boot log via serial console:

```
Uncompressing Linux... done, booting the kernel.
[    0.000000] Booting Linux on physical CPU 0xf00
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.0.8-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.8.3 20140303 (prerelease) (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03) ) #803 SMP PREEMPT Mon Jul 13 20:15:11 BST 2015
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: Raspberry Pi 2 Model B Rev 1.1
[    0.000000] cma: Reserved 8 MiB at 0x3d800000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] [bcm2709_smp_init_cpus] enter (8660->f3003010)
[    0.000000] [bcm2709_smp_init_cpus] ncores=4
[    0.000000] PERCPU: Embedded 11 pages/cpu @bcf01000 s16128 r8192 d20736 u45056
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 251720
[    0.000000] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=853 bcm2708_fb.fbheight=480 bcm2709.boardrev=0xa01041 bcm2709.serial=0xd9cd6d9c smsc95xx.macaddr=B8:27:EB:CD:6D:9C bcm2708_fb.fbswap=1 bcm2709.disk_led_gpio=47 bcm2709.disk_led_active_low=0 sdhci-bcm2708.emmc_clock_freq=250000000 vc_mem.mem_base=0x3ea00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=tty0 console=ttyAMA0,115200 root=UUID=def94b06-98e3-4aab-b8c4-394cdbabd15a rootfstype=ext4 elevator=deadline rootwait zcache gpt
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 988384K/1015808K available (5885K kernel code, 378K rwdata, 1784K rodata, 400K init, 757K bss, 19232K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xbe800000 - 0xff000000   (1032 MB)
[    0.000000]     lowmem  : 0x80000000 - 0xbe000000   ( 992 MB)
[    0.000000]     modules : 0x7f000000 - 0x80000000   (  16 MB)
[    0.000000]       .text : 0x80008000 - 0x8078575c   (7670 kB)
[    0.000000]       .init : 0x80786000 - 0x807ea000   ( 400 kB)
[    0.000000]       .data : 0x807ea000 - 0x80848a64   ( 379 kB)
[    0.000000]        .bss : 0x80848a64 - 0x809061cc   ( 758 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] 	Additional per-CPU info printed with stalls.
[    0.000000] NR_IRQS:608
[    0.000000] Architected cp15 timer(s) running at 19.20MHz (virt).
[    0.000013] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 3579139424256ns
[    0.000037] Switching to timer-based delay loop, resolution 52ns
[    0.000319] Console: colour dummy device 80x30
[    0.002070] console [tty0] enabled
[    0.002136] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.002237] pid_max: default: 32768 minimum: 301
[    0.002601] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002669] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.003987] Initializing cgroup subsys blkio
[    0.004072] Initializing cgroup subsys memory
[    0.004139] Initializing cgroup subsys devices
[    0.004199] Initializing cgroup subsys freezer
[    0.004276] Initializing cgroup subsys net_cls
[    0.004395] CPU: Testing write buffer coherency: ok
[    0.004517] ftrace: allocating 20007 entries in 59 pages
[    0.053399] CPU0: update cpu_capacity 1024
[    0.053483] CPU0: thread -1, cpu 0, socket 15, mpidr 80000f00
[    0.053530] [bcm2709_smp_prepare_cpus] enter
[    0.053683] Setting up static identity map for 0x54eb70 - 0x54eba4
[    0.113297] [bcm2709_boot_secondary] cpu:1 started (0) 18
[    0.113613] [bcm2709_secondary_init] enter cpu:1
[    0.113664] CPU1: update cpu_capacity 1024
[    0.113673] CPU1: thread -1, cpu 1, socket 15, mpidr 80000f01
[    0.133290] [bcm2709_boot_secondary] cpu:2 started (0) 17
[    0.133546] [bcm2709_secondary_init] enter cpu:2
[    0.133577] CPU2: update cpu_capacity 1024
[    0.133585] CPU2: thread -1, cpu 2, socket 15, mpidr 80000f02
[    0.153323] [bcm2709_boot_secondary] cpu:3 started (0) 18
[    0.153562] [bcm2709_secondary_init] enter cpu:3
[    0.153592] CPU3: update cpu_capacity 1024
[... 161 lines cut ...]
device_tree_param=i2c1=on
device_tree_param=i2s=on
device_tree_param=spi=on
gpu_mem=16
kernel=kernel7.img
sdtv_mode=2
sdtv_aspect=3
hdmi_force_hotplug=1
hdmi_drive=2
avoid_pwm_pll=1
```

`blkid | grep mmc`:

```
/dev/mmcblk0p1: LABEL="RaspberryPi" UUID="9875-0503" TYPE="vfat" PARTLABEL="Microsoft basic data" PARTUUID="b35e53ff-74e9-4f3e-9898-45252a2ddde4" 
/dev/mmcblk0p2: LABEL="root" UUID="def94b06-98e3-4aab-b8c4-394cdbabd15a" TYPE="ext4" PARTLABEL="Linux filesystem" PARTUUID="05f3b69b-4999-4fa2-88a4-44491f2171be" 
/dev/mmcblk0p3: LABEL="var" UUID="361d1e72-0767-4e21-8923-b4777bd5a134" TYPE="ext4" PARTLABEL="Linux filesystem" PARTUUID="4525f0ce-b7ca-4260-97e6-79796527d316" 
[... cut ...] " PARTUUID="eb3d5b0a-d0d8-48ba-96e8-3f05b9de1492" 
/dev/mmcblk0p5: LABEL="space" UUID="497b3be3-e0f8-4775-b8ee-93f3ef065668" TYPE="ext4" PARTLABEL="Linux filesystem" PARTUUID="4d97c7e9-66af-4309-9da2-3c1323bdc3de"
```

... and I've just realised that my problem is that I was using `root=UUID=xxx` which requires an initrd to setup correctly, but I had to disable this because my initrd is xz compressed and the stock kernel doesn't support this compression.

Thanks for jogging my memory on this! ;)

closed 2015-07-15 by srcshelton