raspberrypi/linux #1064: Read status of power led using sysfs

jvwag, 2015-07-15:
I want to read the status of GPIO35 on a Raspberry Pi 2 Model B using the sysfs on Raspbian with a 4.0.8-v7+ kernel. I can access other GPIO pins using /sys/class/gpio but the two leds are already accessable by /sys/class/leds/led[0,1].

My assumption was that putting led1 (the red led) into trigger mode input the state of the led could be read using the brightness attribute. This value does not seem to change and is always on 0.

When using non-sysfs tools I can access the state of GPIO35 without a problem: raspi-gpio get 35 will change its level value from 1 to 0 when an under-voltage occurs. I hope to use one pattern to access all GPIO's by file monitoring using select or poll and not revert to a different solution for the leds.

The reason why I'm monitoring GPIO35 is to see if there are voltage issue and report those to the end user, advising to change the adapter and/or cable.

- Does led trigger input does what I think it should do, or could this be a bug?
- Could I configure led1 not to be a led but a gpio in sysfs?

(This is a cross-post, also on http://raspberrypi.stackexchange.com/questions/33479/read-status-of-power-led-using-sysfs-with-select-poll)

pelwell, 2015-07-15:
The leds-gpio module does not currently implement the brightness get mechanism, but I think it could be made to without too much effort. It has already been modified to get ledtrig-input to work, so making further changes isn't an issue.

pelwell, 2015-07-15:
That was easy. It will be in the next release.

jvwag, 2015-07-15:
Wow, thank you Phil! I will try to test your changes if I can setup a build environment.

One more question: will this enable me to use a select(2) or poll(2) on the brightness attribute? On the gpio sys subsystem this will only work if you enable an edge attribute.

pelwell, 2015-07-16:
Unfortunately no. The sysfs interface for LEDs is a straight read/write interface, and the structure of the LEDs subsystem insulates the individual LED drivers from the sysfs entries. However, there are some LEDs modules that add their own sysfs entries, so it should be possible.

closed 2016-10-27 by jvwag