raspberrypi/linux #1076: could you add "MMC_CAP_ERASE" to the bcm2835-sdhost driver mmc->caps initialization?

rasmorthil, 2015-07-19:
I've been using the new "sdhost" driver, and it's overclocking abilities are amazing.  E.g., I can now run my card at 40MB/s read speeds (as compared to 18MB/s under the "mmc" driver).

However the current implementation of this new driver in the "rpi-4.0.y" tree is missing the "MMC_CAP_ERASE" capability for the host controller configuration, and as such this prevents "fstrim" from working ("fstrim" gives an "FITRIM ioctl failed: Operation not supported" error).  Adding the MMC_CAP_ERASE capability fixes this issue, and the "sdhost" driver will then support the TRIM feature like the current "bcm2835-mmc" driver does.

I've enabled this capability in my own kernel build, but it'd be great to get this in upstream so that I don't need to keep building my own kernel just for this simple change...

The patch to bcm2835-sdhost.c is really simple:

    diff --git a/drivers/mmc/host/bcm2835-sdhost.c b/drivers/mmc/host/bcm2835-sdhost.c
    index d65870a..57a6ad3 100644
    --- a/drivers/mmc/host/bcm2835-sdhost.c
    +++ b/drivers/mmc/host/bcm2835-sdhost.c
    @@ -1675,7 +1675,7 @@ int bcm2835_sdhost_add_host(struct bcm2835_host *host)
            /* host controller capabilities */
            mmc->caps |= /* MMC_CAP_SDIO_IRQ |*/ MMC_CAP_4_BIT_DATA |
                    MMC_CAP_SD_HIGHSPEED | MMC_CAP_MMC_HIGHSPEED |
    -               MMC_CAP_NEEDS_POLL | MMC_CAP_HW_RESET |
    +               MMC_CAP_NEEDS_POLL | MMC_CAP_HW_RESET | MMC_CAP_ERASE |
                    (ALLOW_CMD23 * MMC_CAP_CMD23);
    
            spin_lock_init(&host->lock);

Thanks!

pelwell, 2015-07-20:
I'm happy to add it, provided it works. Which card are you using? None of mine (SanDisk class 10, Lexar class 10, no-name class 6) have the required capabilities. In fact there's a discussion in https://github.com/raspberrypi/linux/issues/431 where Gordon says that SD cards don't really support (and in fact don't need) trimming.

rasmorthil, 2015-07-20:
Indeed, I just verified that performing a TRIM request doesn't appear to do anything even though the card capabilities report it as being supported (!).  At least this is the case on my SD card (Samsung 32GB class 6 that "supports" the required capabilities).  Here's how I verified this...:

First, let's create a file, then get the file's inode number:

    root@gandalf:~/tmp# echo ABCDEFG >foobar
    root@gandalf:~/tmp# sync
    root@gandalf:~/tmp# ls -i foobar
    2646 foobar
    root@gandalf:~/tmp# 

Now, we'll find the physical sector of this file:

    root@gandalf:~/tmp# debugfs /dev/mmcblk0p2
    debugfs 1.42.9 (4-Feb-2014)
    debugfs:  stat <2646>
    Inode: 2646   Type: regular    Mode:  0644   Flags: 0x80000
    Generation: 659272254    Version: 0x00000000:00000001
    User:     0   Group:     0   Size: 8
    File ACL: 0    Directory ACL: 0
    Links: 1   Blockcount: 8
    Fragment:  Address: 0    Number: 0    Size: 0
     ctime: 0x55ad0732:479c39fc -- Mon Jul 20 07:35:30 2015
     atime: 0x55ad0701:55fa95e8 -- Mon Jul 20 07:34:41 2015
     mtime: 0x55ad0732:479c39fc -- Mon Jul 20 07:35:30 2015
    crtime: 0x55ad0701:55fa95e8 -- Mon Jul 20 07:34:41 2015
    Size of extra inode fields: 28
    EXTENTS:
    (0):418372
    debugfs:  q
    root@gandalf:~/tmp# 

Next, we'll read the sector to verify that it indeed contains the contents of the file:

    root@gandalf:~/tmp# dd if=/dev/mmcblk0p2 ibs=4096 count=1 skip=418372 | xxd | head -1
    1+0 records in
    8+0 records out
    4096 bytes (4.1 kB) copied, 0.00180056 s, 2.3 MB/s
    0000000: 4142 4344 4546 470a 0000 0000 0000 0000  ABCDEFG.........
    root@gandalf:~/tmp# 

Now, let's delete the file, and we'll see that the physical contents are still there:

    root@gandalf:~/tmp# rm foobar
    root@gandalf:~/tmp# sync
    root@gandalf:~/tmp# dd if=/dev/mmcblk0p2 ibs=4096 count=1 skip=418372 | xxd | head -1
    1+0 records in
    8+0 records out
    0000000: 4142 4344 4546 470a 0000 0000 0000 0000  ABCDEFG.........
    4096 bytes (4.1 kB) copied, 0.00782803 s, 523 kB/s
    root@gandalf:~/tmp# 

What will the magical "fstrim" command do?  (The moment we've all been waiting for...)

    root@gandalf:~/tmp# fstrim -v /
    /: 48234496 bytes were trimmed
    root@gandalf:~/tmp# sync
    root@gandalf:~/tmp# dd if=/dev/mmcblk0p2 ibs=4096 count=1 skip=418372 | xxd | head -1
    1+0 records in
    8+0 records out
    4096 bytes (4.1 kB) copied, 0.00204582 s, 2.0 MB/s
    0000000: 4142 4344 4546 470a 0000 0000 0000 0000  ABCDEFG.........
    root@gandalf:~/tmp#

Absolutely nothing!

So...it would appear that "fstrim" actually doesn't do anything for this particular card.  However keep in mind that we can't actually say that other cards might not actually support this functionality correctly.

Before I say "nevermind", I am thinking that the right thing to do regardless would be to support this in the driver by default anyway.  If a particular card's capabilities indicates that it supports this, shouldn't the driver allow/enable it regardless?  This way the driver is properly supporting (and not arbitrarily limiting) what a particular card's firmware implementation actually may do.

Thanks!

pelwell, 2015-07-21:
I've added the ERASE capability to the sdhost driver - it will be in the next release.

rasmorthil, 2015-07-22:
Great, thanks!

closed 2015-07-22 by rasmorthil

linvinus, 2016-08-16:
@rasmorthil  , 
comment for post https://github.com/raspberrypi/linux/issues/1076#issuecomment-122912035
just want to mention that after `sync`  but before every `dd` you should drop caches
 `echo 3 > /proc/sys/vm/drop_caches`

same test was done under ubuntu 16.04 and sd card `SanDisk Ultra 8Gb` ext4, after trim and drop cache i get zeros instead `ABCDEFG`

rasmorthil, 2016-08-25:
@linvinus - many thanks for your comment!  I believe that the success of this ultimately depends on the appropriate support within the SD card.  For example, my Transcend 64GB MLC card (https://www.amazon.com/gp/product/B00VHW7OKO/) does not zero out the data following the TRIM operation, with or without the drop_caches.  The important thing is that the kernel driver allows this capability, so that if the card does support it then it will work, as it does for your card.

linvinus, 2016-09-06:
@rasmorthil ok, i can confirm, samsung evo plus 32Gb same behavior as yours Transcend.
is this bad or not? what do you think?
samsung evo plus have good performance, but for system disk on raspberry probably not the best choice

rasmorthil, 2016-09-27:
@linvinus It all depends on how many P/E cycles are made to the card.  If you have a large card that is mostly empty, that can help significantly.  (This is also one reason I went with a large MLC card instead of a cheaper, smaller TLC.)  You can also move logging, samba cache, etc. to tmpfs to help reduce writes to the flash.  You may want to use the "fatrace" utility as well to help show what is writing to the card, so that you can optimize your system - see https://www.piware.de/2012/02/fatrace-report-system-wide-file-access-events/.