raspberrypi/linux #1078: Audio Overlay Documentation Confusion

mvastola, 2015-07-20:
In [arch/arm/boot/dts/overlays/README](https://github.com/raspberrypi/linux/blob/rpi-4.0.y/arch/arm/boot/dts/overlays/README) on the rpi-4.0.y branch, the following is stated (currently on lines 81-82):
```text
audio               Set to "on" to disable the onboard ALSA audio interface (default "off")
```

I would imagine this is a simple typo, in which case the documentation needs to be updated, but in the case the documentation is actually correct, please consider this submission to be a feature request to make this configuration setting and the accompanying documentation more intuitive.

Thanks,
Mike

pelwell, 2015-07-20:
No, it is a simple typo that I will correct - thanks - but the behaviour of the system as a whole is more complicated.

In order for the ALSA PCM sound output to be available, two things are required - the creation of a platform device and the loading of a driver module. Currently the platform device is always created - if Device Tree doesn't do it (e.g. using `dtparam=audio=on`), then the board support code will. Enabing audio from Device Tree will cause the module to be loaded automatically, otherwise it will have to be loaded manually, usually from /etc/modules. This approach allows a staged transition from always-on (because it is in /etc/modules) to on-if-enabled-in-DT. Once the config.txt setting has been added to new installs and there is a suitable upgrade script (one which spots snd-bcm2835 in /etc/modules entry, removes it and adds `dtparam=audio=on` to config.txt`), then the hack in the board support code can be removed.

See [this](https://github.com/raspberrypi/linux/pull/981) thread and [this](https://github.com/raspberrypi/linux/commit/1e1ac984e2dfe0d5c290fb9a5e5703c29006fd7c) commit for more details.

pelwell, 2015-07-21:
There is a patch for the typo in the source tree, and it will appear in the next release. With this patch and the above explanation, can we close this issue?

mvastola, 2015-07-21:
Certainly. Thanks a ton!

closed 2015-07-21 by pelwell

NicoHood, 2016-06-10:
I read in the arch wiki that this option in config.txt is now essential. I installed all related alsa packages, pulseaudio etc and also set the config.txt option. But on arch linux I will get no sound output. The installed kodi package can output audio though. I also forced the output to the audio jack, rather than hdmi. I checked pavucontrol and alsamixer. Any ideas why it still does not work?

popcornmix, 2016-06-10:
What is in config.txt?
What does ``aplay -l`` report?

NicoHood, 2016-06-10:
```
[alarm@alarmpi ~]$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

```

# See /boot/overlays/README for all available options

[pi0]
# USB OTG driver
dtoverlay=dwc2

[pi1]
# Overclocking Turbo
arm_freq=1000
core_freq=500
sdram_freq=500
over_voltage=6
temp_limit=70

[pi2]

[pi3]

[all]
gpu_mem=128
disable_overscan=1
hdmi_force_hotplug=1
disable_splash=1
dtparam=audio=on
framebuffer_depth=32
framebuffer_ignore_alpha=1

# Get the monitor setting via
# tvservice -n
# /opt/vc/bin/tvservice -n
[EDID=ACR-M230HDL]
# DVI Monitor Mode
hdmi_drive=1
```

popcornmix, 2016-06-10:
Okay, the dtparam=audio=on is working.
I suspect this is a configuration issue. I'd suggest removing puleaudio and any custom audio config and checking if aplay can play a wav file.

NicoHood, 2016-06-10:
I always use this command to test audio:
`aplay /usr/share/sounds/alsa/Front_Center.wav`

It did not work, even when i had no pulseaudio etc installed. I started from little packages and tried with installing more, but nothing helped.

popcornmix, 2016-06-10:
Afraid we can't offer much support for Arch.
If you can try a raspbian install I suspect it would just work, but if it doesn't then we can try to help.
Open a new issue though, as yours doesn't seem related to this one.

NicoHood, 2016-06-11:
On raspi1 (with arch linux for pi1) aplay -l does not even recognize a soundcard. alsamixer has at least a sound bar that i can move up and down. config.txt is set correct. kodi works here too (even with no additional alsa packages installed, but maybe they were pulled in via dependency). I know that you don't do arch for rpi, but isnt that strange? rpi1 has the latest 4.4-13 kernel.

popcornmix, 2016-06-11:
kodi doesn't use ALSA (it uses OpenMAX for audio), so its behaviour isn't relevant.
You would would be better off asking Arch packaging questions in an Arch forum - I have no knowledge of that.

NicoHood, 2016-06-12:
Thanks for the help popcornmix, I will try to solve the issue. I'd like to add some information here, as people might find this on google. I'll edit this post, if i find out more about it. You may comment if you have further ideas. If not, thats all fine :)

So I plugged in my external usb sound card and tried to play sound with it. And it works.
```
aplay -D plughw:1,0 /usr/share/sounds/alsa/Front_Center.wav
```

So there must be something wrong with the analog output. I might try hdmi output as well. And as the command itself (for analog) gives no error, I assume that there must be something wrong with the sound volume or the sound setting in config.txt (maybe i could try to force the audio jack).

pelwell, 2016-06-12:
Have you used raspi-config to force the audio to use the analogue outputs? By default it will use HDMI in preference.

NicoHood, 2016-06-12:
Sure i did, via the command line. On arch there is no raspi-config