raspberrypi/linux #1082: ARM Optimization making 4.2 impossible to compile

Belphemur, 2015-07-21:
Hello,

When I tried to compile (today) the 4.2 branch I got:
```
arch/arm/kernel/built-in.o:(___ksymtab+mmiocpy+0x0): undefined reference to `mmiocpy'
arch/arm/kernel/built-in.o:(___ksymtab+mmioset+0x0): undefined reference to `mmioset'
```

I have been able to trace this to this patch: 1bd46782d08b01b73df0085b51ea1021b19b44fd .

If I revert it, I have a compiled and working kernel for Raspberry Pi B. I used the bcmrpi_defconfig + zswap module ( with zbud and lz4 compiled in the kernel). No other change done to the default config.

pelwell, 2015-07-21:
Yes, same here. On the Pi 1 we use optimised implementations of memcpy and memset. These need the equivalent mmiocpy and mmioset aliases adding.

A patch is imminent.

pelwell, 2015-07-21:
That works for me now:
```
[    0.000000] Linux version 4.2.0-rc3+ (phil@phil-VirtualBox) (gcc version 4.8.3 20140303 (prerelease) (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03) ) #612 PREEMPT Tue Jul 21 11:20:38 BST 2015
```

Belphemur, 2015-07-21:
Compile works too !
Since I get those stack trace in dmesg, I don't know to what it's linked. This is before applying your patch.

http://pastie.org/10304085

pelwell, 2015-07-21:
That is a different problem - it looks to be an ethernet checksum failure. Either there really is a problem in the hardware or there is a bug in the software.

Can you create a new issue for this, giving as much information as you can - steps to reproduce, frequency etc.?

Belphemur, 2015-07-21:
That what I thought too. Just wanted to be sure it wasn't linked to a memory access failure.

I'll create another issue with my whole dmesg.

closed 2015-07-21 by Belphemur