raspberrypi/linux #1083: IPv6 Checksum error with kernel 4.2

Belphemur, 2015-07-21:
Raspberry Pi B 256 MB.
Current kernel branch 4.2 with bcmrpi_defconfig + zswap + zbud + lz4.

Triggering app: [syncthing](https://github.com/syncthing/syncthing). and swapper.

Router TPLink TL-WR841N with OpenWRT and IPv6 for local lan.
Raspberry Pi have openvpn installed and configured (but not used when the error arise).

Full dmesg log with all the StackTrace:
http://pastie.org/10304133

Ifconfig:
```
br0       Link encap:Ethernet  HWaddr b8:27:eb:fc:dd:77
          inet addr:192.168.42.2  Bcast:192.168.42.255  Mask:255.255.255.0
          inet6 addr: fe80::545e:b8ff:fe97:cb51/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:59523 errors:0 dropped:0 overruns:0 frame:0
          TX packets:24568 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:28257772 (26.9 MiB)  TX bytes:73773783 (70.3 MiB)

eth0      Link encap:Ethernet  HWaddr b8:27:eb:fc:dd:77
          inet addr:192.168.42.2  Bcast:192.168.42.255  Mask:255.255.255.0
          inet6 addr: fe80::ba27:ebff:fefc:dd77/64 Scope:Link
          UP BROADCAST RUNNING PROMISC MULTICAST  MTU:1500  Metric:1
          RX packets:59778 errors:0 dropped:0 overruns:0 frame:0
          TX packets:65460 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:28282513 (26.9 MiB)  TX bytes:76734164 (73.1 MiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:5511 errors:0 dropped:0 overruns:0 frame:0
          TX packets:5511 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:550126 (537.2 KiB)  TX bytes:550126 (537.2 KiB)

tap0      Link encap:Ethernet  HWaddr 56:5e:b8:97:cb:51
          inet6 addr: fe80::545e:b8ff:fe97:cb51/64 Scope:Link
          UP BROADCAST RUNNING PROMISC MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:6622 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:100
          RX bytes:0 (0.0 B)  TX bytes:2583973 (2.4 MiB)
```

P33M, 2015-07-27:
Is this a regression between the current stable (4.0.y) branch and 4.2? 4.2 is still in -rc.

Belphemur, 2015-07-27:
Yes it wasn't happening in the 4.0.y neither 4.1.y.
I know it's still in RC and I'll continue to test until it's stable.

Btw it doesn't trigger a kernel panic and the Raspberry Pi is stable (other than those exception in the dmesg).

What I don't understand is where does those IPv6 packet are coming/going. None of the interfaces have an IPv6 set and ip6tables is completely empty.

P33M, 2015-07-27:
I suspect this is an upstream bug. Bisecting between 4.1 and 4.2 branches is likely to find the culprit - as this bug appears to be unique to your setup then you're best-equipped to do this.

Ringdingcoder, 2015-09-25:
Hmm, I'm getting exactly the same on an x86_64 desktop machine with a Marvell Sky2 Ethernet chip when another machine on the network does DHCP requests.

Stack trace: http://pastie.org/10443904

airend, 2016-03-22:
In case this is the same issue, here's a possible fix: https://github.com/xbianonpi/xbian/issues/814#issuecomment-199881762.

Ringdingcoder, 2016-03-25:
@airend Thanks, I tested these patches on x86_64. Needs torvalds/linux@fdc5432a7b44ab7de as well. The issue has gone away.

popcornmix, 2016-03-31:
Latest ``sudo BRANCH=next rpi-update`` firmware cherry-picks some upstream commits that may fix this issue. Can you update and test?

Belphemur, 2016-04-04:
I can confirm the issue is gone with the same setup and the kernel 4.4.6+

closed 2016-04-04 by Belphemur