raspberrypi/linux #1088: kernel trace cause by hostapd

scottleibrand, 2015-07-25:
Trying to set up an infrastructure-mode wifi network on the pi, and am triggering a stack trace every time I try to connect a client to it.  The client makes a DHCP request (to dnsmasq), gets assigned an IP, and then hostapd stack-traces and disconnects the wifi connection.

```
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.568900] ------------[ cut here ]------------
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.569018] WARNING: CPU: 0 PID: 12016 at net/mac80211/ieee80211_i.h:895 sta_info_alloc+0x
390/0x3b4 [mac80211]()
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.569102] Modules linked in: 8021q garp stp llc usbserial ctr ccm snd_bcm2835 snd_pcm sn
d_seq snd_seq_device snd_timer snd arc4 rt2800usb rt2800lib rt2x00usb rt2x00lib mac80211 cfg80211 crc_ccitt rfkill evdev cdc_acm u
io_pdrv_genirq uio
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.569117] CPU: 0 PID: 12016 Comm: hostapd Tainted: G        W      3.18.11-v7+ #781
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.569159] [<800172bc>] (unwind_backtrace) from [<80012f7c>] (show_stack+0x20/0x24)
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.569182] [<80012f7c>] (show_stack) from [<80520970>] (dump_stack+0x98/0xd8)
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.569205] [<80520970>] (dump_stack) from [<80025618>] (warn_slowpath_common+0x7c/0xa0)
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.569227] [<80025618>] (warn_slowpath_common) from [<800256f8>] (warn_slowpath_null+0x2c
/0x34)
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.569300] [<800256f8>] (warn_slowpath_null) from [<7f0b8ee4>] (sta_info_alloc+0x390/0x3b
4 [mac80211])
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.569519] [<7f0b8ee4>] (sta_info_alloc [mac80211]) from [<7f0d3234>] (ieee80211_add_stat
ion+0x84/0x2bc [mac80211])
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.569768] [<7f0d3234>] (ieee80211_add_station [mac80211]) from [<7f03c024>] (nl80211_new
_station+0x31c/0x470 [cfg80211])
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.569957] [<7f03c024>] (nl80211_new_station [cfg80211]) from [<80479cfc>] (genl_rcv_msg+
0x25c/0x3dc)
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.569981] [<80479cfc>] (genl_rcv_msg) from [<80479130>] (netlink_rcv_skb+0xbc/0xd0)
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.570000] [<80479130>] (netlink_rcv_skb) from [<80479a8c>] (genl_rcv+0x30/0x44)
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.570019] [<80479a8c>] (genl_rcv) from [<80478a24>] (netlink_unicast+0x16c/0x238)
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.570037] [<80478a24>] (netlink_unicast) from [<80478e84>] (netlink_sendmsg+0x2e0/0x3c4)
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.570060] [<80478e84>] (netlink_sendmsg) from [<804326f0>] (sock_sendmsg+0x94/0xb8)
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.570083] [<804326f0>] (sock_sendmsg) from [<80433ce8>] (___sys_sendmsg.part.31+0x294/0x2a4)
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.570105] [<80433ce8>] (___sys_sendmsg.part.31) from [<80434b64>] (__sys_sendmsg+0x5c/0x8c)
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.570126] [<80434b64>] (__sys_sendmsg) from [<80434bac>] (SyS_sendmsg+0x18/0x1c)
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.570147] [<80434bac>] (SyS_sendmsg) from [<8000eac0>] (ret_fast_syscall+0x0/0x48)
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.570156] ---[ end trace c51e57dbe3d8ca94 ]---
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.570166] ------------[ cut here ]------------
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.570269] WARNING: CPU: 0 PID: 12016 at net/mac80211/ieee80211_i.h:895 sta_apply_parameters.isra.22+0x3ac/0x3b4 [mac80211]()
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.570342] Modules linked in: 8021q garp stp llc usbserial ctr ccm snd_bcm2835 snd_pcm snd_seq snd_seq_device snd_timer snd arc4 rt2800usb rt2800lib rt2x00usb rt2x00lib mac80211 cfg80211 crc_ccitt rfkill evdev cdc_acm uio_pdrv_genirq uio
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.570355] CPU: 0 PID: 12016 Comm: hostapd Tainted: G        W      3.18.11-v7+ #781
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.570379] [<800172bc>] (unwind_backtrace) from [<80012f7c>] (show_stack+0x20/0x24)
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.570397] [<80012f7c>] (show_stack) from [<80520970>] (dump_stack+0x98/0xd8)
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.570417] [<80520970>] (dump_stack) from [<80025618>] (warn_slowpath_common+0x7c/0xa0)
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.570438] [<80025618>] (warn_slowpath_common) from [<800256f8>] (warn_slowpath_null+0x2c/0x34)
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.570539] [<800256f8>] (warn_slowpath_null) from [<7f0d2e4c>] (sta_apply_parameters.isra.22+0x3ac/0x3b4 [mac80211])
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.570719] [<7f0d2e4c>] (sta_apply_parameters.isra.22 [mac80211]) from [<7f0d32a0>] (ieee80211_add_station+0xf0/0x2bc [mac80211])
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.570889] [<7f0d32a0>] (ieee80211_add_station [mac80211]) from [<7f03c024>] (nl80211_new_station+0x31c/0x470 [cfg80211])
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.571008] [<7f03c024>] (nl80211_new_station [cfg80211]) from [<80479cfc>] (genl_rcv_msg+0x25c/0x3dc)
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.571051] [<80479cfc>] (genl_rcv_msg) from [<80479130>] (netlink_rcv_skb+0xbc/0xd0)
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.571070] [<80479130>] (netlink_rcv_skb) from [<80479a8c>] (genl_rcv+0x30/0x44)
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.571089] [<80479a8c>] (genl_rcv) from [<80478a24>] (netlink_unicast+0x16c/0x238)
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.571107] [<80478a24>] (netlink_unicast) from [<80478e84>] (netlink_sendmsg+0x2e0/0x3c4)
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.571126] [<80478e84>] (netlink_sendmsg) from [<804326f0>] (sock_sendmsg+0x94/0xb8)
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.571146] [<804326f0>] (sock_sendmsg) from [<80433ce8>] (___sys_sendmsg.part.31+0x294/0x2a4)
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.571166] [<80433ce8>] (___sys_sendmsg.part.31) from [<80434b64>] (__sys_sendmsg+0x5c/0x8c)
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.571186] [<80434b64>] (__sys_sendmsg) from [<80434bac>] (SyS_sendmsg+0x18/0x1c)
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.571205] [<80434bac>] (SyS_sendmsg) from [<8000eac0>] (ret_fast_syscall+0x0/0x48)
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.571211] ---[ end trace c51e57dbe3d8ca95 ]---
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.571235] ------------[ cut here ]------------
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.571333] WARNING: CPU: 0 PID: 12016 at net/mac80211/rate.h:65 ieee80211_add_station+0x2a8/0x2bc [mac80211]()
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.571402] Modules linked in: 8021q garp stp llc usbserial ctr ccm snd_bcm2835 snd_pcm snd_seq snd_seq_device snd_timer snd arc4 rt2800usb rt2800lib rt2x00usb rt2x00lib mac80211 cfg80211 crc_ccitt rfkill evdev cdc_acm uio_pdrv_genirq uio
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.571413] CPU: 0 PID: 12016 Comm: hostapd Tainted: G        W      3.18.11-v7+ #781
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.571435] [<800172bc>] (unwind_backtrace) from [<80012f7c>] (show_stack+0x20/0x24)
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.571453] [<80012f7c>] (show_stack) from [<80520970>] (dump_stack+0x98/0xd8)
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.571472] [<80520970>] (dump_stack) from [<80025618>] (warn_slowpath_common+0x7c/0xa0)
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.571492] [<80025618>] (warn_slowpath_common) from [<800256f8>] (warn_slowpath_null+0x2c/0x34)
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.571592] [<800256f8>] (warn_slowpath_null) from [<7f0d3458>] (ieee80211_add_station+0x2a8/0x2bc [mac80211])
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.571764] [<7f0d3458>] (ieee80211_add_station [mac80211]) from [<7f03c024>] (nl80211_new_station+0x31c/0x470 [cfg80211])
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.571959] [<7f03c024>] (nl80211_new_station [cfg80211]) from [<80479cfc>] (genl_rcv_msg+0x25c/0x3dc)
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.572011] [<80479cfc>] (genl_rcv_msg) from [<80479130>] (netlink_rcv_skb+0xbc/0xd0)
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.572030] [<80479130>] (netlink_rcv_skb) from [<80479a8c>] (genl_rcv+0x30/0x44)
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.572049] [<80479a8c>] (genl_rcv) from [<80478a24>] (netlink_unicast+0x16c/0x238)
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.572068] [<80478a24>] (netlink_unicast) from [<80478e84>] (netlink_sendmsg+0x2e0/0x3c4)
[... 4 lines cut ...]
[... cut ...]
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.574399] [<804326f0>] (sock_sendmsg) from [<80433ce8>] (___sys_sendmsg.part.31+0x294/0x2a4)
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.574421] [<80433ce8>] (___sys_sendmsg.part.31) from [<80434b64>] (__sys_sendmsg+0x5c/0x8c)
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.574443] [<80434b64>] (__sys_sendmsg) from [<80434bac>] (SyS_sendmsg+0x18/0x1c)
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.574465] [<80434bac>] (SyS_sendmsg) from [<8000eac0>] (ret_fast_syscall+0x0/0x48)
Jul 24 22:43:11 raspberrypi3 kernel: [ 3322.574473] ---[ end trace c51e57dbe3d8ca97 ]---
```

scottleibrand, 2015-07-25:
I was able to get things working with a different config, so if this doesn't look like a problem with the kernel itself, feel free to close out the issue.

pelwell, 2015-07-25:
The warning appears to be caused by some missing band configuration data. As this is upstream code, we would ask you to report it to the maintainers of that code. But I think you are running a 3.18 kernel and we are on 4.0, soon moving to 4.1; it is worth updating to a newer kernel before contacting them as the bug may already have been fixed.

scottleibrand, 2015-07-25:
Ok, thanks.  Closing this issue then.

closed 2015-07-25 by scottleibrand