raspberrypi/linux #1090: Enable Mediatek mt7601u wifi driver in 4.2 tree

clivem, 2015-07-28:
Please enable the Mediatek mt7601u wifi driver in the 4.2 tree.

CONFIG_WL_MEDIATEK=y
CONFIG_MT7601U=m

I don't actually use it myself, but remember someone opened an issue in the past about getting a driver included for this wifi dongle, to which the response was something along the lines of, "when it's upstreamed we'll enable it". It got added with 4.2rc3 IIRC.

popcornmix, 2015-07-28:
Added.

clivem, 2015-08-11:
Popcornmix, I just noticed the other "issue" that was closed, someone suggesting moving to 4.2 quickly, to gain in-tree support for the mt7601u. What about back-porting the driver from 4.2 to 4.1. I've been applying the following on top of RPI 4.1 tree....
http://www3.squeezecommunity.org/patches/0001-backport-mediatek-mt7601u-from-v4.2-rc3.patch
Just a suggestion. There do seem to be quite a few RPI users with a mt7601u based dongle and most of them seem to struggle with compiling modules to support it. I have put many gigs of data through the driver patched into RPI 4.1 kernel without any issues.

clivem, 2015-08-13:
Ok, cool. See you have a applied the backport patch to rpi-4.1.y. Also need to make sure the Linux distribution (raspbian????) supplies the firmware for an out-of-the-box experience. The driver needs to load /lib/firmware/mt7601u.bin to bring the dongle up. Don't think that has been added to linux-firmware.git yet. Seem to recall that is obtained from the MediaTek vendor tarball. (File is called MT7601.bin in their tarball.) This is from memory. Not in front of my "work" computer at the moment. I'll post a new comment later today confirming this.

popcornmix, 2015-08-13:
Okay, rpi-update should contain the driver. Can you test it and confirm it works?

Note: We don't normally support out of tree drivers, but as this is from upstream 4.2 and looks safe, I've included it until we move to 4.2. If there are problems reported I'll drop the patch.

@XECDesign would be good to include the firmware file for this in repo.

clivem, 2015-08-13:
Just a quick note.... The firmware I have been using is obtained from downloading the vendor (MediaTek) driver from http://www.mediatek.com/en/downloads1/downloads/mt7601u-usb/
Extract the archive and ....
cp DPO_MT7601U_LinuxSTA_3.0.0.4_20130913/mcu/bin/MT7601.bin /lib/firmware/mt7601u.bin

XECDesign, 2015-08-13:
Added to the todo list.

letetfly, 2015-08-18:
I have testet this 4.1.5 with dwa-127 with mt7601u chip. Only had to manually provide the MT7601.bin to the location /lib/firmware/MT7601u.bin to get i work. Thats it.
Thx for putting it almost in place, much easier and quicker than compiling your self or waiting for 4.2 to officially be released.

heniotierra, 2015-09-04:
Yes, but there's a thing to note. The driver from Mediatek is the old one. The new one (which is to be released with kernel 4.2) was written based on Mediatek's, but is open source and seems like is a better driver.

XECDesign, 2015-09-04:
@popcornmix What's the preferred firmware here?

popcornmix, 2015-09-04:
We're using the driver from 4.2 (which is the kuba-moo one). I believe that recommends using the mediatek firmware as describes by @clivem in https://github.com/raspberrypi/linux/issues/1090#issuecomment-130628097
So, I think mt7601u.bin is fine.

MrEngman, 2015-09-04:
To download and install the firmware I find it rather easier using the command 
```
sudo wget https://github.com/porjo/mt7601/raw/master/src/mcu/bin/MT7601.bin -O /lib/firmware/mt7601u.bin
```

pelwell, 2015-09-16:
@clivem I think this issue is fixed - Close if you agree.

clivem, 2015-09-16:
Yes, closing this "issue"...... The mt7601u driver is working very nicely with those 4x "cherry picked" patches applied. I'm currently stress testing the driver, transferring 16GB files backward and forward across the network. Seeing sustained transfers greater than 8MB/s..... So it's a nice "cheap" alternative to the usual RTL8188CUS recommendation. Seeing sustained transfers greater than 8MB/s..... So it's a nice "cheap" alternative to the usual RTL8188CUS recommendation, when you can still pick up generic MT7601U dongles for less than £3 on ePay, with the external antenna, which might prove useful in less than ideal circumstances, where your Pi is some distance from AP, and the external antenna is the better option, than the built in antenna with the RTL8188CUS nano dongle. YMMV.

I'm not recommending this (or any other vendor)..... But just so others can see what I'm currently testing the mt7601u driver with on Pi. (I have several of these as well as the built-in antenna, nano model.) http://www.ebay.co.uk/itm/271963577350

closed 2015-09-16 by clivem

ClaymorePT, 2016-05-11:
Hello!
I know this issue has been closed for a while, but I'm having trouble in using a mt7601u usb device in ad-hoc mode or AP mode.
Can you tell me if the kernel 4.2 module for mt7601u support these modes? I have already tried to change the operation mode using iwconfig, bu I always get "Operation not supported". 
I check the sources and there is code implemented for the ad-hoc mode, which causes me some confusion about the support.

airtonoliveira-oplen, 2016-07-02:
Hi this link https://github.com/porjo/mt7601/raw/master/src/mcu/bin/MT7601.bin is not working...

clivem, 2016-07-02:
Get it from the linux-firmware github....
http://git.kernel.org/cgit/linux/kernel/git/firmware/linux-firmware.git/tree/mt7601u.bin

airtonoliveira-oplen, 2016-07-02:
file found ok, but dmesg: mt7601u 1-1.5:1.0: Invalid firmware image

clivem, 2016-07-02:
Erm, don't know what the problem is for you. That's the firmware I use.

```
mt7601u 1-1.5:1.0: ASIC revision: 76010001 MAC revision: 76010500
mt7601u 1-1.5:1.0: Firmware Version: 0.1.00 Build: 7640 Build time: 201302052146____
mt7601u 1-1.5:1.0: Warning: unsupported EEPROM version 0d
mt7601u 1-1.5:1.0: EEPROM ver:0d fae:00
usbcore: registered new interface driver mt7601u
```

airtonoliveira-oplen, 2016-07-02:
Sorry, file is ok, but still not working yet, raspbian lite 4.4.13+
  ```
   mt7601u 1-1.5:1.0: ASIC revision: 76010001 MAC revision: 76010500
   mt7601u 1-1.5:1.0: Firmware Version: 0.1.00 Build: 7640 Build time: 201302052146____
   mt7601u 1-1.5:1.0: Error: your device needs default EEPROM file and this driver doesn't support it!
   mt7601u: probe of 1-1.5:1.0 failed with error -22
   usbcore: registered new interface driver mt7601u
```

clivem, 2016-07-02:
I get "unsupported EEPROM" from several different mt7601u devices I have, but have never seen that "Error: your device needs default EEPROM file and this driver doesn't support it!" msg before.

tomatos666, 2016-07-28:
Hello,
i have the same problem. Some one have a working link?

ClaymorePT, 2016-07-28:
I'm able to use the wireless device as a client but not as an Access Point,
under OpenWRT (linux 4.4 kernel)

clivem, 2016-07-28:
> i have the same problem. Some one have a working link?  

Not a working link. Just that the driver developer was aware that there were problems with some devices.

https://github.com/kuba-moo/mt7601u/issues/37#issuecomment-142225566