raspberrypi/linux #1095: 8192cu WLAN breaks internet connection often, seemingly related to rpi-update to kernel 4.0.

ThomasKorimort, 2015-08-04:
Hi!
Since some time my WLAN connection which used to work properly apart from speed with EDIMAX WIFI dongle with RTL8188CUS chipset using RPI 2 B firmware driver 8192cu breaks too often to be useful now. I think this is maybe due to a firmware update. There was warning that the update could make some drivers not work anymore (kernel 4.0). Is this such a case? Further: Why do you provide an update, if some drivers don't work anymore afterwards?
I hope this issue can be solved nicely soon.

Many greetings, Thomas Korimort.

ThomasKorimort, 2015-08-17:
Update: Afte updating another time with Hexxeh script i have looked and saw that suddenly for whatever reaon uname lists again 3.18 trunk of RPi2. I don't know why this is. But still the internet connection problem resides. It seems strange to me as according to the Hexxeh script there should be 4.1 kernel but there is old kernel. For starting i used an image of DEbian 8.0 Jessie not RaspBian. Maybe this is why this Hexxeh script does not work properly? I used the image at https://images.collabora.co.uk/rpi2/ and upgraded Debian Jessie 8.0 from that trunk.

popcornmix, 2015-08-17:
Are you using NOOBS. How full is your sdcard?
Can you post output of rpi-update.
Note: you can force an rpi-update by
``sudo rm /boot/.firmware_revision && sudo rpi-update``

ThomasKorimort, 2015-08-17:
I am not using NOOBS. When i installed my RPi2B some months ago, NOOBS was not supporting Debian 8.0 Jessie. Thus i had to use the above-mentioned Image which is a custom-compiled bootstrap image from some knowledgeable person who knows how to bootstrap a running Debian 8.0 Linux system on the RPi2B.

I followed your procedure of deleting .Firmware_revision Prior to executing rpi-update, but still the 3.8.15 trunk is listed by uname -v. Maybe this is because this Image puts the files into a subfolder instead of the Standard one?

My SD Card is 128 GB and after installing the Image of 5 GB Partition size i enlarged the Partition to the full size of 128 GB. What i would like to have is Debian Jessie for my RPi2B. It was suggested that one could use the Debian HardFP Standard distro for the RPi2B (ARMv7 processor), but i was not clever enough to find out how the Basic Image is put onto the SD Card.

NOOBS i don't want to use as Long as it is not supporting Debian Jessie.

popcornmix, 2015-08-17:
rpi-upate is designed for raspbian. If you are using a different distribution (that presumably doensn't mount /dev/mmcblk0p1 to /boot) then you are on your own. Best ask whoever created the image you are using how best to update the kernel.

ThomasKorimort, 2015-08-17:
ok, thanks. Do you know what is the Basic steps to get a bootable System Image for my RPi2B from the Standard distro of Debian Jessie?

popcornmix, 2015-08-17:
There is an official Jessie raspbian image being worked on now. No definite release date, but hopefully in September.

It is quite easy to edit /etc/apt/sources.list to change wheezy to jessie and apt-get dist-upgrade, but some packages will break.

There a number of unofficial Jessie images linked from the forums, but I can't say for certain which ones work fine.

I'd suggest you wait for the official image if you don't want to have to fix the issues that arise yourself.

ThomasKorimort, 2015-08-17:
yes, good idea. but still i don't understand why my internet problem is there..

popcornmix, 2015-08-17:
What does ``uname -a`` report. It's not clear if your image is even using an official kernel.

ThomasKorimort, 2015-08-17:
actually i have some time, but i am mostly solving mathematical problems and problems related to church law, although recently i am also delving into other law codices and computer topics more and more. Since i know that the Hyperboreans were a technologically superior civilization i got interested again in the technical sciences... :-)

ThomasKorimort, 2015-08-17:
SMP PREEMPT Debian 3.18.5-1~exp1.co1 (2015-02-02) armv7l GNU/Linux 

uname -a is same like uname -v

popcornmix, 2015-08-17:
Yes, that is a kernel not built by us, so I can't help with any issues you have.
Install an official raspbian image, and if you have problems then report it here.

For problems with a third party kernel, you have to take your problems to the people who made the image. I have no idea what kernel tree, patches or config options produced that kernel or the rest of the image.

Ruffio, 2016-08-17:
@ThomasKorimort has your issue been resolved? If so, please close this issue. Thanks.

closed 2017-05-17 by JamesH65