raspberrypi/linux #1106: rpi-4.1.y tree fails to compile (fc98e35c89f1f7dae12db3d50878d96c5552e813)

clivem, 2015-08-18:
Most likely due to the merge of the spi patches yesterday.

  CC [M]  drivers/spi/spi-bcm2708.o
drivers/spi/spi-bcm2835.c: In function 'bcm2835_spi_probe':
drivers/spi/spi-bcm2835.c:772:23: error: 'bcm2835_spi_handle_err' undeclared (first use in this function)
  master->handle_err = bcm2835_spi_handle_err;
                       ^
drivers/spi/spi-bcm2835.c:772:23: note: each undeclared identifier is reported only once for each function it appears in
scripts/Makefile.build:264: recipe for target 'drivers/spi/spi-bcm2835.o' failed
make[2]: *** [drivers/spi/spi-bcm2835.o] Error 1
make[2]: *** Waiting for unfinished jobs....

popcornmix, 2015-08-18:
Should be fixed now.

clivem, 2015-08-18:
Confirm. Fixed. Thank you.

closed 2015-08-18 by popcornmix