raspberrypi/linux #1108: DMA: when spi and i2s dma enabled drivers are both loaded the system will crash

msperl, 2015-08-19:
when both:
* snd-soc-bcm2708-i2s
* spi-bcm2835

are loaded the rpi becomes unresponsive and no messages are printed on the serial console.
Only a reset/power cycle will bring the system back online.

There seems to be an issue with the dma-engine code that will trigger this issue...

HiassofT, 2015-08-19:
I'm seeing a similar issue here (RPi2 with kernel 4.1) with the following config (no external hardware required)
```
dtparam=spi=on
dtoverlay=rpi-dac
 ```
Raspbian boots up fine and the drivers are loaded, but as soon as I try to play back an audio file (eg `speaker-test -c 2`) the RPi locks up hard - no response to pings.

If I disable SPI or switch to spi-bcm2708 no lockups occur.

pelwell, 2015-08-20:
I think I may have found the problem. Channel 12 shouldn't be marked as being available because it doesn't have a dedicated interrupt line; instead, it shares channel 11's (along with 13 & 14). The driver doesn't cope with this, so only channel 11 from this group should be used. The interrupt which would logically belong to channel 12 (<1 28>) is actually a global interrupt for ALL channels. Enabling could easily cause a lockup.

See https://github.com/raspberrypi/linux/pull/1113

msperl, 2015-08-20:
Stupid question: Bulk DMA channel - It does not say anywhere in the documentation anything about this!
Searching for BULK/Bulk/bulk in the PDF does not give any result...

msperl, 2015-08-20:
OK - found it:
```
DMA channel 0 and 15 are fitted with an external 128 bit 8 word read FIFO. This enables efficient memory to memory transfers to be performed. This FIFO allows the DMA to 
```

HiassofT, 2015-08-20:
A quick test with #1113 shows that the rpi-dac + spi test no longer locks up. Thanks a lot @pelwell!

pelwell, 2015-08-20:
Thanks for finding the problem at such a convenient time.

pelwell, 2015-08-26:
Can we close this now?

closed 2015-08-26 by msperl