raspberrypi/linux #1109: DMA: DT: there are 2 DMA channels that are in use but which are NOT documented in the device tree

msperl, 2015-08-19:
there are 2 DMA channels that are in use but which are not documented in the device tree:
```
root@rpi2a:~# head /sys/class/dma/dma0chan*/in_use
==> /sys/class/dma/dma0chan0/in_use <==
1

==> /sys/class/dma/dma0chan1/in_use <==
1

==> /sys/class/dma/dma0chan2/in_use <==
0

==> /sys/class/dma/dma0chan3/in_use <==
0

==> /sys/class/dma/dma0chan4/in_use <==
0
```
I guess these are related to: USB and MMC, but for some reason these are NOT defined in the device tree...

popcornmix, 2015-08-19:
I think chan 0 is used for framebuffer acceleration:
https://github.com/raspberrypi/linux/commit/5980b746d7eafc846fe82697818d077a873145e2

HiassofT, 2015-08-19:
According to dmesg channel 0 is quite certainly used by the framebuffer:
```
[    0.263872] bcm2708-dmaengine 3f007000.dma: DMA legacy API manager at f3007000, dmachans=0x7f35
...
[    0.389476] BCM2708FB: allocated DMA channel 0 @ f3007000
```
Channel 1 could be mmc or usb...

Is there some way to find out at runtime which DMA channel is used by which driver, something like /proc/dma on PCs?

msperl, 2015-08-19:
@popcornmix: that is why I would like to see all the dmas defined in the device-tree...
@HiassofT: I fear there is no way to do that...

pelwell, 2015-08-20:
Which device tree are you looking at? bcm2708_common.dtsi on rpi-4.1.y and rpi-4.2.y says:
```
		mmc: mmc@7e300000 {
			compatible = "brcm,bcm2835-mmc";
			reg = <0x7e300000 0x100>;
			interrupts = <2 30>;
			clocks = <&clk_mmc>;
			dmas = <&dma 11>,
			       <&dma 11>;
			dma-names = "tx", "rx";
			status = "disabled";
		};
```

msperl, 2015-08-20:
I have just tried:
```
root@rpi2a:~# dtc -Ifs /proc/device-tree | grep dmas
Warning (avoid_default_addr_size): Relying on default #address-cells value for /axi/vc_mem
Warning (avoid_default_addr_size): Relying on default #size-cells value for /axi/vc_mem
			dmas = <0x6 0xb 0x6 0xb>;
			dmas = <0x6 0x6 0x6 0x7>;
```
So there is at least one missing - and it is not spi...
A closer look reveals arch/arm/boot/dts/bcm2708_common.dtsi:
```
                i2s: i2s@7e203000 {
                        compatible = "brcm,bcm2708-i2s";
                        reg = <0x7e203000 0x24>,
                              <0x7e101098 0x08>;

                        //dmas = <&dma 2>,
                        //       <&dma 3>;
                        dma-names = "tx", "rx";
                        status = "disabled";
                };
```
(at least with 4.2.y)

So I wonder how i2s is working with dma not configured...

pelwell, 2015-08-20:
```
#define BCM2708_DMA_DREQ_PCM_TX		2
#define BCM2708_DMA_DREQ_PCM_RX		3
```
and
```
	/* Set the DREQ */
	dev->dma_data[SNDRV_PCM_STREAM_PLAYBACK].slave_id =
		BCM2708_DMA_DREQ_PCM_TX;
	dev->dma_data[SNDRV_PCM_STREAM_CAPTURE].slave_id =
		BCM2708_DMA_DREQ_PCM_RX;
```
and
```
	snd_soc_dai_init_dma_data(dai,
			&dev->dma_data[SNDRV_PCM_STREAM_PLAYBACK],
			&dev->dma_data[SNDRV_PCM_STREAM_CAPTURE]);
```

msperl, 2015-08-20:
But that should actually be sitting in the DT and not in the driver...

msperl, 2015-08-20:
but that is the way it is...

msperl, 2015-08-20:
Note that there is an almost identical version of the sound driver that is upstream (also by Florian Meier) that does take the values from the device-tree...

msperl, 2015-08-20:
I guess this is now solved

closed 2015-08-20 by msperl