raspberrypi/linux #1110: DMA: Firmware: please provide more than 5 DMA channels to the linux kernel

msperl, 2015-08-19:
please either provide more static channels so that we can run spi + i2s + USB + MMC concurrently on the system or provide a "registration" system to allow for better sharing of DMA between VC and ARM (say via mbox messages).

HiassofT, 2015-08-19:
+1 for that request

I'm currently facing the issue that the Cirrus audio card driver no longer works with the 4.1 kernel.

Only 3 DMA channels out of the 5 are available, the driver needs SPI to configure the wm5102 (SPI needs 2 DMA channels) and the bcm2708-i2s also needs 2 DMA channels because the wm5102 supports both playback and record. So it's short by 1 DMA channel and bcm2708-i2s won't load...

I noticed a hardcoded limit of 5 in bcm2708-dmaengine.c inside the `#ifdef CONFIG_DMA_BCM2708_LEGACY` block of bcm2835_dma_probe. Does that have anything to do with the 5 channel limit perhaps?

pelwell, 2015-08-19:
I am investigating. It's my first time looking at the DMA driver(s), but the initial observation is that we are using the LEGACY_API mode which pre-configures 5 channels. This may be significant...

msperl, 2015-08-19:
But then the allocation should fail and not loop forever in kernel context (or so it seems)...

Do we need legacy mode?

I know that we can not use the upstream DMA driver as it does not contain all the features - there is a merge request by @notro, but that never got merged upstream...

HiassofT, 2015-08-19:
With the Cirrus driver DMA allocation fails rather gracefully. bcm2708-i2s gets the playback channel and then errors out when trying to acquire the record channel:
```
[    4.695533] bcm2708-i2s 3f203000.i2s: Missing dma channel for stream: 1
[    4.702319] bcm2708-i2s 3f203000.i2s: ASoC: pcm constructor failed: -22
```
At this point all 5 dma channels are marked in_use, if I rmmod bcm2708-i2s channel 4 is freed - nothing too unusual so far.

The hard lock up #1108 might be caused by something else.

msperl, 2015-08-19:
So I have done my testing and have come to the conclusion:
* assuming the sequence: modprobe snd_soc_rpi_dac; modprobe spi-bcm2835
  * I can load a device that does not trigger DMA transfers and everything works fine (modprobe ads7846)
  * but as soon as I load a device that DOES make use of DMA the system is "failing" possibly overwriting memory or similar (modprobe fb_s6d02a1)
* assuming the sequence: modprobe spi-bcm2835; modprobe snd_soc_rpi_dac
  * I can load a device that does not trigger DMA transfers and everything works fine (modprobe ads7846)
  * I can also load a device that DOES make use of DMA without issues (modprobe fb_s6d02a1)
  * no idea if sound would trigger the lockup...

msperl, 2015-08-19:
Recompiling the kernel 4.2 without CONFIG_DMA_BCM2708_LEGACY results in 6 dma-channels in /sys/class/dma and no more lock-ups when playing audio or loading the fbtft device (which initiates a clear screen automatically)

notro, 2015-08-19:
The legacy DMA API is needed because the fb driver needs DMA mem to mem copy which the dmaengine driver doesn't support.
bcm2708-dmaengine is an amalgam of new an old following this PR: https://github.com/raspberrypi/linux/pull/945

This is the part that makes 5 lite channels available to DT clients:
```c
  for (i = 0; i < 5; i++) {
    void __iomem *chan_base;
    int chan_id;

    chan_id = bcm_dma_chan_alloc(BCM_DMA_FEATURE_LITE,
      &chan_base,
      &irq);

    if (chan_id < 0)
      break;

    rc = bcm2708_dma_chan_init(od, chan_base, chan_id, irq);
    if (rc)
      goto err_no_dma;
  }
```

Try adding a section something like this to see if that gives 1 more channel:
```c
  for (i = 0; i < 1; i++) {
    void __iomem *chan_base;
    int chan_id;

    chan_id = bcm_dma_chan_alloc(BCM_DMA_FEATURE_NORMAL_ORD,
      &chan_base,
      &irq);

    if (chan_id < 0)
      break;

    rc = bcm2708_dma_chan_init(od, chan_base, chan_id, irq);
    if (rc)
      goto err_no_dma;
  }
```
I think 7 normal channels should be available.

Not enabling DMA_BCM2708_LEGACY probably results the fb driver not loading.

msperl, 2015-08-19:
Why does the fb thingy require the legacy version - why can't it work the "normal" way?

Then we no longer need this "hacky" thingy and we could eventually move to drivers/dma/bcm2835-dma,c?

notro, 2015-08-19:
> Why does the fb thingy require the legacy version

because

> the fb driver needs DMA mem to mem copy which the dmaengine driver doesn't support

This need will go away when the vc4 driver is in place (or someone adds mem-to-mem support to the dmaengine driver).

pelwell, 2015-08-19:
I'm hoping to put in a dynamic split between the two sides, rather than having to statically split the pool in two.

notro, 2015-08-19:
Is this necessary? I thought only the fb driver used the legacy api, so all available channels except 0 can be given to dmaengine.

notro, 2015-08-19:
How about removing the CONFIG_DMA_BCM2708_LEGACY part from bcm2835_dma_probe() and add the call to bcm_dmaman_probe().
Then hardcode dmachans in bcm_dmaman_probe() to only include the bulk channel 0.

bcm2835_dma_probe() will exclude channels 0,2,3 (not sure why 2 and 3 are excluded), and bcm_dmaman_probe() will have channel 0 for the fb driver to use.

Would that work?

msperl, 2015-08-20:
I was looking a the same - see http://lxr.free-electrons.com/source/drivers/dma/bcm2835-dma.c#L617
So dma 0,2 and 3 are not used/masked.

If we got just one "consumer" of the legacy interface plus it always uses fixed dma 0, then fix that use instead of patching the legacy code - at that point we could even switch to the upstream bcm2835-dma driver.

msperl, 2015-08-20:
just in case: here a patch that would disable DMA for fb_copyarea when LEGACY is disabled:
https://github.com/msperl/linux-rpi/commit/2f4e5e065117b18d4dfd4086c1c4b3af086c3623

I wonder how much more CPU is used using memcopy compared to setting up DMA and then waiting for the interrupt to get triggered (which typically takes 20us on its own) and then the thread woken...

Note that the code is using busy-wait for some transfers (<32k pixel or so it seems), so all that "optimization" of DMA is lost - with actually an extra latency penalty because of the interrupt still needing to get handled...
See:
https://github.com/raspberrypi/linux/blob/rpi-4.1.y/drivers/video/fbdev/bcm2708_fb.c#L595

If it is really of an advantage, then can you share the numbers showing that it produces less Overhead/is faster/...?

popcornmix, 2015-08-20:
The fb_copyarea is used when dragging windows in X. These may be up to 4MB in size (for 32bpp 1080p). There is clearly a cpu benefit (espicially on single core Pi1) to handling that with dma.

msperl, 2015-08-20:
So this leaves us with the option to implement dma_prep_interleaved_dma in DMA engine and switch to that...

pelwell, 2015-08-20:
I have a patch which results in having 7 DMA channels available + the 1 bulk channel available via the legacy API for bcm2708_fb. It was going to be 8+1, but I had to remove channel 12 because it doesn't have a dedicated interrupt line.

pelwell, 2015-08-20:
See https://github.com/raspberrypi/linux/pull/1113.

HiassofT, 2015-08-20:
Thanks @pelwell, the Cirrus driver works fine again with #1113

closed 2015-08-26 by msperl