raspberrypi/linux #1116: "tc filter show dev eth0" - does not show match conditions

J-Kerman, 2015-08-23:
I recently updated to wheezy 2015-05-05 after purchasing the latest Raspberry PI v2 and noticed the "tc filter show" option does not display the IP header match settings. I use my Raspberry PI as a network emulator and have a GUI front-end to retrieve the current emulation settings from TC. Since these match filters are not returned, I cannot display those settings and/or modify them in my GUI.

Here's an example of a filter that captures traffic from src 192.168.0.0/16 to dst 192.168.101.0/24.
>tc qdisc add dev eth0 handle 1: root htb
>tc filter add dev eth0 parent 1: protocol ip prio 0 u32 match ip src 192.168.0.0/16 match ip dst 192.168.101.0/24 flowid 1:1

If you run the command "tc filter show dev eth0", this is what you'll see in Raspbian wheezy:
filter parent 1: protocol ip pref 49152 u32
filter parent 1: protocol ip pref 49152 u32 fh 800: ht divisor 1

On an older build of Raspbian (2014-09-09), Debian jessie (8.1.0), CentOS 6.5, and Slackware 14.1, this is what should be shown:
filter parent 1: protocol ip pref 49152 u32
filter parent 1: protocol ip pref 49152 u32 fh 800: ht divisor 1
filter parent 1: protocol ip pref 49152 u32 fh 800::800 order 2048 key ht 800 bkt 0 flowid 1:1
  match c0a80000/ffff0000 at 12
  match c0a86500/ffffff00 at 16

TC command line tools haven't changed, so perhaps there's something different in the new kernel build that is no longer printing out this information? Anyone have thoughts on how to correct this?

popcornmix, 2015-08-23:
Can you report output of ``uname -a`` for the working and not working raspbian images?

J-Kerman, 2015-08-24:
On the older Raspberry PI v1 with Raspbian release 2014-09, here's the printout:
Linux 3.12.28+ #709 PREEMPT Mon Sep 8 15:28:00 BST 2014 armv6l GNU/Linux

On the newer Raspberry PI v2 with Raspbian release 2015-05, here's the printout:
Linux 3.18.11-v7+ #781 SMP PREEMPT Tue Apr 21 18:07:59 BST 2015 armv7l GNU/Linux

popcornmix, 2015-08-24:
I've just tried with latest rpi-update firmware:
```
root@raspberrypi:/home/pi# uname -a
Linux raspberrypi 4.1.6-v7+ #810 SMP PREEMPT Tue Aug 18 15:32:12 BST 2015 armv7l GNU/Linux
root@raspberrypi:/home/pi# tc qdisc add dev eth0 handle 1: root htb
root@raspberrypi:/home/pi# tc filter add dev eth0 parent 1: protocol ip prio 0 u32 match ip src 192.168.0.0/16 match ip dst 192.168.101.0/24 flowid 1:1
root@raspberrypi:/home/pi# tc filter show dev eth0
filter parent 1: protocol ip pref 49152 u32 
filter parent 1: protocol ip pref 49152 u32 fh 800: ht divisor 1 
filter parent 1: protocol ip pref 49152 u32 fh 800::800 order 2048 key ht 800 bkt 0 flowid 1:1 
  match c0a80000/ffff0000 at 12
  match c0a86500/ffffff00 at 16
```
which looks like the behaviour you wanted. Can you try ``sudo rpi-update`` and reboot and try again?

J-Kerman, 2015-08-24:
Thanks popcornmix. Updating the firmware / kernel to 4.1.6-v7+ has resolved the issue.

Ruffio, 2016-08-17:
@J-Kerman has your issue been resolved? If so, please close this issue. Thanks.

JamesH65, 2017-05-17:
Fixed

closed 2017-05-17 by JamesH65