raspberrypi/linux #1118: "Bad page state" segfaults

sneakernets, 2015-08-24:
"AutoDoom" is a program that intelligently paths through Doom levels. While there some segfaults here and there that show up even on X86/X86-64 systems, surely the kernel shouldn't thrash around this hard?

Next-to-latest firmware as of August 24 2:04 PM CST, on Raspberry PI Model B.

```
[330583.185394] BUG: Bad page state in process AutoDoom  pfn:14256
[330583.193725] page:db6d8418 count:0 mapcount:0 mapping:  (null) index:0x2
[330583.202097] flags: 0x0()
[330583.207817] page dumped because: page still charged to cgroup
[330583.215453] page->mem_cgroup:00010000
[330583.221009] Modules linked in: bnep rfcomm bluetooth rfkill binfmt_misc snd_bcm2835 snd_pcm snd_seq snd_seq_device snd_timer snd uio_pdrv_genirq uio
[330583.221079] CPU: 0 PID: 6114 Comm: AutoDoom Not tainted 4.1.6+ #810
[330583.221091] Hardware name: BCM2708
[330583.221141] [<c0016d88>] (unwind_backtrace) from [<c0013b50>] (show_stack+0x20/0x24)
[330583.221167] [<c0013b50>] (show_stack) from [<c058692c>] (dump_stack+0x20/0x28)
[330583.221195] [<c058692c>] (dump_stack) from [<c00ec4f0>] (bad_page+0xc8/0x128)
[330583.221220] [<c00ec4f0>] (bad_page) from [<c00efc64>] (get_page_from_freelist+0x714/0x9a8)
[330583.221243] [<c00efc64>] (get_page_from_freelist) from [<c00f0048>] (__alloc_pages_nodemask+0x150/0x8ec)
[330583.221277] [<c00f0048>] (__alloc_pages_nodemask) from [<c011543c>] (handle_mm_fault+0xcf4/0x10a0)
[330583.221307] [<c011543c>] (handle_mm_fault) from [<c058c9d8>] (do_page_fault+0x2d4/0x3f0)
[330583.221329] [<c058c9d8>] (do_page_fault) from [<c0009250>] (do_DataAbort+0x44/0xc4)
[330583.221348] [<c0009250>] (do_DataAbort) from [<c058c32c>] (__dabt_usr+0x4c/0x60)
[330583.221358] Exception stack(0xc2e41fb0 to 0xc2e41ff8)
[330583.221372] 1fa0:                                     00000629 0001cba1 046cd460 00000000
[330583.221389] 1fc0: 046cce40 00000629 b6aa2280 046cce38 0001d1c8 00200000 00000638 00002710
[330583.221403] 1fe0: 00000004 beff30a0 b69e50d0 b69e52d0 68000010 ffffffff
[330583.221412] Disabling lock debugging due to kernel taint
[334342.280112] BUG: Bad page state in process AutoDoom  pfn:08960
[334342.287491] page:db538180 count:0 mapcount:0 mapping:00000004 index:0x2
[334342.297668] flags: 0x0()
[334342.301613] page dumped because: non-NULL mapping
[334342.309775] Modules linked in: bnep rfcomm bluetooth rfkill binfmt_misc snd_bcm2835 snd_pcm snd_seq snd_seq_device snd_timer snd uio_pdrv_genirq uio
[334342.309853] CPU: 0 PID: 6365 Comm: AutoDoom Tainted: G    B           4.1.6+ #810
[334342.309866] Hardware name: BCM2708
[334342.309913] [<c0016d88>] (unwind_backtrace) from [<c0013b50>] (show_stack+0x20/0x24)
[334342.309937] [<c0013b50>] (show_stack) from [<c058692c>] (dump_stack+0x20/0x28)
[334342.309966] [<c058692c>] (dump_stack) from [<c00ec4f0>] (bad_page+0xc8/0x128)
[334342.309991] [<c00ec4f0>] (bad_page) from [<c00efc64>] (get_page_from_freelist+0x714/0x9a8)
[334342.310014] [<c00efc64>] (get_page_from_freelist) from [<c00f0048>] (__alloc_pages_nodemask+0x150/0x8ec)
[334342.310046] [<c00f0048>] (__alloc_pages_nodemask) from [<c011543c>] (handle_mm_fault+0xcf4/0x10a0)
[334342.310075] [<c011543c>] (handle_mm_fault) from [<c058c9d8>] (do_page_fault+0x2d4/0x3f0)
[334342.310096] [<c058c9d8>] (do_page_fault) from [<c0009250>] (do_DataAbort+0x44/0xc4)
[334342.310116] [<c0009250>] (do_DataAbort) from [<c058c32c>] (__dabt_usr+0x4c/0x60)
[334342.310126] Exception stack(0xda2ebfb0 to 0xda2ebff8)
[334342.310140] bfa0:                                     b5c5c000 00000000 00049018 00000000
[334342.310156] bfc0: 047442d0 00000002 00347ecc 029b4fc0 04744308 00000002 00347e40 0023ee90
[334342.310169] bfe0: 00000000 be817060 00000000 b6f6b9b8 28000010 ffffffff
[337870.218888] BUG: Bad page state in process AutoDoom  pfn:089ac
[337870.226286] page:db538c30 count:0 mapcount:0 mapping:  (null) index:0x2
[337870.236307] flags: 0x0()
[337870.240429] page dumped because: page still charged to cgroup
[337870.249889] page->mem_cgroup:00080000
[337870.255107] Modules linked in: bnep rfcomm bluetooth rfkill binfmt_misc snd_bcm2835 snd_pcm snd_seq snd_seq_device snd_timer snd uio_pdrv_genirq uio
[337870.255203] CPU: 0 PID: 6365 Comm: AutoDoom Tainted: G    B           4.1.6+ #810
[337870.255215] Hardware name: BCM2708
[337870.255264] [<c0016d88>] (unwind_backtrace) from [<c0013b50>] (show_stack+0x20/0x24)
[337870.255291] [<c0013b50>] (show_stack) from [<c058692c>] (dump_stack+0x20/0x28)
[337870.255319] [<c058692c>] (dump_stack) from [<c00ec4f0>] (bad_page+0xc8/0x128)
[337870.255345] [<c00ec4f0>] (bad_page) from [<c00efc64>] (get_page_from_freelist+0x714/0x9a8)
[337870.255368] [<c00efc64>] (get_page_from_freelist) from [<c00f0048>] (__alloc_pages_nodemask+0x150/0x8ec)
[337870.255402] [<c00f0048>] (__alloc_pages_nodemask) from [<c011543c>] (handle_mm_fault+0xcf4/0x10a0)
[337870.255431] [<c011543c>] (handle_mm_fault) from [<c058c9d8>] (do_page_fault+0x2d4/0x3f0)
[337870.255452] [<c058c9d8>] (do_page_fault) from [<c0009250>] (do_DataAbort+0x44/0xc4)
[337870.255471] [<c0009250>] (do_DataAbort) from [<c058c32c>] (__dabt_usr+0x4c/0x60)
[337870.255482] Exception stack(0xda2ebfb0 to 0xda2ebff8)
[337870.255495] bfa0:                                     b5c98000 00000000 0000d018 00000000
[337870.255512] bfc0: 0937e0c0 00000002 00347ecc 029b4fc0 0937e0f8 00000002 00347e40 0023ee90
[337870.255527] bfe0: 00000000 be817060 00000000 b6f6b9b8 28000010 ffffffff
[337870.255574] BUG: Bad page state in process AutoDoom  pfn:089ae
[337870.264787] page:db538c78 count:524288 mapcount:0 mapping:  (null) index:0x2
[... 110 lines cut ...]
[338209.309138] bf20: da2ebf74 da2ebf30 c01057c4 c0059b00 00000022 000b5bf5 da2ebf44 da2ebf48
[338209.320710] bf40: c013d73c 00000000 00000000 00000022 00000003 000b1000 00000000 c000f908
[338209.332280] bf60: da2ea000 00000000 da2ebfa4 da2ebf78 ffffffff ffffffff b6f6b9b8 b6f6b9b8
[338209.343838] bf80: 28000010 ffffffff 00000000 b6f6b9b8 28000010 ffffffff 00c5387d 00c5387d
[338209.355406] bfa0: 00000000 da2ebfb0 c058c32c c0009218 b5ca2000 00000000 00003018 00000000
[338209.366972] bfc0: 09b911f0 00000002 00347ecc 029b4fc0 09b91228 00000002 00347e40 0023ee90
[338209.378536] bfe0: 00000000 be817060 00000000 b6f6b9b8 28000010 ffffffff 00000000 00000000
[338209.390125] [<c00fac18>] (isolate_lru_pages.isra.49) from [<c00fcac8>] (shrink_active_list+0xc4/0x494)
[338209.402863] [<c00fcac8>] (shrink_active_list) from [<c00fd4d4>] (shrink_lruvec+0x63c/0x6b0)
[338209.414619] [<c00fd4d4>] (shrink_lruvec) from [<c00fd648>] (shrink_zone+0x100/0x2b0)
[338209.425741] [<c00fd648>] (shrink_zone) from [<c00fd93c>] (do_try_to_free_pages+0x144/0x4a8)
[338209.437483] [<c00fd93c>] (do_try_to_free_pages) from [<c00fdf18>] (try_to_free_pages+0x278/0x434)
[338209.449749] [<c00fdf18>] (try_to_free_pages) from [<c00f03ec>] (__alloc_pages_nodemask+0x4f4/0x8ec)
[338209.462182] [<c00f03ec>] (__alloc_pages_nodemask) from [<c011543c>] (handle_mm_fault+0xcf4/0x10a0)
[338209.474522] [<c011543c>] (handle_mm_fault) from [<c058c9d8>] (do_page_fault+0x2d4/0x3f0)
[338209.485968] [<c058c9d8>] (do_page_fault) from [<c0009250>] (do_DataAbort+0x44/0xc4)
[338209.496962] [<c0009250>] (do_DataAbort) from [<c058c32c>] (__dabt_usr+0x4c/0x60)
[338209.507661] Exception stack(0xda2ebfb0 to 0xda2ebff8)
[338209.514409] bfa0:                                     b5ca2000 00000000 00003018 00000000
[338209.525802] bfc0: 09b911f0 00000002 00347ecc 029b4fc0 09b91228 00000002 00347e40 0023ee90
[338209.537182] bfe0: 00000000 be817060 00000000 b6f6b9b8 28000010 ffffffff
[338209.545474] Code: e5827004 e8870024 e5857000 eaffffe9 (e7f001f2)
[338209.553226] ---[ end trace 7000c4b639cb9053 ]---
[338209.559468] note: AutoDoom[6365] exited with preempt_count 1
```

popcornmix, 2015-08-24:
Are you overclocking?
Any non-default command line options?
Any non-default config.txt options?
Any kernel modules loaded?
Any peripherals connected to Pi?

sneakernets, 2015-08-24:
> Are you overclocking?

Yes, at High, but this has been happening even when not overclocking. This is the first time I was able to grab a crash log, though. I can attempt to reset this and see if I can trigger it again...

> Any non-default command line options?

Not that I know of.

> Any non-default config.txt options?

Some NTSC related tweaks, but since I was using X thru SSH once in a while, I didn't have it plugged up to my TV.

>Any kernel modules loaded?

No more than what's default, I think. I disabled bluetooth services since I'm not using my bluetooth keyboard.

>Any peripherals connected to Pi?

Just my network cable and the 2A power supply that comes with the Canakit.

popcornmix, 2015-08-24:
Disable overclock and confirm the same kernel backtrace occurs.
Was this a standard raspbian sdcard install?

sneakernets, 2015-08-24:
> Was this a standard raspbian sdcard install?

Yes, from the website. I wasn't savvy enough to roll my own thing :)
I am letting it run through now!

sneakernets, 2015-12-02:
I wanted to provide an update on this:  it seems that every overclock setting that has "Overvolt:6" causes these odd kernel panics. Looking on various forums, I could be overheating the RG2. Given that the the board is usually a hot potato running several processes when these crashes happen, I bet that's the problem...

popcornmix, 2015-12-02:
Be aware that over_voltage will increase power consumption so may make a borderline power supply become unreliable. A better power supply may improve the chance of a successful overclock.

Anyway, this doesn't sound like a kernel issue, so I'll close.

closed 2015-12-02 by popcornmix