raspberrypi/linux #1119: How do we handle merging of drivers that (ab)use peripherals for esoteric purposes?

P33M, 2015-08-26:
Some of the peripherals in BCM2835 are generic enough that they can be repurposed as bitstream output/sampling input style applications, such as #710 using PWM serialisation to drive neopixels and the possibility of using AUX_SPI1/2 as a generic low-speed IR/remote TX/RX. I've previously trialled transmitting SPDIF over the PWM peripheral with an LED on a GPIO and it sort-of works.

This creates multiple drivers for the same hardware - how do we handle this? Multiple compatible strings in the base DT?

pelwell, 2015-08-26:
> Multiple compatible strings in the base DT?

I'm not sure what you mean by this, but I'll give you my answers anyway.

#### Answer 1

The philosophy of DT is that you describe the hardware, in a way which should be OS- and user-neutral. Compatible strings are then to identify the components, nothing more.

I understand that viewpoint, but it is limiting. It's almost as if there was a layer missing; although the module aliases provide a compatible string->module mapping, it appears to be assumed to be a one-to-one or many-to-one relationship, involving no user choice. I'd be curious to see if modprobe supports one-to-many, with blacklisting providing the selection mechanism.

But we already use DT in a more pragmatic way. Our tree supports multiple equivalent drivers for the same hardware, with the substring "bcm2708" indicating downstream and "bcm2835" indicating upstream (or in some cases not-yet-upstreamed), so one user-friendly way to handle driver selection is with multiple compatible string, and overlays are the obvious way to change them.

#### Answer 2

Perhaps the "standard" drivers could be extended to include the new functionality as a different mode of operation, or perhaps present an interface on top of which could be layered this new functionality? That seems like a better model. Those layered drivers would then obviously be loaded from /etc/modules (unless we want to abuse DT in a different way - spidev anybody?)

P33M, 2015-08-27:
By this I mean a many-to-one relationship with drivers -> hardware. There's a lot of ephemeral programs out there that make "use" of peripherals in unconventional ways which it would be good to centralise (i.e. pull into the kernel, again to reduce the number of people whacking I/O registers from userspace).

I'm wary of making users select via blacklisting/removal of blacklist - we had this in the days before DT and it was awkward - much better would be a positive selection guided through existing functionality - overlays or similar.

pelwell, 2015-08-27:
So that was my answer 1, and is always there as a fallback. I still think we should consider answer 2 on a case-by-case basis.

P33M, 2015-08-27:
Ok. Looks like at least an outline for Answer 2 (auxiliary interfaces with common enables) is going to appear in mainline soon: http://lists.infradead.org/pipermail/linux-rpi-kernel/2015-August/002193.html

closed 2015-08-27 by P33M