raspberrypi/linux #1121: Packages.gz does not get updated when new kernel images become available

nu11nu11, 2015-08-29:
I wondered why no kernel patches were available from Raspbian repos since April 2015. I took a look on the repos and found the contents of Packages.gz not being up to date, thus not hinting apt-get to the fact a newer kernel is available.

`apt-cache policy linux-image-3.18.0-trunk-rpi2` insists on this being the latest version:
`3.18.5-1~exp1+rpi16`

Looking into it, I found this file on the servers:
http://mirrordirector.raspbian.org/raspbian/pool/main/l/linux-3.18/linux-image-3.18.0-trunk-rpi2_3.18.5-1~exp1+rpi19_armhf.deb

So I looked at the Packages file and found the culprit there - however, no idea on the reason why it didn't get updated.

Contents of http://mirrordirector.raspbian.org/raspbian/dists/jessie/main/binary-armhf/Packages.gz:

```
Package: linux-image-3.18.0-trunk-rpi2
Source: linux-3.18
Version: 3.18.5-1~exp1+rpi16
Architecture: armhf
Maintainer: Debian Kernel Team <debian-kernel@lists.debian.org>
Installed-Size: 59921
Pre-Depends: debconf | debconf-2.0
Depends: kmod | module-init-tools, linux-base (>= 3~), debconf (>= 0.5) | debconf-2.0, initramfs-tools (>= 0.110~) | l
inux-initramfs-tool
Recommends: firmware-linux-free (>= 3~), irqbalance
Suggests: linux-doc-3.18, debian-kernel-handbook, fdutils
Breaks: at (<< 3.1.12-1+squeeze1), initramfs-tools (<< 0.110~)
Provides: linux-modules-3.18.0-trunk-rpi2
Homepage: https://www.kernel.org/
Priority: optional
Section: kernel
Filename: pool/main/l/linux-3.18/linux-image-3.18.0-trunk-rpi2_3.18.5-1~exp1+rpi16_armhf.deb
Size: 15855288
SHA256: 144012f453bb7a175dd68f5fcf21249cec0b0d287ac43b620b6296bb495c8b72
SHA1: a48c2b20e897c4d43bc857b0b1e4b99ae9b1c778
MD5sum: 6e2b41e354f259c7b3b86f64e4f1846a
Description: Linux 3.18 for RaspberryPI2
 The Linux kernel 3.18 and modules for use on Raspberry PI 2.
```

mati75, 2015-08-29:
Please report that on Raspberry Pi forums, here is only kernel code sources.

XECDesign, 2015-08-30:
Actually, that isn't even about this kernel. It's the Debian linux kernel with patches from this repo applied on top. https://www.raspbian.org/RaspbianBugs

closed 2015-08-30 by XECDesign