raspberrypi/linux #1128: Kernel bump to 4.1 breaks Minecraft

XECDesign, 2015-09-09:
After upgrading the kernel, the minecraft windows goes black a few seconds after entering a world.
Although I am not 100% sure about this, it only seems to happen when using NOOBS, but not on a raw image.

The issue is first introduced here:
https://github.com/Hexxeh/rpi-firmware/commit/00bd2e394c777a14bfd21ff6ffbbd0d45c1ca3aa

Previous discussion:
https://github.com/raspberrypi/rpi-sense/issues/4
https://www.raspberrypi.org/forums/viewtopic.php?f=78&t=119545

XECDesign, 2015-09-09:
After compiling a few kernels, I can now also confirm that the exact commit that introduced the problem is a15106c0300ea7958fa7e1cc62c573d999eb4ee2

pelwell, 2015-09-09:
I can reproduce this:
1) start with a clean NOOBS 1.4.1 download
2) install Raspbian and reboot
3) sudo apt-get update
4) sudo apt-get upgrade
5) reboot
6) play Minecraft, and watch the overlay window disappear after about 5 seconds.

If you can get the desktop cursor back (you may need to hit Esc), moving the desktop window will cause the Minecraft widow to appear for a few seconds, then disappear again. If you position the window at the edges of the desktop you can see the Minecraft window in the overscan area even when the rest of the window disappears, enough to tell that the game is still operational.

pelwell, 2015-09-09:
This bug was due to the modification to bcm2708_fb to use the new rpifirmware driver. The old interface, via the old mailbox driver, used two channels - the common channel for manipulating properties, and a special channel for providing the initial framebuffer settings. All values conveyed down the dedicated channel could also be set as individual properties, and this is what the rpifirmware driver does, but this has the unintended consequence of bypassing an extra bit of initialisation that ensures the framebuffer is behind any other display layers (such as the Minecraft window). Without this initialisation, the framebuffer sits at the same depth as other layers, and whichever is modified last tends to end up on top.

The final piece of the puzzle is that while Minecraft is running, every 5 seconds it unblanks the display - perhaps to stop the display going to sleep. When this happens, the framebuffer is effectively re-added, placing it on top of the Minecraft window which was initially having been created after the framebuffer. Moving the Minecraft window brings it to the front again, only to have it pushed back at the next 5-second unblank.

Although the bug was introduced by a kernel change, it will be fixed in the next firmware release by making sure the depth is initialised correctly when the framebuffer is allocated.

XECDesign, 2015-09-09:
Oh, interesting. So it doesn't sound like NOOBS should have anything to do with it. Any idea why everything seems to work fine on a different card with the same kernel and firmware, but without NOOBS?

pelwell, 2015-09-10:
No, I've no idea.

[ Long pause ]

Yes, I understand it now. In my earlier investigation I discovered that arm_display_present (the function that normally sets the layer to -127) was being bypassed with the new rpifirmware driver. What I didn't see, because it didn't happen, was the call to it from arm_display_init that initially configures the framebuffer. With a non-NOOBS installation this path *is* taken, so the layer is being initialised correctly (even though only once), but NOOBS prevents this (by setting disable_splash) because we don't want two rainbow splash screens, hence the layer remains zero.

pelwell, 2015-09-10:
A new firmware was pushed last night - please close when you are happy this has fixed the problem.

XECDesign, 2015-09-11:
Works great. Thanks for the fix and the explanation.

closed 2015-09-11 by XECDesign