raspberrypi/linux #1129: Compute Module ignores i2c_arm from either confit.txt or dt

clivem, 2015-09-11:
dtparam=i2c_arm=on in config.txt on compute module does not result in either i2c0 or i2c2 being enabled. I'd like it to enable i2c1 for compatibility with A+,B+ 2B.

The ignoring of anything referencing i2c_arm on CM also results in the lack of rtc being enabled using the "stock" i2c-rtc overlay definition, (config.txt: dtoverlay=i2c-rtc,ds1307), which references i2c_arm. Which of course can be worked around by specifying either i2c0 or i2c1 in the dts instead, but again, for consistencies sake and to allow the "stock" overlay definition to be used without any mods on all models including CM, it would be nice if either a ref to i2c_arm in either an overlay file or config.txt on CM, caused i2c1 to be enabled on CM.

pelwell, 2015-09-14:
This could easily have been a DT/kernel issue, but as you've worked out the arm/vc aliases are created by the firmware. The thinking behind the absence of the aliases was that ownership of the ports is more flexible on the CM, and that CM users are likely to be creating their own .dtbs anyway.

Having said that, there's probably no harm in having the aliases created anyway. They should appear in a future release.

popcornmix, 2015-09-14:
@pelwell's change should be in latest rpi-update firmware

clivem, 2015-09-14:
Tested using i2c_arm alias on CM with latest firmware. Working! Thank you.

closed 2015-09-15 by pelwell