raspberrypi/linux #1131: Internal error: Oops: 5 [#1] PREEMPT SMP ARM (rpi-4.1.y gitf1e8cc4)

clivem, 2015-09-12:
I've been doing some stress testing of rpi-4.1.y HEAD on Pi2B. A combination of generating load with "haveged -n 0 > /dev/null" and "stress -c2 -i2 -m2" causes an OOPS within minutes....

http://www3.squeezecommunity.org/logs/screenlog.0

clivem, 2015-09-13:
popcornmix,

I've gotten to the bottom of cranky behaviour, oopses and div by zeros.... Long story short, me running around like a headless chicken for the last 24 hours trying to debug this, can you cherry pick and apply these 4 upstream patches to both 4.1 and 4.2 trees for the mt7601u driver....

http://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/patch/drivers/net/wireless/mediatek/mt7601u?id=bed429e1ae8b7ee207e01f3aa60dcc0d06a8ed4d
http://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/patch/drivers/net/wireless/mediatek/mt7601u?id=d9517c0a5d7468a7ea63086057604fcb0fff480e
http://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/patch/drivers/net/wireless/mediatek/mt7601u?id=4513493d188d5e3052aee68eda85eaaa1a4e41c2
http://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/patch/drivers/net/wireless/mediatek/mt7601u?id=78623bfb6f4cbdba3183621e8e0e781611217022

clivem, 2015-09-15:
Can you please get those upstream mt7601u driver patches (mentioned above) into the downstream RPI 4.{1,2} trees? 

(Not a problem on single core bcm2835 where I did most of my testing, but without the spinlock patch, will most certainly cause issues on multi-core bcm2836, whereby people using the mt7601u driver un-patched are going to run into all sorts of "undefined" behaviour. eg. random crashes, oopses, locks, div by zero.....)

pelwell, 2015-09-15:
I've backported those patches, and verified only that it compiles OK. They will be in the next release. 

Please confirm ASAP that the driver still functions, and close if you feel the issue has been addressed.

clivem, 2015-09-15:
I've already compiled and tested both rpi-4.1.y and rpi-4.2.y, with those 4 patches applied. (Which I'd already done prior to asking for them to be applied and did again after the rpi-4.1.y tree was updated to 4.1.7 yesterday.) But I'll have another quick look before closing.....

clivem, 2015-09-15:
Tested.... Closing....

closed 2015-09-15 by clivem